knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #1: M600 - Filament Change funktioniert bei mir nicht

gloomytree, 2020-06-19:
Hi, erstmal ein fettes Dankeschön für das kompilieren der Firmware, ich hätte das wegen dem LCD nicht alleine hinbekommen. Ich habe sie gestern auf meinen Any geflasht und festgestellt, dass M600 nicht funktioniert. Der Drucker reagiert nicht auf den Befehl und druckt einfach weiter. 

Ich habe gesehen, dass bei dir im Repo ADVANCED_PAUSE_FEATURE deaktiviert ist. Ist das beabsichtigt? Das ist für M600 notwendig. 

Doku: https://marlinfw.org/docs/gcode/M600.html

knutwurst, 2020-06-21:
Hey :)
Du hast vollkommen Recht. Das ADVANCED_PAUSE_FEATURE musste aufgrund eines Bugs deaktiviert werden und ich habe ganz übersehen, dass dadurch M600 auch kaputtgeht :/

Im nächsten Release sollte es aber wieder laufen.

Ich lasse das Issue noch mal offen und schließe es, wenn der Fehler beseitig ist, ja? Du bekommst dann eine Mail.

knutwurst, 2020-06-21:
Fixed in db6707d ;) 
Wenn du dir die Firmware aus den sourcen selber baust, klappt es schon. Das neue Feature und der Bugfix kommen sonst aber in die 1.0.5.

closed 2020-06-21 by knutwurst

gloomytree, 2020-06-21:
> Fixed in [db6707d](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/commit/db6707d78e10d209d31d9b6e507ed15265645e4a) ;)
> Wenn du dir die Firmware aus den sourcen selber baust, klappt es schon. Das neue Feature und der Bugfix kommen sonst aber in die 1.0.5.

Awesome! Vielen Dank nochmals für die ganze Arbeit und das Teilen!