knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #2: Denkfehler bei fertig kompilierten Firmwares?

lordhamsterrad, 2020-06-20:
Ich habe TMC-Treiber verbaut und die Stecker nicht gedreht. Deswegen dachte ich, die TMC-Firmware wäre die richtige, aber dann laufen die Achsen verkehrt herum. Mit der Version, die eigentlich für die Standardtreiber vorgesehen ist, funktioniert es jedoch. Wurde da beim Kompilieren etwas durcheinandergebracht oder mache ich einen Denkfehler? Benutze Version 1.0.4

knutwurst, 2020-06-21:
Hey ;)
Magst du mir sagen, welchen Drucker bzw welche .hex Datei du verwendet hast? Also welche geht und welche nicht?

Es *kann* natürlich immer mal sein, dass sich ein Fehler eingeschlichen hat, jedoch höre ich das zum ersten Mal.

Kann es sein, dass du Treiber hast, bei denen die Ausgänge schon gedreht sind? Ist gibt da einige aus China, die schon vom Werk aus anders verdrahtet sind.

lordhamsterrad, 2020-06-21:
Moin,
Danke für die schnelle Antwort. Ich habe nochmal in meinen Unterlagen gegraben und tatsächlich sind an meinen Treibern die Motoranschlüsse schon gedreht 🤦 . Da hatte ich wohl was fasch in Erinnerung, tut mir leid, hätte ich auch vorher drauf kommen können.

closed 2020-06-21 by lordhamsterrad