knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #3: frontal fan Fan2

jekysan, 2020-06-24:
hello, when I use the gcode M106 S255 without the printer printing, the fan runs at maximum power, while if I use the gcode during a print, it starts at maximum power and then stops after 2-3 seconds and resumes at a speed much lower than the maximum one. I use a noctua a4x20 flx. Can you find a solution?

knutwurst, 2020-06-24:
Hello,
this has nothing to do with the firmware or the printer. If you set the fanspeed via usb or the touchscreen, the M106 command is processed but it is also processed when you load a gcode file with an object you want to print. You can open the *.gcode file with an editor and search for the M106 command. I'm sure you'll fine one. And this entry overrides your manual M106 and sets the speed like it was set in the slicer.

knutwurst, 2020-06-26:
I am not quite sure what you are going to do, but you can also simply delete every M106 line or comment it out with a semicolon. This should work for you if you don't want to set the speed in the slicer software.

jekysan, 2020-06-30:
ok, now work again, but there is the same problem the fan not work at maximum speed. if i send the command without the print the fan work at maximum speed, with the print the fan work with a slow speed

knutwurst, 2020-07-04:
Can you revert to the original firmware and use the same gcode from your SD card to verify that the problem does not persist?

Also can you post the gcode you want to print, so I can check it by myself?

jekysan, 2020-07-05:
no unfortunately I can't use the original firmware, I'm using 2209 drivers

this is the gcode, the link expires in a week

https://we.tl/t-M9OhiXTlwd

knutwurst, 2020-07-11:
You have a bug in your GCODE (or slicer settings) ... the nozzle temperature is set to zero degrees (0°C) and the cold extrusion prevention jumps in and disabled the extrusion and the fan.

knutwurst, 2020-07-11:
I tested your gcode and can not see any issues at all. The fan starts at around 70% and when I push it to 100% it stays there. It must be a hardware issue on your side.

I'm sorry I can't help. :/

closed 2020-07-11 by knutwurst