knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #4: New Display in i3 Mega S not fully supported

ketchup67, 2020-07-04:
There is a new display in the i3 Mega S built in.
The special menu is not shown anymore.

knutwurst, 2020-07-04:
Does everything else work as expected or is just the special menu missing? Did you insert an SD Card?

ketchup67, 2020-07-04:
The sd card was inserted.
All other standard functions seem to be fine.

Subfolders and chinese letters are not shown, just empty rows. Only files
inside the root folder were shown.

Due to the bed leveling issues the special submenu would be really very
help- and usefull.

knutwurst, 2020-07-04:
That's technically impossible. Maybe your firmware update did not succeed? Either the display and the file listing works correctly or not at all, because everything is based on the same code.

Can you verify that you've got the Trigorilla Motherboard with Atmega2560 and not the 32bit "Trigorilla Pro"?

ketchup67, 2020-07-04:
I also thought it is technically not possible, but i wasted the whole day
with this simple update.
Several times in Cura, but unfortunately there is no feedback of the
commands. So i installed Octopie on my Raspberry where i can see in the
terminal what is going on.
During the boot procedure your firmware is showing up without any problems.
Also the flash procedure went through successfully. How can i get the
information which board is used in the printer without opening it?

knutwurst, 2020-07-05:
"Oliver Köster <notifications@github.com> schrieb am..." -> are you german? ;) 

Can you tell me which firmware is shown in the display, when you click on Tools -> More -> Info?

ketchup67, 2020-07-05:
Yes, i am german, too.
Bei Info wird V116 angezeigt. Dies ist vermutlich die TFT Firmware Version

knutwurst, 2020-07-05:
(In english - maybe someone else is interested in this topic too)

Could you please install the original firmware and show me a photo of the info-screen on the TFT? I'll investigate the issue. "V116" is the custom firmware label, which is provided by marlin but it is commonly ignored by the display, so I didn't change the value. Normally something like "1.1.2" or "1.4.0rc" is shown.

ketchup67, 2020-07-05:
I already did this, too. I installed the Anycubic Software which is
available on their Website. The display showed V115 after the update was
finished.

knutwurst, 2020-07-05:
Can you upload a photo?

ketchup67, 2020-07-05:
Sure. If you want i can support you with fotos and logfiles (from Octopie).
By the way the Repetiers software was even not able to send out the gcodes.

knutwurst, 2020-07-05:
Photos of the tft info screen before and after the update and also a photo of the non-working special menu would be great. I'll dig into it. Since the Anycubic Mega P (Pro) arrived, there is also a new display firmware out there... sadly anycubic didn't want me to have the firmware :(

ketchup67, 2020-07-05:
Attached the photos after your firmware update. So now i will install the
original Anycubic software again and send you a picture then, too.
As you can see, the original folders or subfolders are not showing again.

knutwurst, 2020-07-05:
Right here, I cannot see any attachments if you reply to an email. You need to log into github and upload the photos there. ;)

ketchup67, 2020-07-05:
Sorry, my mistake.
[image]
[image]
[image]

knutwurst, 2020-07-05:
[image]

Damn! looks like you got the display from the new "Mega P" which also looks like the "Mega X".
That's weird... I'd really like to see the insides of the printer ;)

Seems that I now need another printer with this particular display to edit the firmware. My hope is, that the file list inside the display uses a sanitiser to get rid of special characters (like the ".." or  the "<" of the special menu). Maybe I can work around the issue...

ketchup67, 2020-07-05:
Here the photo after the original firmware update...
Please let me know how i can support you.

[image]

knutwurst, 2020-07-05:
[mega_s_new_tft_01.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/4874847/mega_s_new_tft_01.hex.zip)
Can you please try this firmware and let me know if you see any differences in the file menu?

ketchup67, 2020-07-05:
Yes, sure thank you very much. I am outside right now but 2 - 3 hours later
i will be back home again and try it. Then i will give you feedback asap.
Again, thank you very much!

ketchup67, 2020-07-05:
Right now i flashed your modified firmware. I would say partly succesfull. Please take a look at the attached photos.
The special menu is showing up now but unfortunately not accesable.

[image]
[image]

ketchup67, 2020-07-05:
Ich habe das neue Update bereits geflasht und bereits auf Github einen
Eintrag gemacht.

knutwurst, 2020-07-05:
Wow.. okay. That's exactly what I was trying to achieve!
Give me some time. I'll do my best to get your screen working ;)

ketchup67, 2020-07-05:
Thank you very very much again!

knutwurst, 2020-07-05:
Oh and I forgot: You need to click on "refresh" instead of print, to accept entries in the special menu. But at this point, you might be stuck inside, because the menu handling is still the old one.

ketchup67, 2020-07-05:
I did - i did the rtfm thing :-P

knutwurst, 2020-07-06:
Could you please try this firmware?
If I am correct, you should see and now use the special menu... and maybe I can create a version, where both displays are supported. Also my goal is to use the "Print" key as "enter".

[get_into_special_menu.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/4878677/get_into_special_menu.zip)

ketchup67, 2020-07-06:
I flashed it right now. The situation is same. No reaction if trying to accept the special menue.

[image]
[image]
[image]

knutwurst, 2020-07-06:
[get_into_special_menu_2.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/4879139/get_into_special_menu_2.zip)

Next one... now with debugging output and hopefully working directory-listing. The missing directories are our problem atm to get the correct index.

ketchup67, 2020-07-06:
Here the debug output

[image]

knutwurst, 2020-07-07:
Can you make a photo of the file listing when the original sd card is inserted?

ketchup67, 2020-07-07:
Attached the file view of the SD Card with the Windows Explorer and what you can see on the TFT:

[image]
[image]

knutwurst, 2020-07-07:
The double entry at the top is weird... 
But try this. Maybe you can see folders now.
I really need this touch display to test it properly :/ 

[get_into_special_menu_3.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/4883375/get_into_special_menu_3.zip)

ketchup67, 2020-07-07:
Here are the new pictures including the console output:

[image]
[image]
[image]

knutwurst, 2020-07-07:
Can you click on some entries and show the log output? It's hard to tell what the display does with the data and what the callbacks are. I try to get the thing to do the development and testing by myself :)

ketchup67, 2020-07-07:
By pressing <Special Menu>.gcode:

Recv: TFT Serial Command: A13 <Special Menu>.gcode
Recv: TFT Serial Command: A26
Recv:  TFT Serial Debug: SelectedDirectory: <Special Menu>.gcode
Recv: TFT Serial Debug: Enter Special Menu
Recv: TFT Serial Command: A8 S0
Recv: 0: <Special Menu>.gcode
Recv: 1: Owl_pair.gcode
Recv: 2: parameter.gcode
Recv: 3: /Files_Engish_Mega S.gcode

By pressing /Files_English_Mega S.gcode:

Recv: TFT Serial Command: A13 parame~1.gco
Recv: echo:Now fresh file: parame~1.gco
Recv: File opened: parame~1.gco Size: 13691
Recv: File selected
Recv: TFT Serial Debug: File open successful... J20
Send: M27
Recv: Not SD printing
Recv: ok
[...]

By pressing arrow down, a new page is opening with the second photo i did send.
Pressing now /Robot.gccodeese_Mega S.gcode the display is looking like this with this message:

[image]

Recv: TFT Serial Command: A13 /Files_Japanese_Mega S.gcode
Send: M27
Recv: Not SD printing
Recv: ok
[...]

Pressing /D__-__Mega S.gcode this happens:

[image]

Recv: TFT Serial Command: A13 /Robot.gcode
[...]
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
[...]

knutwurst, 2020-07-07:
Next ;) 
Please use "reload" to select the item.. not "print"! It's not implemented yet.

[get_into_special_menu_4.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/4884058/get_into_special_menu_4.zip)

ketchup67, 2020-07-07:
A big advantage ! But a little complicated to move up and down, because it always jumps out of this point ad must navigate back to there.

Send: M500
Recv: echo:Settings Stored (724 bytes; crc 25040)
Recv: ok
[...]
Recv: TFT Serial Command: A8 S0
Recv: 0: <Special Menu>.gcode
Recv: 1: Owl_pair.gcode_         .gcode
Recv: 2: parameter.gcode_        .gcode
Recv: 3: /Files_Engish_Mega S_    .gcode
[...]
Recv: TFT Serial Command: A13 <Special Menu>.gcode
Recv: TFT Serial Command: A26
Recv:  TFT Serial Debug: SelectedDirectory: <Special Menu>.gcode
Recv: TFT Serial Debug: Enter Special Menu
Recv: TFT Serial Command: A8 S0
[...]
Recv: TFT Serial Command: A26
Recv:  TFT Serial Debug: SelectedDirectory:
Recv: echo:SD card ok
Recv: TFT Serial Command: A8 S0
[...]
Recv: TFT Serial Command: A26
Recv:  TFT Serial Debug: SelectedDirectory:
Recv: echo:SD card ok
Recv: TFT Serial Command: A8 S0
Recv: TFT Serial Command: A8 S4
Recv: TFT Serial Command: A8 S8
Recv: TFT Serial Command: A8 S12
[...]
Recv: TFT Serial Command: A8 S16
[...]
Recv: TFT Serial Command: A13 <Exit>.gcode
Recv: TFT Serial Command: A26
Recv:  TFT Serial Debug: SelectedDirectory: <Exit>.gcode
Recv: TFT Serial Debug: Enter Special Menu
Recv: TFT Serial Command: A8 S0
Recv: 0: <Special Menu>.gcode
Recv: 1: Owl_pair.gcode_         .gcode
Recv: 2: parameter.gcode_        .gcode
Recv: 3: /Files_Engish_Mega S_    .gcode
[...]
Recv: TFT Serial Command: A13 <Special Menu>.gcode
[...]
Recv: TFT Serial Command: A26
Recv:  TFT Serial Debug: SelectedDirectory: <Special Menu>.gcode
Recv: TFT Serial Debug: Enter Special Menu
Recv: TFT Serial Command: A8 S0
[...]
Recv: TFT Serial Command: A13 <Preheat Ultrabase>.gcode
[...]
Recv: TFT Serial Command: A13 <Preheat Ultrabase>.gcode
Recv: TFT Serial Command: A8 S4
[...]
Recv: TFT Serial Command: A13 <Next Mesh Point>.gcode
[...]
Recv: TFT Serial Command: A8 S0
[...]
Recv: TFT Serial Command: A13 <Preheat Ultrabase>.gcode
Recv: TFT Serial Command: A26
Recv:  TFT Serial Debug: SelectedDirectory: <Preheat Ultrabase>.gcode
Recv: TFT Serial Debug: Enter Special Menu
Recv: Special Menu: Preheat Ultrabase
Recv: TFT Serial Command: A8 S0
Recv: ok
[...]
Recv: TFT Serial Command: A8 S4
[...]
Recv: TFT Serial Command: A13 <Next Mesh Point>.gcode
[... 279 lines cut ...]
Recv: ok
[...]
Recv: TFT Serial Command: A8 S4
Recv: TFT Serial Command: A8 S8
[...]
Recv: TFT Serial Command: A13 <Z Down 0.02>.gcode
[...]
Recv: TFT Serial Command: A26
Recv:  TFT Serial Debug: SelectedDirectory: <Z Down 0.02>.gcode
Recv: TFT Serial Debug: Enter Special Menu
Recv: Special Menu: Z Down 0.02
Recv: TFT Serial Command: A8 S0
Recv: ok
Recv: ok
Recv: ok
[...]

here are the photos:

[image]
[image]
[image]
[image]
[image]
[image]

ketchup67, 2020-07-07:
Also i choosed Z up but in the log it shows down

knutwurst, 2020-07-07:
Can you show me the main menu with the folders? Are they displayed? I really need to investigate, what causes the drawing-bug at the top. I feed the data the same way I did with the original touchscreen... so something has changed.

ketchup67, 2020-07-07:
Attached the photos and the logfile. Folders are shown but not accessable. 
Pressing a folder title results into going back to the first menu.

Recv: TFT Serial Debug: SD card removed... J01
[...]
Recv: echo:SD card ok
Recv: TFT Serial Debug: SD card inserted... J00
[...]
Recv: TFT Serial Command: A8 S0
Recv: 0: <Special Menu>.gcode
Recv: 1: Owl_pair.gcode_         .gcode
Recv: 2: parameter.gcode_        .gcode
Recv: 3: /Files_Engish_Mega S_    .gcode
[...]
Recv: TFT Serial Command: A8 S4
Recv: 4: /Files_Japanese_Mega S_  .gcode
Recv: 5: /Robot_                  .gcode
Recv: 6: /D__-__Mega S_           .gcode
[...]
Recv: TFT Serial Command: A13 /Files_Japanese_Mega S_  .gcode
[...]
Recv: TFT Serial Command: A26
Recv:  TFT Serial Debug: SelectedDirectory: Files_Japanese_Mega S_  .gcode
Recv: echo:Cannot enter subdir: Files_Japanese_Mega S_  .gcode
Recv: TFT Serial Command: A8 S0
Recv: 0: <Special Menu>.gcode
Recv: 1: Owl_pair.gcode_         .gcode
Recv: 2: parameter.gcode_        .gcode
Recv: 3: /Files_Engish_Mega S_    .gcode
[...]

[image]
[image]

ketchup67, 2020-07-07:
Here the log and photos with the original firmware. Unfortunately no debug enabled

Send: M20
Recv: echo:SD card ok
Recv: Begin file list
Recv: OWL_PA~1.GCO
Recv: PARAME~1.GCO
Recv: echo:Cannot open subdir
Recv: _OS~1/
Recv: echo:Cannot open subdir
Recv: /
Recv: echo:Cannot open subdir
Recv: /
Recv: /ROBOT/AI3M_R~1.GCO
Recv: /ROBOT/AI3M_R~2.GCO
Recv: /ROBOT/AI3M_R~3.GCO
Recv: /ROBOT/AI3M_R~4.GCO
Recv: /ROBOT/AI93B1~1.GCO
Recv: /ROBOT/AIAD07~1.GCO
Recv: /ROBOT/AI2C23~1.GCO
Recv: /ROBOT/AI5CC8~1.GCO
Recv: echo:Cannot open subdir
Recv: /
Recv: End file list
Recv: ok
Send: M105
Recv: ok T:25.3 /0.0 B:25.9 /0.0 @:0 B@:0
Send: M105
Recv: ok T:25.3 /0.0 B:25.9 /0.0 @:0 B@:0
Send: M105
Recv: ok T:25.2 /0.0 B:25.9 /0.0 @:0 B@:0
Send: M105
Recv: ok T:25.2 /0.0 B:25.9 /0.0 @:0 B@:0
Send: M105
Recv: ok T:25.2 /0.0 B:25.9 /0.0 @:0 B@:0
Send: M105
Recv: ok T:25.2 /0.0 B:25.9 /0.0 @:0 B@:0

[image]
[image]
[image]

ketchup67, 2020-07-08:
Hi, 
sorry for my question but do you maybe have any updates?

knutwurst, 2020-07-08:
I actually do have a normal job, so I can't fiddle around all the time with the firmware... but here you go. Try it ;)
Also check out the official facebook group: https://www.facebook.com/groups/3094090037303577/

[list_dir_and_special_menu_01.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/4890513/list_dir_and_special_menu_01.zip)

ketchup67, 2020-07-08:
I am very sorry about, it was really not my intention to push you.
I am not in facebook, so i do not have access to the link you offered right now.
Thank you really very much for your hard work in your free time for my issue i brought to you

knutwurst, 2020-07-08:
Hey.. all's fine ;)
Could you test the firmware i provided? *push

ketchup67, 2020-07-08:
Thx for pushing :-P (rofl)

Now things are getting more strange. 
The first pictures shows what is shown in the menu.
By pressing "parameter.gcode_" nothing happens. 

[image]

But if i leave the print menu and going back into there 
the second photo is showing what will be displayed then......

[image]

Attached the log:

Recv: TFT Serial Command: A26
Recv:  TFT Serial Debug: SelectedDirectory:
Recv: echo:SD card ok
Recv: TFT Serial Command: A8 S0
Recv: 0: <Special Menu>.gco
Recv: 1: Owl_pair.gcode_                     .gco
Recv: 2: parameter.gcode_                    .gco
Recv: 3: /Files_Engish_Mega S_                .gco
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.12 /0.00 B:27.67 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.19 /0.00 B:27.73 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.22 /0.00 B:27.73 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.22 /0.00 B:27.70 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.19 /0.00 B:27.73 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.12 /0.00 B:27.73 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.16 /0.00 B:27.73 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.16 /0.00 B:27.70 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.19 /0.00 B:27.70 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.16 /0.00 B:27.73 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.16 /0.00 B:27.70 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.22 /0.00 B:27.70 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.09 /0.00 B:27.73 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.19 /0.00 B:27.70 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.12 /0.00 B:27.73 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.16 /0.00 B:27.70 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.16 /0.00 B:27.73 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.22 /0.00 B:27.73 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:27.16 /0.00 B:27.70 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok

knutwurst, 2020-07-08:
That's expected :D 
I made a mistake and wrote to many bytes into a way too small array... so this is a classic buffer-overflow which crashes the display. But I'm on it... wait a sec.

knutwurst, 2020-07-08:
Now it should work.. it works on my display. It's ugly because of the ".gcode" in every single line, but it works.
Please try it and see if you can select directories and the special menu. 

[list_dir_and_special_menu_02.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/4891955/list_dir_and_special_menu_02.zip)

ketchup67, 2020-07-08:
Ok, that´s what i am getting here. Pls also take a look into the log. I always choosed "Z up" but it is always showing "Z down" 
in the log

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Recv: TFT Serial Command: A13 <Start Mesh Leveling>.gcode
Recv:  T:25.56 /0.00 B:26.11 /0.00 @:0 B@:0
Recv: TFT Serial Command: A26
Recv:  TFT Serial Debug: SelectedDirectory: <Start Mesh Leveling>.gcode
Recv: TFT Serial Debug: Enter Special Menu
Recv:  DEBUG: Special Menu Selection: <Start Mesh Leveling>.gcode
Recv: Special Menu: Start Mesh Leveling
Recv: TFT Serial Command: A8 S0
Recv: ok
Send: M27
Recv:  T:25.53 /0.00 B:26.08 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:25.56 /0.00 B:25.97 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:25.56 /0.00 B:26.16 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:25.56 /0.00 B:26.16 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
Recv: ok
Send: M27
Recv:  T:25.56 /0.00 B:26.11 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:25.62 /0.00 B:26.11 /0.00 @:0 B@:0
Recv: MBL G29 point 1 of 25
Recv: X:10.00 Y:10.00 Z:0.00 E:0.00 Count X:800 Y:800 Z:0
Recv: ok
Recv: Not SD printing
Send: M27
Recv: ok
Recv: Not SD printing
Recv: ok
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv: TFT Serial Command: A8 S4
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:25.69 /0.00 B:26.16 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:25.59 /0.00 B:26.14 /0.00 @:0 B@:0
Recv: TFT Serial Command: A13 <Z Up 0.1>.gcode
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:25.56 /0.00 B:26.14 /0.00 @:0 B@:0
Recv: TFT Serial Command: A26
Recv:  TFT Serial Debug: SelectedDirectory: <Z Up 0.1>.gcode
Recv: TFT Serial Debug: Enter Special Menu
Recv:  DEBUG: Special Menu Selection: <Z Up 0.1>.gcode
Recv: Special Menu: Z Up 0.1
Recv: TFT Serial Command: A8 S0
Recv: ok
Recv: ok
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv: TFT Serial Command: A8 S4
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:25.59 /0.00 B:26.08 /0.00 @:0 B@:0
Recv: TFT Serial Command: A8 S8
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv:  T:25.53 /0.00 B:26.08 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: ok
Recv: TFT Serial Command: A13 <Z Down 0.02>.gcode
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:25.59 /0.00 B:26.08 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Recv: TFT Serial Command: A26
Recv:  TFT Serial Debug: SelectedDirectory: <Z Down 0.02>.gcode
Recv: TFT Serial Debug: Enter Special Menu
Recv:  DEBUG: Special Menu Selection: <Z Down 0.02>.gcode
Recv: Special Menu: Z Down 0.02
Recv: ok
Recv: TFT Serial Command: A8 S0
Recv: ok
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv:  T:25.59 /0.00 B:26.14 /0.00 @:0 B@:0
Recv: TFT Serial Command: A8 S4
Send: M27
Recv: Not SD printing
Recv: ok
Recv: TFT Serial Command: A8 S8
Send: M27
Recv: Not SD printing
Recv: ok
Recv: TFT Serial Command: A13 <Z Down 0.02>.gcode
Recv:  T:25.66 /0.00 B:26.11 /0.00 @:0 B@:0
Send: M27
Recv: Not SD printing
Recv: ok
Send: M27
Recv: Not SD printing
Recv: ok
Recv: TFT Serial Command: A26
Recv:  TFT Serial Debug: SelectedDirectory: <Z Down 0.02>.gcode
Recv: TFT Serial Debug: Enter Special Menu
Recv:  DEBUG: Special Menu Selection: <Z Down 0.02>.gcode
Recv: Special Menu: Z Down 0.02
Recv: ok
Recv: TFT Serial Command: A8 S0

ketchup67, 2020-07-08:
It´s funny. 2 weeks i am the owner of this printer now, but i only printed 1 time and it ended up with that the work piece detached from the bed. About i was not there when it happened, it was a mess. 
The lower holding plate is not straight and flat. I am not able to get a really good leveling here.
Maybe i just should send the printer back....

akula3, 2020-07-08:
Don't worry, ketchup67 !
Just open an issue at Anycubic directly. They will manage it.
On this way they send me new Hotbed, Mainboard and Hotend Fans.

ketchup67, 2020-07-08:
Really? I am impressed, because i bought it from the Anycubic ebay shop and i didn´t mention hat i could highlight it there directly. Nevertheless, my hope was to solve it with this firmware without more trouble.

akula3, 2020-07-08:
I bought my Mega S from Aliexpress. They redirected me to Anycubic.

ketchup67, 2020-07-08:
May i ask you, when did you bought your Mega S? Is there also this new display built in?

akula3, 2020-07-08:
No, it's the older one. I bought my printer before Christmas '19.
But, I don't like the stupid original TFT from Anycubic and replaced it by RepRapDiscount Full Graphic Smart Controller. It is more informative and offer live changes in the print process.

ketchup67, 2020-07-08:
That´s great ! But for me the standard one would be really fine...if the print print problem would be solved (!)
I did read somethíng about the BL-Touch modification. A little complicated and also i guess the warranty will be gone 
with and especially about this modification. So i will not touch it now ;)

knutwurst, 2020-07-09:
Next one... without directories and with another serial protocol handling.

[list_dir_and_special_menu_03.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/4895484/list_dir_and_special_menu_03.zip)

@akula3 , aren't you the one from thingiverse with the TMC2209 and UART? 👯‍♂️ 
I didn't forget you but I still did not get any suitable Trinamic drivers to test my setup but I am confident that UART with RX+TX will work on the Trigorilla board.

akula3, 2020-07-09:
Yes Oliver, i'm and we had the discussion on thingiverse you mentione.
I belive too, that it should work, but I'm not sure which two pins on trigorilla are interrupt capable.
However, this is not the current topic. If you want, I can open here in your repo an extra theme for UART.

ketchup67, 2020-07-09:
Here we go again:
After leaving the Print menu and going back inside to there the basic gcodes are not shown anymore. 
Moving directly the special menu - "Z - up" ist still "Z - down"

[image]
[image]
[image]
[image]
[image]
[image]
[image]

The log:

ketchup67, 2020-07-09:
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: ok
Send: M20
Recv: Begin file list
Recv: OWL_PA~1.GCO 6870525
Recv: PARAME~1.GCO 13691
Recv: /ROBOT/AI3M_R~1.GCO 29980794
Recv: /ROBOT/AI3M_R~2.GCO 15165665
Recv: /ROBOT/AI3M_R~3.GCO 8726931
Recv: /ROBOT/AI3M_R~4.GCO 11986786
Recv: /ROBOT/AI93B1~1.GCO 15082478
Recv: /ROBOT/AIAD07~1.GCO 24305812
Recv: /ROBOT/AI2C23~1.GCO 7064063
Recv: /ROBOT/AI5CC8~1.GCO 11031816
Recv: End file list
Recv: ok
Send: M420V1
Recv: echo:Invalid mesh.
Recv: echo:Bed Leveling OFF
Recv: echo:Fade Height OFF
Recv: ok
Send: M851
Recv: echo:Unknown command: "M851"
Recv: ok
Send: M155 S2
Recv: ok
Recv:  T:28.97 /0.00 B:29.49 /0.00 @:0 B@:0
Recv:  T:28.97 /0.00 B:29.46 /0.00 @:0 B@:0
Recv:  T:28.97 /0.00 B:29.40 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.49 /0.00 @:0 B@:0
Recv: TFT Serial Command: A8 S0
Recv: 0: <Special Menu>.gcode
Recv: 1: Owl_pair.gcode
Recv: 2: parameter.gcode
Recv:  T:28.97 /0.00 B:29.46 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.37 /0.00 @:0 B@:0
Recv:  T:28.97 /0.00 B:29.52 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.43 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.49 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.49 /0.00 @:0 B@:0
Recv:  T:28.94 /0.00 B:29.43 /0.00 @:0 B@:0
Recv: TFT Serial Command: A13 <Special Menu>.gcode
Recv:  T:29.00 /0.00 B:29.43 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.49 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.43 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.43 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.43 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.49 /0.00 @:0 B@:0
Recv: TFT Serial Command: A26
Recv:  TFT Serial Debug: SelectedDirectory: <Special Menu>.gcode
Recv: TFT Serial Debug: Enter Special Menu
Recv:  DEBUG: Special Menu Selection: <Special Menu>.gcode
Recv: TFT Serial Command: A8 S0
Recv:  T:29.00 /0.00 B:29.43 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.46 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.49 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.49 /0.00 @:0 B@:0
Recv: TFT Serial Command: A13 <Exit>.gcode
Recv:  T:29.00 /0.00 B:29.46 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.49 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.52 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.55 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.52 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.55 /0.00 @:0 B@:0
Recv: TFT Serial Command: A8 S0
Recv:  T:29.00 /0.00 B:29.49 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.43 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.40 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.52 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.55 /0.00 @:0 B@:0
Recv: TFT Serial Command: A8 S4
Recv:  T:29.00 /0.00 B:29.55 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.55 /0.00 @:0 B@:0
Recv: TFT Serial Command: A13 <Start Mesh Leveling>.gcode
Recv:  T:29.00 /0.00 B:29.55 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.55 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.52 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.49 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.52 /0.00 @:0 B@:0
Recv: TFT Serial Command: A26
Recv:  TFT Serial Debug: SelectedDirectory: <Start Mesh Leveling>.gcode
Recv: TFT Serial Debug: Enter Special Menu
Recv:  DEBUG: Special Menu Selection: <Start Mesh Leveling>.gcode
Recv: Special Menu: Start Mesh Leveling
Recv: TFT Serial Command: A8 S0
Recv: ok
Recv:  T:29.00 /0.00 B:29.52 /0.00 @:0 B@:0
Recv:  T:29.00 /0.00 B:29.49 /0.00 @:0 B@:0
Recv:  T:28.97 /0.00 B:29.49 /0.00 @:0 B@:0
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Send: M113 S2
Recv:  T:29.03 /0.00 B:29.49 /0.00 @:0 B@:0
Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
Recv: ok
Recv:  T:29.00 /0.00 B:29.49 /0.00 @:0 B@:0
Recv: MBL G29 point 1 of 25
Recv: X:10.00 Y:10.00 Z:0.00 E:0.00 Count X:800 Y:800 Z:0
Recv: ok
Connection closed, closing down monitor
Changing monitoring state from "Operational" to "Offline"

knutwurst, 2020-07-10:
> However, this is not the current topic. If you want, I can open here in your repo an extra theme for UART.
Not necessary ;) I'll create a development branch for this when I have the hardware. I'll also mention you so you can see the progress.

> Moving directly the special menu - "Z - up" ist still "Z - down"
This is because the menu items are not in the correct place.  This is the main issue, I am trying to fix, but it is also a bit tricky to do without a display which I can debug in realtime. ... I really need one. The behaviour is completely different from the "old" display, which works just fine with this dev branch. (the .gcode at every line still looks ulgly, but one issue at the time^^)

[45 more replies]

closed 2020-07-27 by knutwurst