knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #5: Mesh Leveling im GCode notwending?

Fehldruck, 2020-07-07:
Hey. Du schreibst ja dass das Mesh nach einem neustart gerestored wird. Miss ich da noch die beiden Commands im gcode nach G28 einfügen oder sogar g28 und die beiden meshbefehle weglassen? Sorry für die doofe frage. Bin neu im 3D Druck. Und muss ich bevor ich ein neues mesh mache ein m502 durchführen?

LG

knutwurst, 2020-07-07:
Moin :)
Ich sag's nur ungern, aber das hier ist ein Bugtracker. Ein "Issue" ist ein Fehler, der in der Firmware vorhanden ist, und den ich hiermit tracken kann um ihn zu beheben. Allgemeine Fragen kannst du gern in der offiziellen Facebook-Gruppe stellen: https://www.facebook.com/groups/3094090037303577/

Aber... wir wollen ja mal nicht so sein :D 
Da es sich hierbei um eine (angepasste/umgebaute) Marlin-Firmware handelt, gilt auch weiterhin die Dokumentation für die GCODEs. Beispielsweise der G28 ist das "homing", also auf Nullposition fahren. Den solltest du auf jeden Fall im Startcode deines Slicers haben! -> Siehe: https://marlinfw.org/docs/gcode/G028.html

M502 (Werkseinstellungen) brauchst du nie machen. Höchstens nachdem zwischen Firmwares wechselst. Aber selbst das Initialisieren geschieht automatisch. Das ist übrigens das Selbe, als wenn du im Menü auf "Load FW Defaults" gehst.

Nachdem du dein Mesh fertig hast, solltest du aber einen M500 (Save EEPROM) machen. Kannst du auch über das Menü. Damit wird alles gespeichert.

Im Normalfall wird nach einem G28 aber das Mesh-Leveling abgeschaltet und man muss es wieder mit M420 aktivieren. Dies passiert automatisch. Im Startcode musst du aber "M420 S1" einfügen, vorzugsweise nach einem G28. Dann ist das Mesh leveling auf jeden Fall an! Und selbst wenn der Drucker kein Mesh gespeichert hat, so wird der Befehl einfach ignoriert. Es ist also immer ratsam M420 S1 mit im Startcode zu haben.

Der Einfachheit halber, hänge ich dir mal meine Start- und Endcodes an. Die kannst du dann anpassen so wie du magst:

Startcode:
```
G21                                 ; metric values
G90                                 ; absolute positioning
M420 S1                             ; Enable leveling
M82                                 ; set extruder to absolute mode
M107                                ; start with the fan off
M300 S1000 P500                     ; BEEP heating done
G28 X0 Y10 Z0                       ; move X/Y to min endstops
G0 Z0.15                            ; lift nozzle a bit
G0 Y15                              ; go away from clips
G92 E0                              ; zero the extruded length
G1 X50 E25 F500                     ; Extrude 25mm of filament in a 5cm line.
G92 E0                              ; zero the extruded length again
G1 E-2 F500                         ; Retract a little
G1 X120 F4000                       ; Quickly wipe away from the filament line
```

Endcode:
```
M104 S0          ; Extruder off
M140 S0          ; Heatbed off
M107             ; Fan off
G91              ; relative positioning
G1 E-5 F300      ; retract a little
G1 Z+10 E-5 ; X-20 Y-20 F{travel_xy_speed} ; lift print head
G28 X0 ;Y0       ; homing
G1 Y180 F2000    ; reset feedrate
M84              ; disable stepper motors
G90              ; absolute positioning
M300 S440 P200   ; Make Print Completed Tones
M300 S660 P250   ; beep
M300 S880 P300   ; beep
```

Fehldruck, 2020-07-07:
Vielen Dank für die schnelle Hilfe. Ich kopier mir morgen den gcode raus. Und der Facebookgruppe hab ich schon eine Anfrage geschickt. Also miss ich nicht vor dem M420 noch das M501 setzen was in anderen Anleitungen steht?

knutwurst, 2020-07-07:
Du bist auch angenommen ;) 

Nein... das steht vielleicht in anderen Anleitungen, damit man "Müll" loswird. Aber man muss das nicht tun. Mit einem M501 ist ja alles weg, was man gespeichert hat. Ok, der i3 Mega kann nicht so viel, aber wenn ich bedenke, ich müsste alles beim Kossel zurücksetzen, würde ich schreien :D

EDIT: Ich hab im Code oben das angepasst. Aber wie gesagt.. in meiner Firmware wird das aktivierte Mesh level nach dem G28 sowieso wiederhergestellt. Insofern ist es egal. Du musst es aber irgendwo im Startcode haben.

Fehldruck, 2020-07-07:
Dankeschön für deine Hilfe :) Wieder was gelernt :)

closed 2020-07-07 by knutwurst