knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #6: Endstop beeps

Dalton0190, 2020-07-08:
since you have activated the endstop beep in release 1.0.5 it starts beeping as soon as Z and X axis are moving.
i can reproduce that when i manually move the axis via touchscreen or in octopi, but only when both axis (Z and X) have been moved.
same test does not beep when i only move Z and Y or when i move X and Y
beeping sounds then all time long until i disable the motors in the setup menu
all axis are physically far away from the endstops, so it must come from some firmware stuff
i have tried the original anycubic firmware, works fine (except axis directions because of the TMC2208)
Video of the issue:   https://youtu.be/o64bxG3nOZ4

knutwurst, 2020-07-08:
Can you do me a favour and do a G28 homing before moving the axis? Does the issue persist? I can no reproduce the issue, so it might be that it was always there, but you didn't hear it.

I can deactivate the endstop beeps, but the problem itself persists on your particular printer. There is something wrong with your endstops.

Dalton0190, 2020-07-08:
in homeposition it stops beeping
it must be something related to the left Z endstop, when i press it manually then it stops beeping regardless of the other endstops

but when i install the original firmware 1.1.5 from anycubic it works normal

knutwurst, 2020-07-08:
The original firmware disables the endstop beeps or doesn't let you move the axis before homing. Something like that...

Dalton0190, 2020-07-08:
in the original firmware the endstop beeps are activated and i can move before homing
with the custom firmware it starts beeping also after homing when the Z endstop gets free again

knutwurst, 2020-07-08:
That's strange. But disabling the endstop beep does not solve the problem. I'll take a look into it!

Dalton0190, 2020-07-09:
i made a custom build with disabled endstop beep and everything seems to work fine, i dont know where this is coming from...

knutwurst, 2020-07-09:
So now you can't hear the endstop beep anymore, but the endstop is still triggered. Good for you that Marlin ignores the endstops, when not printing.

For me it looks like a hardware issue.

Did you use the precompiled 1.0.5 Version or did you build by yourself? The current codebase differs a bit from the latest release version.

Dalton0190, 2020-07-09:
i used your master branch to build

knutwurst, 2020-07-10:
Can you please use the precompiled .hex 1.0.5 to see if the behaviour differs?

Dalton0190, 2020-07-11:
> Can you please use the precompiled .hex 1.0.5 to see if the behaviour differs?

was the same problem with the precompiled 1.0.5
the strange thing is that the uncontrolled beeping stops immediately when i disable the motors while it still beeps when i trigger the endstops by hand

knutwurst, 2020-07-11:
That's really strange and since you are the only one I know, who has that problem, I assume it's a hardware issue :/ 
As you might already have seen: There is a new version 1.0.6 which fixes some issues and also disables the beeps again.

I also close this topic. Maybe you can find out, what exactly causes the endstop beep on your hardware and provide some information which I might implement in the future ;)

closed 2020-07-11 by knutwurst