knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #7: Mesh Level Step 0.01

icebear79, 2020-07-27:
The Step 0.01 (up or down, I didnt remember now) does a movement way more then 0.01 down to the Bed!

knutwurst, 2020-07-27:
When pressing +0.01 the movement should be +0.03 and -0.02 afterwards to ensure a motor movement. This seems not to work correctly and I reverted it. There are some printers, which don't move at all, when pressing on +/- 0.01 so the feature itself needs to be revised.

In the meantime... can you live with the 0.02 steps? You can also compile the version I pushed a few minutes ago by yourself if you really need the feature right now.

icebear79, 2020-07-27:
Yes, it is okay for me to live with the 0.02 and the 0.1 Steps. 

I think only 0.1 wouldn't be fine enough. 0.02 does it well; so, I think there is no need to implement the 0.01 Steps. Kick 'em out ;-).

Regards
Ice

closed 2020-07-27 by knutwurst