knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #8: Display freezes

Sydney1101, 2020-07-31:
Hi. I modified your version for my Mega X and it works fine. I have only one issue with your version - the function of the touch display. That happens mostly, when I use the filament in function. After that I can change the menu pages, but the printer will not accept any touch action (i.e. preheat function). Do you have any idea to fix or localize the problem?

Best Greetings
Andreas

knutwurst, 2020-08-01:
Hello :)
The official Mega X support should come within the next week. Did you only edit the print volume size or did you change something else?

And that do you mean by "filament in function"? The one integrated in the original display code or the filament change function in the special menu?

Can you make a detailed plan how to reproduce the issue?

Thanks in advance!

Sydney1101, 2020-08-01:
Hi. It is a little bit more than modify the volume size ;-) After some time, when I print with OctoPrint, I want to transfer filament through the nozzle. So I use on the menu Tools - Infill - Filament in. It works mostly one times. If it occure, the temperature goes down and I have to restart the printer. But I don't know what have to be happen bevor that problem occurs... :(

knutwurst, 2020-08-01:
Ok?! I'll take a look at the original X firmware, but It can't be much more than the print volume size, since the printers are almost identical. When the temperature goes down, the board may had a reset or the firmware froze. Maybe because the TFT didn't respond.

Which commands are you sending via octoprint when using "infill"?
Can you do a test with an SD card print and pronterface to track the errors?

Sydney1101, 2020-08-01:
Yes, I can do it. If you want to create a Mega X version, I can share my modifications... ;) So, I can use your version without anything do change *g*. BTH I have changed the junction deviation to 0.1 otherwise my edges are too round...

knutwurst, 2020-08-02:
`Which commands are you sending via octoprint when using "infill"?
Can you do a test with an SD card print and pronterface to track the errors?`

Could you please answer the questions? That would be really helpful :)

What I need is:
- A complete precedure to reproduce the bug, including the gcodes you sent.
- A complete Log of the console output of your printer especially he complete error message.

Also, yes, please share your modifications if you like :)

Sydney1101, 2020-08-16:
Sorry for the long time.... I was very busy and I find the reason of my "touch" problem. It was the temperature sensor on the hotend which sometimes measures wrong values. I will look at your mega x version, If I found some points of my version - I will sent you my values.

knutwurst, 2020-08-17:
Good morning :D 

Unfortunately I don't own a Mega X, so I haven't been able to get a good value for the junction deviation so far. Good to know 0.2 works for you. I calculated 1.6, which should be exactly right with a standard acceleration of 400 mm/s and 4 mm jerk. Since I just couldn't test it, I switched to classic jerk (only for the Mega X).

Which compile errors do you get?

I suspect that you did not set the board in configuration.h. Please also have a look at line 46 in configuration.h. Maybe something will clear up then;)

Since the Mega X and BLTouch support, I had to compile for 24 binary files, which was no longer possible without scripting and PlatformIO.

Sydney1101, 2020-08-17:
Yes, the board type was missing 👍 I played with the Linear Advance and it produces quit good results by speeds up to 60mm/s. Only the sound of the extruder is very special...

Sydney1101, 2020-08-24:
No 👍

closed 2020-08-24 by Sydney1101