knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #9: BLTouch Probing Error

Sceleratis, 2020-08-12:
I just got my BLTouch all installed and wired up. It *seems* like it's working. Entering commands in OctoPrint's terminal seems to make it do stuff, and M119 seems to correctly return whether or not the sensor is triggered. For some reason actually probing the bed just fails but I don't know why. Anytime I try to probe the bed, whether from the special menu or from OctoPrint, it homes, releases the probe, starts moving down, the probe hits the bed, the probe is retracted and the bltouch turns red and the printer just stops. In OctoPrint, the error "Probing failed" is thrown and the printer is disconnected.

I'm using an AnyCubic i3 Mega-S (with the DGUS display) and a bltouch I bought through Amazon.
Any thoughts? Is there something I'm missing here or could this be something with the firmware? Also is there anyway for us to set the bltouch z offset from the nozzle through the special menu (for when this does eventually hopefully work?)

knutwurst, 2020-08-12:
If everything appears correct and the automatic leveling starts, but stops at the second measuring point and the error message "Probing Failed" appears in Pronterface, this is an indication that the Z-offset was not set correctly and the sensor suddenly reaches a point which is below the height of the nozzle.

I also added this common issue to the "Troubleshooting" section.

Let me know if it fixes your issue. ... Also make sure, all settings (especially the offset) is saved. Doublecheck with M503.

Sceleratis, 2020-08-12:
Also just noticed it's returning triggered when the probe is closed (default state) is this normal?

I should also mention with the current release, with the DGUS display the special menu is no longer working correctly. Trying to navigate around it leads to it showing no options, but then tapping any of the blank spaces shows Exit, Prehead Ultrabase, Fil Change Pause, and Fil Change Resume. Trying to select any of these options, or trying to move to the next page, will again just cause all of the menu options to disappear.

knutwurst, 2020-08-12:
> Also just noticed it's returning triggered when the probe is closed (default state) is this normal?
Yes. The default state is "up" with red light.

> with the DGUS display the special menu is no longer working correctly
Oh! That's not so good :o
When you go back to an older release, does it work correctly?

Sceleratis, 2020-08-12:
It was working correctly on the previous release. Only started happening after installing the newest one.

knutwurst, 2020-08-12:
That's weird, because I didn't change any display code... can you please confirm that 1.1.0 is working?

Sceleratis, 2020-08-12:
Rolled back to 1.1.0, menu seems to be working fine.

Sceleratis, 2020-08-12:
Scratch that, entering the "tune hotend" submenu just produced the same result as before.

knutwurst, 2020-08-12:
And the error wasn't there before you installed the BLTouch Sensor?

Sceleratis, 2020-08-12:
> And the error wasn't there before you installed the BLTouch Sensor?

Not sure. Going to disconnect the sensor and test now

Sceleratis, 2020-08-12:
Still does it with the sensor disconnect

knutwurst, 2020-08-12:
And the Special Menu was working before? When did the error occur at first?
Seems strange that you are the first one with this kind of bug :/

Sceleratis, 2020-08-12:
I never had a reason to use the special menu as I control everything from octoprint so I'm not sure. I only noticed it now while trying to figure out the bltouch stuff.

pahowart, 2020-08-12:
I just completed a BLTOUCH upgrade on a MEGA S using the previous knutwurst release. The BLTouch was a V2.1 Clone. I had to reverse the two wire signal and ground leads to get it to probe correctly. It was definitely correct all of the way through as I confirmed with multimeter, reversed on the BLtouch board maybe. Worked great after that. Unfortunately I did this for a friend and don't have the unit to confirm issues with latest release.

Sceleratis, 2020-08-12:
Well, the good news is after going through the BLTouch configuration again I now finally have it working the way it should, so I guess it was just an issue with the Z offset that caused my original problem (as you said.)

Bad news is I guess I found a new menu bug that for some reason only affects me so far (?)

knutwurst, 2020-08-12:
Ok.. since the original issue is somehow "fixed", I'll close this issue. @Sceleratis could you open a new bugticket with a detailed way to reproduce the bug and also some photos or a video to show the issue?

I'll investigate the issue and ask other ppl on facebook which are using the 1.1.0 or 1.1.1 right now.

Thanks in advance!

closed 2020-08-12 by knutwurst