knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #11: FW version still shows 1.1.0 after 1.1.1 update

Sc0th, 2020-08-13:
Recv: FIRMWARE_NAME:Marlin 2.0.5.4 (knutwurst, Github) MARLIN-AI3M_VERSION:1.1.0 SOURCE_CODE_URL:https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S PROTOCOL_VERSION:1.0 MACHINE_TYPE:Knutwurst's Anycubic i3 MEGA EXTRUDER_COUNT:1 UUID:b6153152-86fb-4cd8-bda7-91c6a04c467d

knutwurst, 2020-08-13:
No? I mean ... I can not reproduce the issue:

`>>> m115
SENDING:M115
FIRMWARE_NAME:Marlin 2.0.5.4 (knutwurst, Github) MARLIN-AI3M_VERSION:1.1.1 SOURCE_CODE_URL:https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S PROTOCOL_VERSION:1.0 MACHINE_TYPE:Knutwurst's Anycubic i3 MEGA EXTRUDER_COUNT:1 UUID:b6153152-86fb-4cd8-bda7-91c6a04c467d`

On 2020-08-10 I set the Version to 1.1.1, just before the new Release Tag.

See: 
[image]

Sc0th, 2020-08-13:
How peculiar - perhaps specific to - i3_MEGA_S_TMC_v1.1.1.hex ?

The firmware update completes ok (via Octoprint - avrdude) and I get all the right beeps in all the right places, but alas seems I am still running 1.1.0

Wonder where I am going wrong, this is not a new procedure to me :(

knutwurst, 2020-08-13:
Huh! You are right!
Going to fix it now... thank you!

Sc0th, 2020-08-13:
glad to know I am not losing my mind after all! thank you!

knutwurst, 2020-08-13:
No, it's just that the i3_MEGA_S_TMC_v1.1.1.hex is almost binary compatible to 1.1.0 because all new features were regarding BLTouch and the Mega X support. Your firmware is not affected at all. The bugfixes I listed came in between 1.1.0 and 1.1.1 and were never published in a release version.

So.. it's fine to use 1.1.0 or the one in the 1.1.1 package.

Anyway I wanted to release 1.1.2 to fix some minor issues and also allow way faster homing and bed probing. Also you can see the distribution date with M115 to be absolutely safe ;)

Sc0th, 2020-08-13:
Got it! thank you again, your work is very much appreciated.

knutwurst, 2020-08-13:
Here you are... have fun ;)
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/tag/1.1.2

closed 2020-08-13 by knutwurst