knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #13: Z-Achse zu schnell

merk-1337, 2020-08-23:
Die Z-Achse bei der Datei Mega-X TMC ist zu schnell konfiguriert, wodurch ein unschönes quietschen beim verfahren ausgelöst wird.
Problem tritt bei mehreren Anwendern auf.

knutwurst, 2020-08-23:
Bei welcher Version? Aktuelle 1.1.3 oder selbst gebaut aus dem Master Branch?

merk-1337, 2020-08-23:
Mit der aktuellen Version.
Kann momentan nicht testen der Drucker läuft noch bis morgen. 
Aber es gibt drei Geschwindigkeiten um manuel zu fahren. 
Bei low ist das Geräusch komplett weg bei Medium noch leicht vorhanden. 

Ich denke dazwischen sollte der perfekte Wert liegen.

knutwurst, 2020-08-23:
Bitte verbinde dich einmal über USB mit Pronterface und dem Drucker und versuche die Werte zu testen. Das würde mir wirklich helfen. Ein "dazwischen sollte der perfekte Wert liegen" ist schwer in Zahlen zu fassen ;) 

Danke :)

merk-1337, 2020-08-23:
Alles klar werde ich machen und, dir anschließend Bescheid geben.

merk-1337, 2020-08-23:
Also getestet zwischen 100 bis 1000 mm/min
Am besten bis max. 155 mm/min

knutwurst, 2020-08-23:
Was?! 
Ähm... Z läuft momentan mit 20m m/s. Du musst auf jeden Fall darunter bleiben. Wo kommen die 100-1000 her?

Tom-Moveman, 2020-08-23:
Knutwurst, rechne mal um 600 mm/min sind 10mm/s somit sind dein 20 mm/s 1200 mm/min. Aber das weißt du sicherlich ;) 20 mm/s auf Z sind schon sehr sportlich

knutwurst, 2020-08-23:
@Tom-Moveman Ah! mm/**min**! Das hab ich überlesen :D 
Und die 20mm/s habe ich mir ja nicht ausgedacht. Die habe ich aus der Original-Firmware: https://github.com/ANYCUBIC-3D/MEGA_X_V1.2.6/blob/38c6846c4c4300a0f68cb3c48e64e245db3c6f77/Configuration.h#L508

Fand die Werte auch sehr merkwürdig, da der Mega S mit 6mm/s auf Z ausgeliefert wird.

Tom-Moveman, 2020-08-23:
Die Werte sind durchaus merkwürdig, anscheinend treuen die Herren und Damen ihrem Drucker viel zu. Ich denke 10mm/s wäre ein guter Wert, aber nur meine Meinung

merk-1337, 2020-08-23:
Ich würde sogar auf den selben Wert wie beim Mega S gehen. 
Dann wären wir auf der sicheren Seite.
Ich denke auch wer Drucken will muss Zeit haben :D

knutwurst, 2020-08-23:
Der Mega X verwendet andere Über-/Untersetzungen. Ich kann nicht einfach 1:1 die Mega S Werte nehmen. 

Was ich jetzt brauche ich jemand, der es live an seinem Drucker ausprobiert. Ich hab keinen Mega X. Ich kann nichts tun. Ich programmiere auch blind... wie schon damals mit dem neuen Display. Das ist alles etwas schwierig.

merk-1337, 2020-08-23:
Ja ich kann es testen. Wäre kein Problem

knutwurst, 2020-08-23:
@merk-1337 , danke :)

Ich brauche die maximale Feedrate für Z und E0.

merk-1337, 2020-08-23:
Für Z wären wir dann bei 10mm /s
E0 würde ich nichts ändern, passt so wie bei der aktuellsten Version.

knutwurst, 2020-08-23:
Danke! <3
Bist du dir beim Extruder sicher?

Ich setze dann mal 10 oder 8mm/s für Z. Solange es leise ist passt das.:)

kommt in die Version 1.1.4 rein. Außerdem noch diverse andere Fixes (siehst du in den Commits der letzten Tage)

merk-1337, 2020-08-23:
Beim Extruder nicht sicher, habe aber die Firmware seit ner Woche und hatte mit der Extrudergeschwindigkeit keine Probleme. Auch die Kalibrator war perfekt.
Hab jetzt auch nochmal Z mit 8mm/s laufen lassen und würde sogar sagen besser wie 10.
Perfekt, das Geräusch macht mich schon Wahnsinnig.

merk-1337, 2020-08-23:
Solltest du weitere Hilfe beim testen rund um den Mega X haben, melde dich.

Tolle Arbeit sollte unterstützt werden 🥇

knutwurst, 2020-08-23:
Sehr schön! Vielen Dank für die Hilfe! 
Morgen im Laufe des Tages kommt die 1.1.4 mit einem Haufen Bugfixes und Verbesserungen (z.B. nicht mehr das .gcode im Special Menu etc).

Änderung der Feedrate ist hier drin: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/commit/e53792cbcebbaeab05d8ea54036b3ce56ce0551f

Ich schließe dann hier mal :)
Schönen Abend noch!

closed 2020-08-23 by knutwurst