knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #14: Filament sensor false positives

NiTeDK, 2020-08-23:
Hi there

Thanks for sharing a good firmware :-)

I seem to have a problem with the filament sensor that I didn't have with the stock firmware on my Anycubic i3 Mega S. Sometimes during a print, it'll halt and say that it has run out of filament even though it hasn't. If I wait till the printhead has moved aside, I can just press continue, and it'll resume printing. I've experienced this over several days with several different models and several different filaments. And several times during the same print - although it doesn't happen every time I print something.

I thought that I read somewhere that the "sensitivity" of the sensor could be adjusted in the code. That I could tell it to check several times over a longer period of time to avoid these false positives. But alas, I haven't been able to find it, so now I'm writing this instead =)

I hope you can help.

Best regards,
Claus

knutwurst, 2020-08-24:
Hello Claus,
With the Anycubic printer, the filament runout routine is controlled by the display and processed in the firmware. This only triggers after 3 seconds. This means that your filament sensor reports "no filament" for more than 3 seconds. If your sensor triggers now while printing, it is defective. However, you can (temporarily) deactivate it in the Special Menu.

If I make the filament runout test even longer, the extruder pulls it in too far and you can no longer change it.

You can permanently deactivate the sensor by commenting _#define ANYCUBIC_FILAMENT_RUNOUT_SENSOR_ out in configuration.h.

In anycubic_touchscreen.cpp line 971 you can edit the routine as you wish ;) 

Best wishes
Olli

NiTeDK, 2020-08-24:
Hey Olli

Thank you for a fast and detailed reply!

You're probably right about the 3 seconds but I don't understand it since I just measured about 51 seconds from the sensor exit to the extruder entrance with a random but fast print. Anyway, I think I'll just try messing around with the 3 seconds to learn something about making 3d printer firmware - and I think I'll try making it ask several times too for a better chance of catching a false positive.

Thanks again :-)

Best regards,
Claus

knutwurst, 2020-08-27:
Your're welcome :)

closed 2020-08-27 by knutwurst