knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #17: 20 sec Verzögerung & Fipsen

maansch, 2020-08-28:
Hallo,
Hardware: mega S aus 2019, TMC 2008 von Watterott

hab heute die „i3_MEGA_S_v1.1.5.hex“ installiert.
Hab zwar TMC drin, die Stecker sind aber gedreht.

Mir ist folgendes aufgefallen.
Wenn du Pause während des Druckes wählst, wird kurz (1 Sekunde) der Confirm Screen angezeit. (Vermute ich mal er ist nicht zu lesen so schnell ist der weg.)
Der Drucker stoppt dann nach 20 sec,  beim Resume dauert es auch 20 sec bis er wieder mit dem druck beginnt.

Wenn du den Druck abbrechen willst (STOP) dauert es ebenfalls 20sec bis der Drucker reagiert.

Eine anderer Punkt ist, so bald der Druck beginnt ertönt ein hoch hochfrequentes Fipsen das mit Druck stopp verschwindet. Hatte ich mit der original ANY FW nicht.

knutwurst, 2020-08-31:
Hallo,
ja, das ist vollkommen normal und ein (gewolltes?) Verhalten von Marlin. Das macht jeder Drucker, der eine Start/Pause/Stop Routine hat.

20 Sekunden sind vielleicht etwas viel, aber es kommt darauf an, wieviele Daten noch im Puffer sind und welche Stecken abgefahren werden. Bei kleineren Objekten mit kürzeren Wegen, dürfte die Zeitspanne kürzer sein.

Man kann die Pause-Funktion auf zwei Arten implementieren:
1. "Simple Pause" -> Hartes Unterbrechen genau an der Stelle, an welcher sich die Düse gerade befindet. Der Vorteil ist, dass der Befehl sofort ausgeführt wird, der Nachteil allerdings, dass die Funktion den Druck quasi zerstört. Die Temperatur wird nicht runtergefahren, die Düse bewegt sich nicht von Druckobjekt weg und allgemein wird nichts dafür getan, dass man danach noch weiterdrucken kann. Da die Düse heiß bleibt, schmilzt sie sich langsam in das Druckobjekt, aber klar.. man kann auch sofort fortsetzen und wieder Filament durchdrücken. Nützt nur nichts ;) 

2. "Advanced Pause Feature" -> Ein Einschleusen der "Bitte um Pause", was bedeutet, dass erst die wichtigsten letzten Bahnen noch aus dem Puffer gedruckt werden und erst wenn alles weitere abgearbeitet ist, die Pause zum Zug kommt. Hierbei werden auch Dinge gemacht wie ein "wipe" (Düse abstreifen), retract, zurückfahren, Druckbett weiterhin heizen, aber Düse abkühlen lassen etc. Aus dem Grund (kalte Düse) dauert es auch, bis man den Druck wieder aufnehmen kann. Es muss nämlich vorher via PID sichergestellt werden, dass die Temperatur wieder erreicht wird, bevor der Druck weitergehen kann. 

Der "Confirm Screen" ist dabei nur ein "Ich bin das Display und weiß nicht wie weit die Firmware gerade ist, deswegen zeige ich mal irgendwas an um den User zu unterhalten und bis ich wieder was vom Mainboard höre". Das geht recht schnell, aber eben nicht unsichtbar.

Also dass Stop/Pause/Resume ne Weile braucht ist bekannt und gar kein Bug. 

Beantwortet das deine Frage ungefähr? ;)

maansch, 2020-08-31:
Hi,
danke für die ausführliche Antwort. 👍

closed 2020-08-31 by maansch