knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #18: TMC_2208

kotarokun, 2020-08-29:
Hello,

i got an question about TMC_2208 Driver. I read about these driver, they would get an issue while using linear advance or other Settings.
I just want to know if there are any information about that? what Settings sould i forget, if i had these Driver for Extruders and Steppers?

thanks for Help. (Gerne auch auf Deutsch;-) )

knutwurst, 2020-08-31:
Hey :)

So it's a question and not a bug, right? :P

I don't know there are any problems. In general, Linear Advance requires a lot of computing power, which the Atmega2560 hardly has. That's why I only activated the feature in the firmware, but never preconfigured it.

In any case, I only see the extruder as problematic when it is running with a TMC2208 in StealthChop mode at 12V. For LA you need a lot of strength and quick changes of direction. This causes problems here and you should consider switching the extruder to at least the SpreadCycle mode (with a solder bridge) or swapping the motor driver for an A4899.

The other axes are not affected. Only the extruder torments itself ;)

Sydney1101, 2020-09-02:
I have no problems with TMC and linear advance on my Mega X. The new sound of the extruder is very special when you are using the linear advance option - but the quality will be better ;-) The lifetime of the extruder will be shorter, based on the continuous changes of the speed. But that is my opinion..

knutwurst, 2020-09-12:
@kotarokun ? ... is this still valid?

knutwurst, 2020-10-02:
-closed: no update-

closed 2020-10-02 by knutwurst