knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #19: Filament Runout Sensor

rogodra, 2020-08-31:
Hello,

Until now, I was using a Marlin 1.1.9 distribution where activating #define FILAMENT_RUNOUT_SENSOR
Where printing from the octoprint, when filament was missing the printing also stopped.
I have seen your firmware and I see it very complete, but when activating this option I get this error:

```
sketch\src\MarlinCore.cpp: In function 'void event_filament_runout()':

sketch\src\MarlinCore.cpp:395:63: error: cannot convert 'const char*' to 'PromptReason' for argument '1' to 'void host_action_prompt_begin(PromptReason, const char*, char)'

       host_action_prompt_begin(PSTR("FilamentRunout T"), false);

                                                               ^

exit status 1
```

Could it be resolved in some way?

Thank you for everything!

knutwurst, 2020-08-31:
The problem is that you are trying to activate the Marlin's own filament runout function, but I don't use it at all and have bypassed it completely because the TFT controls the filament runout on the Anycubic. This is why the function is deactivated in the source code.

What you are looking for is the define ANYCUBIC_FILAMENT_RUNOUT_SENSOR. But that cannot be queried via USB / Octoprint. It is activated by default.

I hope this helps and answers your question?

rogodra, 2020-08-31:
It's a shame, I'm forced to go back to the previous version with marlin 1.1.9.

The davidramiro firmware allows the use of the filament sensor in both TFT/SD and USB/octopi, activating:

FILAMENT_RUNOUT_SENSOR
NOZZLE_PARK_FEATURE
ADVANCED_PAUSE_FEATURE
EMERGENCY_PARSER
ACTION_ON_PAUSE "pause"
ACTION_ON_RESUME "resume"
These last two in Marlin 2.0 I think are:
#define HOST_ACTION_COMMANDS
#define HOST_PROMPT_SUPPORT

Thanks for replying. You are doing a great job. I would try to implement it, but I don’t have enough knowledge.

knutwurst, 2020-09-01:
That could be a feature for the future. I'll dig into it and see what I can do. Unfortunately I'm in the hospital right now and can't test any code that I've written. Maybe I should ask a nurse if I can get a 3D printer :D

rogodra, 2020-09-01:
Thanks again. Good luck and I hope your stay in the hospital is nothing serious. Many moods!

closed 2020-10-01 by knutwurst