knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #20: [BUG] (short description)

MiOrt, 2020-09-03:
### Bug Description

Falsche Voreinstellung ausgewählter Thermistor. Eingestellt ist Thermistor 5. Der von Anycubic verbaute ist aber nach Angaben der Website https://www.anycubic.com/products/5-pcs-ntc-3950-100k-thermistor-with-1-meter-wiring-and-female-pin-head der Typ 11.

Geändert werden müsste Zeile 554 in
#define TEMP_SENSOR_0 11

knutwurst, 2020-09-03:
Ich habe mich an den Original-Sourcecode von Anycubic gehalten. Da Typ 5 und 11 sowieso beides 100k Thermistoren sind, sehe ich da kein Problem.

Siehe: https://github.com/ANYCUBIC-3D/MEGA_S/blob/753c24c06faa1c714bfa6d01cf10d6ff22f5f0d1/Configuration.h#L258

Beim Mega X dasselbe: 
https://github.com/ANYCUBIC-3D/MEGA_X_V1.2.6/blob/38c6846c4c4300a0f68cb3c48e64e245db3c6f77/Configuration.h#L258

MiOrt, 2020-09-04:
Moin,

prinzipiell sind das ja alles 100k Thermistoren. Die Typen unterscheiden sich jedoch durch die Kennlinie, sonst bräuchte man ja gar keine Auswahl anbieten. Somit unterscheidet sich die tatsächliche Temperatur beim identischen Samplingwert. Mir ist das mit dem Typ 5 statt 11 erst aufgefallen, als ich ein Posting in irgend einem Forum las und dann den Hinweis auf der Produktseite von Anycubic sah. Bei den Typen 5 und 11 sind die hinterlegten Kennlinien-Tabellen wie folgt:

#if ANY_THERMISTOR_IS(5) // 100k ParCan thermistor (104GT-2)
// ATC Semitec 104GT-2 (Used in ParCan)
// Verified by linagee. Source: http://shop.arcol.hu/static/datasheets/thermistors.pdf
// Calculated using 4.7kohm pullup, voltage divider math, and manufacturer provided temp/resistance
const short temptable_5[][2] PROGMEM = {
  {    1 * OVERSAMPLENR, 713 },
  {   17 * OVERSAMPLENR, 300 }, // top rating 300C
  {   20 * OVERSAMPLENR, 290 },
  {   23 * OVERSAMPLENR, 280 },
  {   27 * OVERSAMPLENR, 270 },
  {   31 * OVERSAMPLENR, 260 },
  {   37 * OVERSAMPLENR, 250 },
  {   43 * OVERSAMPLENR, 240 },
  {   51 * OVERSAMPLENR, 230 },
  {   61 * OVERSAMPLENR, 220 },
  {   73 * OVERSAMPLENR, 210 },
  {   87 * OVERSAMPLENR, 200 },
  {  106 * OVERSAMPLENR, 190 },
  {  128 * OVERSAMPLENR, 180 },
  {  155 * OVERSAMPLENR, 170 },
  {  189 * OVERSAMPLENR, 160 },
  {  230 * OVERSAMPLENR, 150 },
  {  278 * OVERSAMPLENR, 140 },
  {  336 * OVERSAMPLENR, 130 },
  {  402 * OVERSAMPLENR, 120 },
  {  476 * OVERSAMPLENR, 110 },
  {  554 * OVERSAMPLENR, 100 },
  {  635 * OVERSAMPLENR,  90 },
  {  713 * OVERSAMPLENR,  80 },
  {  784 * OVERSAMPLENR,  70 },
  {  846 * OVERSAMPLENR,  60 },
  {  897 * OVERSAMPLENR,  50 },
  {  937 * OVERSAMPLENR,  40 },
  {  966 * OVERSAMPLENR,  30 },
  {  986 * OVERSAMPLENR,  20 },
  { 1000 * OVERSAMPLENR,  10 },
  { 1010 * OVERSAMPLENR,   0 }
};

#if ANY_THERMISTOR_IS(11)
// QU-BD silicone bed QWG-104F-3950 thermistor
const short temptable_11[][2] PROGMEM = {
  {    1 * OVERSAMPLENR, 938 },
  {   31 * OVERSAMPLENR, 314 },
  {   41 * OVERSAMPLENR, 290 },
  {   51 * OVERSAMPLENR, 272 },
  {   61 * OVERSAMPLENR, 258 },
  {   71 * OVERSAMPLENR, 247 },
  {   81 * OVERSAMPLENR, 237 },
  {   91 * OVERSAMPLENR, 229 },
  {  101 * OVERSAMPLENR, 221 },
  {  111 * OVERSAMPLENR, 215 },
  {  121 * OVERSAMPLENR, 209 },
  {  131 * OVERSAMPLENR, 204 },
  {  141 * OVERSAMPLENR, 199 },
  {  151 * OVERSAMPLENR, 195 },
  {  161 * OVERSAMPLENR, 190 },
  {  171 * OVERSAMPLENR, 187 },
  {  181 * OVERSAMPLENR, 183 },
  {  191 * OVERSAMPLENR, 179 },
  {  201 * OVERSAMPLENR, 176 },
  {  221 * OVERSAMPLENR, 170 },
  {  241 * OVERSAMPLENR, 165 },
  {  261 * OVERSAMPLENR, 160 },
  {  281 * OVERSAMPLENR, 155 },
  {  301 * OVERSAMPLENR, 150 },
  {  331 * OVERSAMPLENR, 144 },
  {  361 * OVERSAMPLENR, 139 },
  {  391 * OVERSAMPLENR, 133 },
  {  421 * OVERSAMPLENR, 128 },
  {  451 * OVERSAMPLENR, 123 },
  {  491 * OVERSAMPLENR, 117 },
  {  531 * OVERSAMPLENR, 111 },
  {  571 * OVERSAMPLENR, 105 },
  {  611 * OVERSAMPLENR, 100 },
  {  641 * OVERSAMPLENR,  95 },
  {  681 * OVERSAMPLENR,  90 },
  {  711 * OVERSAMPLENR,  85 },
  {  751 * OVERSAMPLENR,  79 },
  {  791 * OVERSAMPLENR,  72 },
  {  811 * OVERSAMPLENR,  69 },
  {  831 * OVERSAMPLENR,  65 },
  {  871 * OVERSAMPLENR,  57 },
  {  881 * OVERSAMPLENR,  55 },
  {  901 * OVERSAMPLENR,  51 },
  {  921 * OVERSAMPLENR,  45 },
  {  941 * OVERSAMPLENR,  39 },
  {  971 * OVERSAMPLENR,  28 },
  {  981 * OVERSAMPLENR,  23 },
  {  991 * OVERSAMPLENR,  17 },
  { 1001 * OVERSAMPLENR,   9 },
  { 1021 * OVERSAMPLENR, -27 }
};
#endif

Hier sieht man die Unterschiede recht deutlich. Theoretisch stimmen also die tatsächlichen Werte an der Nozzle nicht mit den gewünschten überein, was ich jedoch mangels passendem Kontaktthermometer nicht prüfen kann.

Warum das auch in der Anycubic-Software so gemacht wird, auf der Ersatzteilseite von denen aber explizit Typ 11 angegeben ist, weiß ich nicht. ich schreibe Anycubic an und bitte um Aufklärung. 

Es druckt ja anscheinend nicht schlecht, fraglich ist nur, was da an der Nozzle tatsächlich vor sich geht.

knutwurst, 2020-09-04:
@MiOrt, du hast Recht. Das sieht wirklich ganz anders aus. Ich habe jetzt nicht verfolgt, welchen Wert OVERSAMPLENR annehmen kann, aber stimmt schon.. wir wollen ja nicht, dass die Kiste falsch misst. Ich bin immer davon ausgegangen, dass Anycubic wissen, was sie da tun.

> Warum das auch in der Anycubic-Software so gemacht wird, auf der Ersatzteilseite von denen aber explizit Typ 11 angegeben ist, weiß ich nicht. ich schreibe Anycubic an und bitte um Aufklärung.

Ja, mach das mal! Ich bin gespannt und wenn es tatsächlich ein Fehler von Anycubic ist, dann bedeutet das, dass auch andere Drucker davon betroffen sind.

MiOrt, 2020-09-04:
Ich habe mal Kontakt aufgenommen, werde berichten, sobald die Antwort da ist.

Prinzipiell ist der falsche Typ ja relativ unkritisch, wenn die Kennlinien wenigstens ähnliche Formen haben (hab ich nicht geprüft). Da man die Temperatur ja eh durch Versuche am aktuellen Material ermittelt, ist der reale Wert dann wurscht, solange er nur immer gleich weit vom realen Wert abweicht ist. Blöd ist nur, dass die Werte keine Aussagekraft bei Diskussionen haben, da der Gegenüber möglicherweise seine Temp mit einer anderen Kennlinie ermittelt hat und somit die Werte nicht vergleichbar sind. Das könnte (muss nicht) die endlosen Threads rund um die richtigen Einstellungen erklären. Wäre dann auch blöd, weil die Hersteller ja auch die Angaben zur Temperatur machen und man mit der falschen Kennlinie schon ganz schön daneben liegen kann.

Vielleicht liefert Anycubic aber auch alle Geräte standardmäßig mit dem Typ 5 aus und ich lag falsch. Vielleicht können die Typ 5 nicht als Ersatzteil liefern und nehmen stattdessen den Typ 11? Wer weiß, ich melde mich.

Sydney1101, 2020-09-05:
Ich habe mal mit einem Thermometer außen am Block meines Mega X (hinter der Silikonsocke) gemessen - während er gerade druckte. Eingestellt: 205°C, Hotend: 190°C, Düse (Lüfter aktiv): 185°C. Alles ohne Wärmeleitpaste o.ä. gemessen. Nur mal so als Anhaltspunkt. Ich werde nochmal messen, wenn der Druck fertig ist (ohne Lüfter)....

Naja, so ganz egal ist es ja nicht - gerade bei höheren Temperaturen. Da fängt man grob bei der Mitte des Herstellerwertes an und spielt da rum. Falls dann was nicht passt hat man sich recht schnell die Düse zugebacken. Hatte mein Spass mit PLA+ von SUNLO. Der Unterschied in der Kennlinie ist doch recht groß.

Sydney1101, 2020-09-05:
So. Meine Finger haben sich gefreut.... Neue Messung ohne Lüfter. Eingestellt: 205°C, Hotend: 200°C, Düse 180°C

knutwurst, 2020-09-05:
Danke noch mal ihr beiden :D 
Ähm, @Sydney1101 , kannst du denn einen Test mit dem Sensortyp 11 machen? Ich kann dir sonst auch eine Firmware damit bauen und hier hochladen, wenn das hilft. Dann schauen wir mal, wie sich das verhält. 

Kommt dann in das nächste Release rein.

Sydney1101, 2020-09-05:
Ja klar. Kannst Du mir eine bauen? Ansonsten müsste ich wieder alles manuell anpassen ;-)

knutwurst, 2020-09-05:
Bitteschön :) 
[i3_MEGA_X_TMC_v1.1.6_BETA.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/5177949/i3_MEGA_X_TMC_v1.1.6_BETA.hex.zip)

Sydney1101, 2020-09-05:
Eingestellt: 205°C, Hotend: 172°C, Düse: 172°C. Von daher war der alte Typ korrekt - würde ich mal sagen ;-)

knutwurst, 2020-09-05:
Na sowas :) 
Auch nach PID Tuning?

Sydney1101, 2020-09-05:
Nein. Das habe ich nicht gemacht, aber mit OctoPi die Aufzeichnung der Temperatur kontrolliert. Irgendwann schwingt sich der Regler ja ein - auch ohne perfektes PID-Tuning ;-)

Sydney1101, 2020-09-05:
Ohne Wärmeleitpaste in der Hülse kann man den Temp.-Fühler im Hotend eh vergessen ... ;-)

MiOrt, 2020-09-05:
Vielen Dank schonmal für die Tests und die Mühe. Ich bin unterwegs, habe auch von Anycubic noch keine Antwort auf meine Anfrage. Keine Ahnung wie lange die Sachen m Regelfall brauchen, aber ich poste umgehend, wenn sich was tut.

knutwurst, 2020-09-17:
So... also da (auch nach meinen Tests) und nach Anleitung von E3D mit den Original-Hotends der Thermistor auf 5 stehen muss, schließe ich das hier mal :)

Notfalls machen wir es wieder auf, aber es sieht soweit ja nicht nach einem Fehler aus.

closed 2020-09-17 by knutwurst

MiOrt, 2020-09-17:
Alles klar. Hab immer noch keine Antwort von Anycubic.