knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #21: BLTouch leveling stoppt mit Error an Position 2,1

sitaro, 2020-09-07:
Hallo,

Das Leveling mit dem BLTouch stoppt immer mit einem Fehler bei Position 2,1.
____

Recv: Error:Probing Failed
Recv: X:84.50 Y:172.50 Z:10.00 E:0.00 Count X:6760 Y:13800 Z:8000
Recv: ok

-------
Der Fehler ist stabil reproduzierbar.

Drucker: Anycubic Mega X

TMC2208

BLTouch original

Danke für die Hilfe

Mathias

sitaro, 2020-09-07:
Falls es weiterhilft, hier der debug
[bltouch-debug.txt](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/5184467/bltouch-debug.txt)

knutwurst, 2020-09-12:
Hey, entschuldige die späte Antwort.
Hast du dich an die Anleitung gehalten? In diesen Fällen liegt es fast immer daran, dass die Probe zu tief fahren muss und damit unter der Nulllinie steht. Ich habe in der [Version 1.1.6](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/tag/1.1.6) die Toleranz auf bis zu 10mm erweitert. Trotzdem ist es essenziell, dass du das Bett vorher noch per Hand richtig levelst.

sitaro, 2020-09-12:
Hallo, 

Mit der neuen Version ist das Problem gelöst, Danke für Deine Arbeit.

Grüße aus Frankfurt 
Mathias Paezolt

closed 2020-09-17 by knutwurst