knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #22: [BUG] Junction Deviation causing stuttering on some curves

brunoosti, 2020-09-12:
Have you been getting stuttering on curves with junction deviation enabled? It translates as rough surface on the outer wall. Maybe try printing a big cone with 1.2mm walls to reproduce the issue (couldn't notice this on single wall tests).
Tried everything on the slicer (plus disabling linear advance) before reverting back to original firmware and testing more without JD on your updated firmware. I'm using classic jerk now and the print surface is supper smooth. 
It seams there is a problem with this feature: https://github.com/MarlinFirmware/Marlin/issues/17920

Sydney1101, 2020-09-12:
Which slicer do you use? The problem is also based on how the slicer works...

brunoosti, 2020-09-12:
Yep.. unfortunately Cura.

knutwurst, 2020-09-12:
Hey :) 
Could you please try this? -> https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/How-to-set-up-Cura

brunoosti, 2020-09-12:
Hey!! 
Tested my resolutions even higher thinking that might be the problem. No success.

knutwurst, 2020-09-12:
That's weird.. 
Do you print via SD or USB?

brunoosti, 2020-09-12:
(i meant lower my resolution, using higher values)
I print via SD.
But wait! Should I enable Junction Deviation on cura? Never found this setting. Only had my calculation for 1000 accelleration and 10 jerk, using a JD of 0.04. Maybe i got your last sentence wrong.
Than i just disabled it in the firmware files reuploading to my printer.

brunoosti, 2020-09-12:
the files I was having problem with where wide curves, part of a helmet. So, maybe, to replicate the issue, try printing 1/4 of a 2mm thick wall sphere that takes almost all of i3m-s bed.

Sydney1101, 2020-09-13:
Yes, the problem occur on wide curves and small walls or oval objects. In the other cases the print quality is better.

knutwurst, 2020-09-17:
Can you tell me which exact configuration you need? I can build some test firmwares for you and I'd like to see the results :)

brunoosti, 2020-09-17:
Wow thanks : )))
I followed your guidelines on the source code and built a mega-s hex file for TMC2208s with reversed connectors on the board. That is it. Had LA, s-curve and JD (at 0.04 for 1000 accel and 10 jerk in cura) all enabled. See my config files,
 maybe it can help. Also, one of the STLs I was having problems printing before disabling JD and the cura profile I'm using for this project. 
[JD stuttering.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/5238941/JD.stuttering.zip)

knutwurst, 2020-09-17:
@Sydney1101 , can you tell me what firmware you need? (printer and addons)

@brunoosti, I'll take a look tomorrow when I have some free time... in the meantime, can you please try a print without enabling the Jerk settings in cura and also disable LA?

Good night ;)

brunoosti, 2020-09-17:
sure! 
What you mean is to uncheck the "enable jerk control" (and acceleration control) in cura and let the printer deal with the file through it's firmware (it seams this may be the culprit.... I've been printing with that box enabled the whole time, choosing the values I used to calculate my 0,04 JD: 10 and 1000. I'll get back with the results soon).
I've tested with K value of 0 with no success. Let's see if the above confirms itself.

brunoosti, 2020-09-17:
No deal... Updated my firmware, enabling JD, m502 m500 in pronterface, unchecked jerk and acceleration boxes in cura and set K to 0. Placed the print half way int the bed to get the problematic portion. Still stuttering on that STL I've attached before (mostly on inner walls). 
By the way how did you get the JD value of 0.013? Jerk is 10 and accel is 1600. Shouldn't it be 0.025?
Tested with your default values and with 1000 accell and 10 jerk

knutwurst, 2020-09-18:
Damn... and with "classic jerk" it works fine? Maybe I should go back to the classic jerk settings...

 > By the way how did you get the JD value of 0.013? Jerk is 10 and accel is 1600. Shouldn't it be 0.025?
I calculated the JD value by using this thread: http://blog.kyneticcnc.com/2018/10/computing-junction-deviation-for-marlin.html but I found out that the value was not perfect when printing with high speeds. So I fiddled around with the config and ended up with 0.013 which gave me perfect results.

Also... when looking at the original Marlin sources and the bug entry found [here](https://github.com/MarlinFirmware/Marlin/issues/17342) I think I need to test more.

Maybe you can help me with that issue? You can compile the FW by yourself, right? If not, I can provide some hex files to test. I'd really appreciate your help :)

Sydney1101, 2020-09-18:
@knutwurst Anycubic Mega X + TMC. I also follow the marlin thread. So, the "bug" is not so easy to solve.... When JD works as expected, the results are quit good. Therfore, I currently I agree to decativate JD. Otherwise, there will be problems with some curves.. My slicer: ideaMaker 3.6.1

Sydney1101, 2020-09-18:
@brunoosti Here are the default values of the Mega X firmware:
Maximum Acceleration (mm/s2):
M201 X400 Y400 Z60 E10000
Accelerations: P=printing, R=retract and T=travel
M204 P400.00 R3000.00 T400.00
Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
M205 S0.00 T0.00 B20000 X4.00 Y4.00 Z0.10 E8.00

Where did you get the value of 1600?

knutwurst, 2020-09-18:
This evening I'll try the files @brunoosti provided and test them on my Mega S. I'm also trying to figure out, why some ppl are having issues with USB printing. After some test results from other users, I'll provide all fixes in the new release.

@Sydney1101 : If you need a specific Version now for testing, I can upload it in the comments ;)

> Where did you get the value of 1600?

This is the default for the MEGA S and P. Your Mega X uses 400.

brunoosti, 2020-09-18:
@knutwurst Yes! I'm not experienced or a programmer at all, but I manage to compile from your source code as everything is well explained in there : ). Maybe I can help with the easy stuff.

@Sydney1101  as knutwurst said: mega s' acceleration. I even tried with slower values and disabling stuff in the slicer (like compensate wall overlaps and some other settings I can't remember right now) only solving the stuttering by giving up junction deviation in the firmware.

Sydney1101, 2020-09-18:
@brunoosti Sorry, I thought you have a Mega X ;-)

Sydney1101, 2020-09-18:
I currently use JD values 0,2. I know the value is very high, but my results on curves are better. But based on the missing solution for that problem (Marlin threat), I would prefer a version without JD ;-) The red picture shows my similar problem with curves and low JD values:

[https://github.com/MarlinFirmware/Marlin/issues/17920 ](url)

brunoosti, 2020-09-18:
No prob : )
Maybe you'd be better slowing the outer wall only (I use 30mm/s) and enabling "retract before outer wall" to "reset" pressure in the nozzle and don't have overextrusion / uneven layers close to the z seam, because of the faster inner walls, like I was experiencing. That and linear advance made a big difference.

brunoosti, 2020-09-18:
take a look at BLGecko comment. Maybe it could help in some way. 
https://www.reddit.com/r/3Dprinting/comments/ccvn0z/is_junction_deviation_better_than_using_normal/

brunoosti, 2020-09-23:
see this: https://github.com/Ultimaker/Cura/issues/6748#issuecomment-697358893
"I had the same issue printing the benchy hull. I increased the serial buffers on marlin configuration as I am using an external BTT TFT35 and I also increased Maximum resolution and Maximum deviation to 0.25mm on Cura and it solved the problem even with junction deviation enabled."

knutwurst, 2020-09-23:
Hey you two ;) I'm doing some tests in my Facebook group. I'll use your test data for that, @Sydney1101. It seems like I'm making progress in terms of print quality. Junctiomn Deviation also requires too much computing power and actually leads to blobs.

Here are a few test firmwares for Mega, Mega S and Mega X (with and without TMC):
[Firmwaretest_2020_v1_and_v2.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/5269944/Firmwaretest_2020_v1_and_v2.zip)

If a BLTouch is available, you have to pull out the servo connector.

Please let me know which version works best.

thanks <3

michaelwc, 2020-09-24:
Unfortunately Mega X TMC 032 has broken my printer. Stuck in some kind of boot loop and unable to connect via USB and reload new Firmware! Pronterface just loops through connection and locks up. Cura just doesnt connect and I cant load firmware.

knutwurst, 2020-09-24:
@michaelwc  OH NO! That was not intended :/ 
But since the firmware is not overwriting the bootloader, you are still able to flash via Tools like Cura 15.04 or the popular [XLoader](https://github.com/knutwurst/xLoader) which can flash almost everything using AVRDude under the hood.

Let me know if it worked for you.

michaelwc, 2020-09-24:
Hi Knutwurst.
Cura would not do it. Also, Pronterface just keep looping through. I had to pull the knutwurst sources and flash from PlatformIO to get a base image. Loop broken.

However, I have now reverted back to the Anycubic FW so I can print again.  I will wait for JD (or Linear Accel) to be fixed in custom firmware before flashing again. I would try another FW from you but I need to get some printing done and printer is tied up for the time being.

Thanks,
M

knutwurst, 2020-09-24:
@michaelwc Only Cura 15.04 is capable of flashing non-initialized ATMEGA Boards. That's why XLoader and PlatformIO also work... they don't care if it's a printer or just a microcontroller ;)

Could you test the 10/20/30 Versions? (not the x1, x2 etc).

Thank you in advance :) 
The Problems with Junction Deviation and USB printing will be resolved in the 1.1.7.

michaelwc, 2020-09-28:
OK, did some testing....

Knutwurst testing |   |   |  
-- | -- | -- | --
Version | K | Result | Notes
0.0.10 | 0.11 | Fail | Fails layer 3, Silky smooth finish though.
0.0.20 | 0.11 | Fail | Fails Layer 7, Rough finish, uneven extrusion
0.0.30 | 0.11 | Fail | fails Layer 7, Rough finish, uneven extrusion
0.0.31 | 0.11 | Fail | Fails layer 9,Rough finish, uneven extrusion
0.0.32 | 0.11 | Fail | Fails to boot, boot loop.

The closest to ideal is 0.0.10. I am going to try again with some different slicing parms to see if I can tweak out the LA issue.

michaelwc, 2020-09-28:
OK, I'm gonna walk away from this for now. I ran some other tests on 0.0.10 and can see this is very much speed and complexity related. When I run a print at 40mm, I get some nice glassy finish. Print continues to quite high up the layers before E0 gives out. However, When I ramp the speed right up (100mm) I get some nasty stuttering and the extrusion finish is nasty and inconsistent.
Feels like some kind of buffer overflow or processing surge? Im not too familiar with the Trigorilla board and how to determine processor saturation etc.

knutwurst, 2020-10-01:
There is a bug in 1.1.6 and all testing firmwares. Please downgrade to 1.1.5. I'll do my best to fix it in the upcoming version.

brunoosti, 2020-10-01:
I've been printing on 1.1.6 for a week nonstop with junction deviation disabled  with no problem. Even using filament runout sensor a couple of times.

knutwurst, 2020-10-01:
@brunoosti , yes.. via SD everything seems to be ok, but via USB/Octoprint, the print quality is really poor on 1.1.6, so I needed to take this version down.

DanJunior78, 2020-10-05:
@knutwurst how's it going with your tests? I'm right now unhappy with my Mega X compared to my i3 Mega S. It was on stock Firmware the same problem. All edges are really bad. All round and not sharp.
If I compare both printers the Mega X is always,like stopping in the corners and starting again. might it make sense to increase the low acceleration values in the firmware. Can't believe that the difference is so high between the i3 Mega and the Mega X,at least on the X Direction.

Sydney1101, 2020-10-06:
@DanJunior78 I have a Mega X and my way to solve this problem is to set JD to a factor of 0.2 or max. . Only in this cases, the acceleration is high enough to create sharp corners. I will try some tests with higher accelerations and lower JD.

DanJunior78, 2020-10-06:
@Sydney1101 it was impossible to get better results on my Mega X. My Mega S is printing perfect. I've done some tests on my Mega X and have it running like my i3. The factory accelerations are a shame. I've opened a request to @knutwurst to change the factory values to my tested once ( I've open a change request...you can see a pic of the result.
But on JD I would never go so high. On the cubes the Z on top has big differences using even 0.16 compared to the once of the i3 of 0.013. 
I was little bit curious seeing stock values for the Mega X which are 20% of the i3...for the X and Y...but the Extruder and Z was same. 
But mine is working fine now.

Sydney1101, 2020-10-07:
@DanJunior78 I will continue our conversation in the firmware issue.

knutwurst, 2020-10-07:
Hey ;)
I'll release the 1.1.7 BETA 2 within this week, which also includes the feedrate and Acceleration values you provided @DanJunior78 . Also the Octoprint-Layershift-Issue should be gone. Seems like an odd behaviour when pushing the Special Menu to the file list without actually having an SD card inserted. It's really strange...

DanJunior78, 2020-10-07:
@knutwurst thanks, that's great. Please use the 4th print values with a JD of 0.013, like for the i3. The tmc2208 Uref for the y axis has to be increased to around 1,05V. The motor itself is not getting too hot.

knutwurst, 2020-10-07:
To be honest, I'd like to go back to the classic jerk mode, because JD causes a CPU overload on very small edges or bug round shapes. Maybe you can provide a good classic jerk value?

DanJunior78, 2020-10-07:
To be honest, I've never tried it before. I've been using my i3 with the stock board and never touched it, I've even used your setting to prepare the firmware for the MKS board which I've installed last weekend. I would start from the i3 values and I could do some tests on my Mega X.

Sydney1101, 2020-10-07:
Based on the current curve problems, I will be happy, if you go back to classic jerk 👍

brunoosti, 2020-10-07:
How JD is behaving on 32 bit boards?? Same problem on curved thin walls? Anyone knows?
I reverted back to 1.1.5 with classic jerk and the printer is fine (and TMC2208s with reversed connectors, if anyone would want my hex file). Have never printed so smoothly. I only wished linear advance would be more effective on bowden setups (after calibration reached the best corners at 30mm/s on outer walls, with retract before outer wall to compensate pressure on the nozzle from the faster inner walls).

DanJunior78, 2020-10-07:
I'm running my i3 with a 32 bit MKS board (changed the Trigorilla 1.4) there I have no issues at all. On my Mega X with a 8 bit board I've optimized the factory settings for the acceleration. Since them the printing quality is looking same like on my i3.[[image]
The white one was the i3 with the Trigorilla board. The black one with the MKS board with 50mm/s and the green one with my Mega X with 40 mm/s (slightly over extrusion).
Maybe I'll remove the Trigorilla board on my Mega X too. 
Somehow the TFT from Anycubic doesn't make me really happy and all the missing possibilities to get the UART functions,too.

knutwurst, 2020-11-12:
Fixed in 1.1.7 Beta 3

closed 2020-11-12 by knutwurst