knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #23: Problem with Octoprint

jhonny811, 2020-09-14:
Good evening, I installed the MEGA_S_TMC_BLT_10_v1.1.6.hex release, I connected the printer to the raspberry with octoprint on raspbian, I started a print and I noticed that the axes were jerking. How can I solve? Thank you

knutwurst, 2020-09-17:
What do you mean by "the axes were jerking"? Can you upload a video? Also I need your gcode :)

Did you try via SD? What hardware do you use?

Tom-Moveman, 2020-09-17:
I have the same problems with the firmware, a lot of jerking and blobs with Octoprint.  I have the feeling it hasn't been tested on Octoprint, or the developer doesn't have an Octoprint.  you can only print the firmware for SD, the quality via USB is poor.

MiOrt, 2020-09-17:
Hi, I am using Version 1.1.5 and was having a lot of trouble especially on larger prints using repetier-server. After testing different settings I finally reduced the input buffer setting from 127 to 63 byte and since then everything works as expected. I had a lot of entries in the log like "Warning: Communication timeout - resetting communication buffer." Maybe there is also such a setting in Octoprint and the server simply tries to send too many bytes at once to the printer?

jhonny811, 2020-09-17:
With the MEGA_S_TMC_BLT_10_v1.1.5.hex release, it doesn't give me any problems

Tom-Moveman, 2020-09-17:
Unfortunately, the buffer is not customisable in Octoprint, otherwise you could fix something. In the firmware, this is simply not taken up or poorly realised. I'm beginning to feel that it's being promised more than it actually involves. I'm going to test another version from other developers soon... There were no such problems in Ramiro's 1.1.9.

knutwurst, 2020-09-17:
Hey guys ;) 

@jhonny811 , would you do me a favour and test these two specific firmware files?

It's pretty odd that 1.1.5 differs from 1.1.6 since the versions share the same codebase ... I only switched the compiler.

Also @Tom-Moveman  and @MiOrt , it would be great if you could test the firmwares, but I don't know if you're using the BLT ones.

I'll try my best to figure out what's going on here.

Good night ;) 

[MEGA_S_TMC_BLT_10_v1.1.6_AB_Testing.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/5241669/MEGA_S_TMC_BLT_10_v1.1.6_AB_Testing.zip)

jhonny811, 2020-09-17:
this weekend I will test

knutwurst, 2020-09-18:
And this one... especially for USB.
[MEGA_S_TMC_BLT_10_v1.1.6.1_C.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/5245248/MEGA_S_TMC_BLT_10_v1.1.6.1_C.hex.zip)

Tom-Moveman, 2020-09-28:
so, here's my feedback, with the test version no improvement.  it is still jerking.  I then swapped USB cables, reinstalled Octoprint and even tested another Pi, the problem persisted.  I then tested a different firmware and the problem was gone immediately.  maybe it is in your version because of the many toggles or wrong settings.  My test is now over, I am now using a different version because unfortunately I did not get good results here.

knutwurst, 2020-10-01:
On Facebook, there were also complaints about the print quality... and it only occurred on 1.1.6, not on 1.1.5. I'll figure out what's wrong there. Please downgrade to 1.1.5.

Tom-Moveman, 2020-10-01:
Sorry but I won't use your versions anymore.  I wonder, are you even testing this?  do you even have octoprint?  a secret beta test on the user is not great.  in the instructions everything is described as so great and easy.  sorry but test the stuff beforehand.  so trust is destroyed.

linuxq, 2020-10-01:
I used all Knutwurst versions on my Mega-S with Octoprint. Even 1.1.6 is running good here! Ok, I don't have BLT or other drivers. Oliver will only be able to test on his setup. Thus there always is a risk of bugs and that is what I took into consideration before flashing too.

knutwurst, 2020-10-01:
@Tom-Moveman, as @linuxq said, I can only test everything on my setup and not on yours. I don't use Octoprint, but use the SD card or a Mac, which I use to print via USB.

All I'm reading from you right now is that you have problems with my firmware and that you complain that I don't test all 36 configurations individually - with firmware that you get completely free. You complain that I don't have a QA department even though I'm the only developer who does this in my spare time? For real?
I even don't own a Mega X or the new DGUS display and everything works anyway.

As already said, the problem only occurs with 1.1.6. I only know that since today. If you had tested a different version beforehand, you could have told me that too, but instead you complain that I don't use Octoprint.

If you can do better then do it yourself :)

@ jhonny811: Your bug is verified and I'm doing my best to fix it. Do not worry. I can do it. I'm just on vacation and don't have a printer on hand. Please use 1.1.5 until the new version is available.

Thank you :)

Tom-Moveman, 2020-10-01:
What are you talking about here?  Marlin is free.  Other developers also do this in their free time. Ramiro, Stef, etc. not just you. Learn to deal with criticism.  But apparently it is true what is in my Spanish forum about you, talk badly to everyone else and just want fame.  For me the discussion is over, you have shown that what they say is true, get rude when someone says something against you ... you still have to learn a lot ...

knutwurst, 2020-10-01:
@Tom-Moveman , just show me where I was rude in your spanish forum ;) 
And please don't forget: You don't have to use my firmware. If you can explain what the problem is and if you can do it better, please open a pull request or create a bug ticket with a explanation how to reproduce the error.

> In the firmware, this is simply not taken up or poorly realised. I'm beginning to feel that it's being promised more than it actually involves.

This is your opinion. Please see above.

I wish you the very best with your alternative firmware, which works so much better <3

XSSammy, 2020-10-09:
@knutwurst I've got a stock i3 Mega that I print too via octoprint.  Happy to help with any diagnosis you need with this bug.

kentbrock, 2020-10-28:
I do not have any issues with Octoprint runnning on a RPI 3B together with Knutwurst 1.1.6. Benchy, calibration cube and other prints looking pretty good via Octoprint or in other words I do not the the difference between SD Card or Octoprint. 

Printer: Anycubic Mega S with TMC2208 V3 (all axes running at 0.95V - Extruder running at 1.1V)
Octoprint: Only Webcam and Filament Manager Pluging running

knutwurst, 2020-10-30:
@Roemer , @XSSammy and all others with print quality issues... would you mind to test the [1.1.7 BETA 2](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/tag/1.1.7-beta.2) ? 
This would help me a lot!
Thanks in advance!

Gewbster, 2020-11-02:
Hy,

I just started using your Firmware, I used Marlin 1.1.9 before that. 
Not sure if that's the same issue, I noticed that the speed wasn't constant when the printer has to change many directions simultaneity  ( like in a circle) the print head stuttered and that resulted in bad print quality. 
By the way I'm also using Octoprint. 
I could fix the issue by changing the Maximum resolution and maximum deviation settings in Cura, that is resulting in less short printer moves and so easier for the processor to calculate. 
not sure if that's the same issue as  jhonny811 and  Tom-Moveman  described, but maybe it helps. 

best regards

knutwurst, 2020-11-12:
1.1.7 Beta 3 is out and all issues should be fixed.

closed 2020-12-13 by knutwurst