knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #24: Printable volume too large

Oderik, 2020-09-26:
On my Mega S, when a print is done, the bed moves to max y axis. The physical range is smaller then the configured range, so the print ends with noisy attempt to move the bed where it can't be.
Could you please revert the bedsize to it's original values? I guess it included tolerances.

knutwurst, 2020-09-27:
Hi,
did you configure the print bed size in your slicer? (see: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Set-new-print-bed-size)

The extra 10mm should be no problem at all and normally it's an issue with the Y endstop.

Oderik, 2020-09-27:
Thanks for the quick response!
I updated Cura like on the screenshot before experiencing the issue.
Do you suggest that the physical endstop at y=0 could be misplaced or what do you mean by endstop issues?
The nozzle seems to placed at a reasonable distance to the edge of the bed (stops just before leaving it) while on the "virtual" end it leaves the bed. Can I fix this by configuration without altering the firmware?

(Sorry, I know this is actually support for my printer and maybe not specific to your firmware)

knutwurst, 2020-10-01:
> Do you suggest that the physical endstop at y=0 could be misplaced or what do you mean by endstop issues?

Yes... it might be, that the endstop is a bit misplaced, but also for the older i3 Mega versions with a smaller bed, I will revert the size, so it is only applied to the Mega S and Mega P.

Oderik, 2020-10-01:
I do have a Mega S where this occurs. But I solved it for me by removing the line "G1 Y180 F2000" from my end G-Code in Cura.
When I use the device touch screen controls or OctoPrint controls, the motors do not try to leave their valid area but stop in time.

knutwurst, 2020-10-01:
Ah, great! So it's not quite firmware related? Can I close the issue? ;)

Oderik, 2020-10-02:
I'd say yes, you can close it. Thanks anyway!

closed 2020-10-02 by knutwurst

linuxq, 2020-10-06:
I have the same problem on my Mega-S with 1.1.5. I changed the print bed size in cura und use your endcode in Cura.