knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #25: [BUG] Crash into bed during leveling with BLTouch v3.1 on Mega-S

McApe, 2020-09-27:
### Bug Description

During leveling started with the special menu the nozzle crashes into the bed with BLTouch v3.1 on Mega-S with the precompiled firmware file.

### My Configurations

I used MEGA_S_BLT_10_v1.1.6.hex

### Steps to Reproduce

1. flash firmware (loaded fw defaults and save eeprom afterwards)
2. follow configuration steps from https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/BLTouch-Installation-(deutsch)#konfiguration
3. start step 20 from configuration above 

**Expected behavior:** Nozzle should not crash into the bed, leveling

**Actual behavior:** Nozzle crashes into the bed

#### Additional Information

Got it working with following changes:
1. uncomment the following defines  
```
#define BLTOUCH_SET_5V_MODE
#define BLTOUCH_FORCE_SW_MODE
#define BLTOUCH_FORCE_MODE_SET  // this is needed only once
```

2. build and flash firmware
3. do the whole leveling (don't know if this realy is needed)
4. comment #define BLTOUCH_FORCE_MODE_SET
5. build and flash firmware again
6. it now works as expected

btw i have a trigorilla board that has no logo and is labeled "v0.0.2".
see https://www.reddit.com/r/anycubic/comments/gstlml/got_a_trigorilla_v002_board_in_my_2020_anycubic/

knutwurst, 2020-10-01:
@McApe , Thanks for the detailed bug description! This is very helpful and I will see whether these changes also work for other BLTouch versions and then incorporate these into the firmware as the default so that BLT 3.1 users no longer have any problems.

Thank you for your help!

closed 2020-10-01 by knutwurst

DarKOrange75, 2020-10-21:
Hi, in your BL Touch Guidelines it says "When extended, it must protrude less than the nozzle." 
Are you sure about that? How can the sensor be triggered if it is higher than the nozzle when extended? The nozzle will then crash onto the bed before the sensor be triggered.

knutwurst, 2020-10-21:
@DarKOrange75 OH! You are right!
Of course, the probe must protrude further than the nozzle! mea culpa!

And thank you! I already fixed the text... If you see some more mistakes (also grammatical mistakes or sth like that, please let me know. :)

DarKOrange75, 2020-10-21:
Wow that was fast! Thank you @knutwurst :-)