knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #26: [FR]Build for the 32-bit Trigorilla Pro version of i3 mega

Blue-Beaker, 2020-10-02:
### Description

Build for the 32-bit Trigorilla Pro version (also known as v4) of i3 mega

### Feature Workflow

1.Build for i3 mega with trigorilla-pro board in the IDE
2.Flash the firmware into printer with stm32flash, flash loader demonstrator etc, not with cura etc.
3.Reboot printer normally and the firmware work

#### Additional Information

Steps to flash the trigorilla-pro board can be found here:
https://github.com/napyk/trigorilla-pro

DanJunior78, 2020-10-06:
The board is not supported from Marlin firmware...i am not sure if @knutwurst want to dig in this rare board, which is anyway born to die, because you can´t change the drivers (i had the same board in my i3 Mega and i´ve changed it last week to an MKS)

Blue-Beaker, 2020-10-07:
> The board is not supported from Marlin firmware...i am not sure if @knutwurst want to dig in this rare board, which is anyway born to die, because you can´t change the drivers (i had the same board in my i3 Mega and i´ve changed it last week to an MKS)

I have successly compiled marlin 2.0.7 for my i3 mega and worked, just changed some pin defs in the marlin config of trigorilla-pro pinouts.

Blue-Beaker, 2020-10-07:
and this board need stm32flash etc. with jp1 disconnected (bootloader mode) to flash firmware.

knutwurst, 2020-10-21:
@Blue-Beaker, did you manage to get the Touchscreen running? Can you use the original Marlin menu? Can you post some Photos? I am also interested in this and I think that's quite interesting for many people out there :)

TaylorWatson, 2020-10-22:
> The board is not supported from Marlin firmware...i am not sure if @knutwurst want to dig in this rare board, which is anyway born to die, because you can´t change the drivers (i had the same board in my i3 Mega and i´ve changed it last week to an MKS)

Can you link the board you purchased to replace? I also have this board in my Mega-S and want to replace it.

DanJunior78, 2020-10-22:
There are many possibilities. I've placed a MKS SGen L board with TMC 2209, TFT35 and a LCD 12864... TFT makes not too much sense...in my Mega X I'll put a Robin Nano V2.
I've seen an interesting board from Bigtreetech...they have on the TFT the dual mode, mixed LCD 12864 with TFT funktinalities.
So...there is a big choice if you're not afraid of something like that...in the pictures. Finding out, what goes where...lol this[image][image]

dimitarpenev, 2020-10-23:
Hi Guys, I just bought  Mega-S and got Trigorila V0.0.2 board. 
It is Atmega 2560  based. It is 8 bit AVR with 16MHz clock.
I am very new to the 3D printers. 
Can you let me know what benefits I would get if migrate to the 32 bit 100MHz board you have mentioned above?

DanJunior78, 2020-10-23:
It's like with anything else...the benefit related to the calculation speed, if there would be something what needs to get heavily calculated. But mainly I did the modification, because I had the 32bit board with fixed drivers. The printer was loud. I couldn't change to LVs or TMCs.
There is also not a big price difference that you won't buy a 32 bit board as replacement then a 8 bit one.

dimitarpenev, 2020-10-23:
Ok but in terms of printing quality and speed. 
Should I expect any difference (with the current Merlin software) if I upgrade to  MKS SGen L for example
My 8 bit board uses drivers as a separate modules.

DanJunior78, 2020-10-23:
No, you won't,e.g.my Mega X is still running the stock 8 bit board...and there is no difference,because we can't print on high speeds which needs some more processing power.

dimitarpenev, 2020-10-23:
Thank you DanJunior78 for letting me know!

knutwurst, 2020-11-12:
Since the Trigorilla Pro mainboard is more of a niche product in the Mega S, I refrain from installing the support for it, as the effort is much higher than the benefit. Especially since you could and would use the display in "Marlin mode" with real text on the tft.
The Trigorilla Pro is actually used in the Anycubic Predator and there is [already a marlin port for this](https://github.com/SXHXC/Marlin-Anycubic-Predator-Trigorilla-PRO). Actually, this could be rewritten for the Mega S, but that would be a completely new project.

I'm sorry to have to say it, but ... no .. I will not support the 32 bit board with this firmware.

closed 2020-11-12 by knutwurst