knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #27: [FR] Adjustment of Mega X Firmware settings

DanJunior78, 2020-10-06:
[image]
### Description

Modify standard firmware values for the Anycubic Mega X

### Feature Workflow

I´ve done some tests with my Mega X because the original firmware acceleration values have finally reduced the printing quality.
I´ve done several prints with different values.
Now some comments from the picture from print result left to right.
The first print has been with a Stock Mega X with Anycubic PLA which came with the printer (with 40mm/s).
The second print has been done after modifying the printer with your firmware based on TMC2208 and Bltouch (with 30mm/s) with PETG.
The third print has following values: (30 mm/s)
M205 J0.16
Maximum feedrates (units/s):
M203 X300.00 Y300.00 Z18.00 E80.00
Maximum Acceleration (units/s2):
M201 X800.00 Y800.00 Z60.00 E10000.00
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
M204 P1500.00 R3000.00 T1500.00
Edges are now better, but still not straight.

The fourth print has following values: (50 mm/s)
M205 J0.16
Maximum feedrates (units/s):
M203 X400.00 Y400.00 Z18.00 E80.00
Maximum Acceleration (units/s2):
M201 X1500.00 Y1500.00 Z60.00 E10000.00
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
M204 P1500.00 R3000.00 T2000.00

This one isn´t bad at all. But i was wondering where the limit could be.

The fifth print has following values: (50 mm/s)
M205 J0.013
Maximum feedrates (units/s):
M203 X500.00 Y500.00 Z18.00 E80.00
Maximum Acceleration (units/s2):
M201 X2000.00 Y2000.00 Z60.00 E10000.00
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
M204 P2000.00 R3000.00 T2000.00

This one was above the capabilities of the setup. Maybe with higher Vref on the TMC2208 possible. But the layer shift started.

The sixth print has following values: (40 mm/s)
M205 J0.013
Maximum feedrates (units/s):
M203 X450.00 Y450.00 Z18.00 E80.00
Maximum Acceleration (units/s2):
M201 X1800.00 Y1800.00 Z60.00 E10000.00
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
M204 P1800.00 R3000.00 T2000.00

This one is also not too bad. 

I was able to compare to my modded Anycubic i3 Mega S (upgraded M). The black cube on the right, ok, there is only one, lol. I´ve replaced the 32 bit Trigorilla v1.4 board by a MKS SGen L with Bltouch and TMC2209, running in UART mode. I was unhappy with the loud stock board. 
If you check the Firmware values for all other Mega versions i was wondering, why they can at least run on the X axis with values like 3000 mm/s and the Mega X only with 400 mm/s ? So, even an Ender 3 is able to run with higher values and the X axis is mechanically the same.

As you can see, Anycubic did a really bad job on the setup. I have seen you can´t test the Mega X, because you don´t have one. I want to contribute my tests to verify that the printer is able to print better :o) and you may update your firmware, which i really find exceptional (at least i was starting to understand how Marlin is programmed and i could fix my MKS setup).

My friend has also the Mega X and the same trouble. So, i am able to say...that would be a nice update.

DanJunior78, 2020-10-06:
Ähhhmmmm...yes, the heatbed. There might be a bug on the Firmware. I am clarifying it. I was setting up M420 Z0.8 to prevent the software correction of the Mesh above 1mm...and...yes...it scratched little bit my surface...

knutwurst, 2020-10-07:
HI @DanJunior78 !
Thank you very much for your Input!

I'll take a look into it. So as far as I can see, the best/safe values are:
> Maximum feedrates (units/s):
  M203 X400.00 Y400.00 Z18.00 E80.00
  Maximum Acceleration (units/s2):
  M201 X1500.00 Y1500.00 Z60.00 E10000.00
  Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
  M204 P1500.00 R3000.00 T2000.00

Correct? I am currently running a BETA Test. Maybe you can participate? Unfortunately I don't own a Mega X, so I cannot verify these values. I can only assume what's correct and did this by looking into Anycubic's firmware sources.

DanJunior78, 2020-10-07:
Yes,only the junction deviation I would reduce to the once from the i3. Thanks for your great job on the firmware.

Sydney1101, 2020-10-07:
I tried 1500 on my Mega X with the TMC2208 modification and the original voltages adapted to the TMC controllers (not 1:1). My currently limit is less than 1500. My last run without problems was at 1200. As you wrote in the other thread, the voltage could be raised - but the standard values of a firmware should cover the normal environments and not a modification (for experts) ;-)

If I have time to open my printer,I will also make test runs with raised voltage and temperature monitoring,  :-D

knutwurst, 2020-10-13:
@DanJunior78, I had some users test these values and came to the conclusion that the acceleration values you suggested are a bit too high. Presumably they would destroy the printer in the long run, since the Mega X moves a lot of mass.

I'm trying to figure out how to get the edges a little sharper without putting too much stress on the hardware.

DanJunior78, 2020-10-13:
Ok, yes. You can see from the second dice that the edges are also not too bad. The values are much smaller

knutwurst, 2020-10-30:
I fiddled around a bit and came to the conclusion, that I'll not increase the default values. Sorry @DanJunior78 . It's a good Idea, but every printer is different and the tweaks can also be applied via GCODE. Your extremely high values might be perfectly fine to print small cubes, but when printing large objects, the print often fails. Advanced users like you can always override the values to their needs but I don't want do "force" others to use values which might not be safe.

closed 2020-10-30 by knutwurst