knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #28: Updated screenshot of board

HowardEdidin, 2020-10-10:
### Description

I do not see where the x, y, and z endstop connectors are plugged into

### Feature Workflow

1.When adding an external ABL one must unplug the existing end stop connector
2. It is plugged into the ABL  controller 
3. One example is EXABL from TH3D

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

knutwurst, 2020-10-21:
Hi @HowardEdidin ,
I don't think I fully understand your question. Are you missing instructions on where which plug goes? There is tons of video footage from Anycubic itself. Or what exactly are you missing?

See: https://www.youtube.com/watch?v=O-0fdBljgDo

knutwurst, 2020-11-12:
Is this still an issue?

closed 2020-12-13 by knutwurst