knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #29: [BUG] (Octoprint+Filamentsensor doesnt work)

bigfoot9992, 2020-10-14:
Hello, 

the Filamentsensor doesn`t work if you use Octoprint for printing.

Please activate it.

Thanks

knutwurst, 2020-10-15:
The filament sensor is not deactivated, but is controlled differently in Marlin 2.0. It was therefore not so easy to operate it without anycubic touchscreen. But you are not the first with this request and I will implement the feature in the coming days so that the sensor also works via USB.

ChristophDerndorfer, 2021-01-07:
@knutwurst, did you have a chance to work on this topic? Looking at the release notes for 1.1.7 and 1.1.8 I didn't see any references to the filament sensor (unless I missed something).

I just received the plugs used for the filament sensor and my plan for the weekend was to re-wire the sensor to the GPIOs on the Raspberry Pi running OctoPrint (plus one of the filament sensor plugins) - rather than the Mega S motherboard as it is now. But of course having the filament sensor handled by the firmware would be a much more elegant solution.

Plus, I'm not sure how the current firmware handles a disconnected filament sensor, would that be an issue and I'd have to manually disable it via the touchscreen menu for every print? In that case my work-around would have been to try and wire the sensor to both the printer motherboard and the RPi but I have no idea whether that would actually work. :D

knutwurst, 2021-01-08:
Hi @ChristophDerndorfer ,
I'm sorry but the feature is more complex than I expected, so I did not work on this :( 
In the official facebook support group there is one member who also used a free gpio PIN to use the sensor for octoprint. That's a workaround but it prevents the display to know the filament state.

ChristophDerndorfer, 2021-01-08:
@knutwurst, thanks for the update and info. Do you have any idea if the possible work-around I mentioned above (wiring the filament sensor to the printer's motherboard and RPi GPIO) could work?

DunklerPhoenix, 2021-01-18:
I used a two wire / two way switch and cutted the wire of the filament sensor.
So i can choose if it's used by octoprint or by the printer.
(You can also just add another sensor)

But it's not a good idea to try connecting the sensor to pi and printer at the same time. The pi will turn off.

ChristophDerndorfer, 2021-01-19:
@DunklerPhoenix, thanks for the information, I was afraid that it wouldn't work. But is there any downside to the display / printer not being connected to the filament sensor? I've been meaning to rewire mine to the RPi but simply haven't found the time to do it yet.

DunklerPhoenix, 2021-01-20:
Yes there is one downside.
The printer shows a message that the filament sensor is missing (even it's disabled through the special menu).
This message comes at boot and if octoprint connects, but it doesn't block any functions. It's only a bit annoying.

bigfoot9992, 2021-01-27:
its bad that it doesn´t work with marlin 2.0. I don´t understand why but ok...

ChristophDerndorfer, 2021-02-27:
@knutwurst, should I let this issue stay stale so it gets closed or do you see any reasonable chance to tackle this in the foreseeable future?

knutwurst, 2021-03-27:
I'm so sorry. I don't think I will get around to adding this functionality for the time being. It's on my agenda, but I just can't find the time because a lot seems more important.

Please leave the issue open. I take care of it. I just don't know when yet.

ChristophDerndorfer, 2021-03-27:
No worries, I appreciate all the work you put into this project and this issue here isn't a showstopper by any means. Have a great weekend!

frobert, 2021-04-17:
Would be awesome to have that ;)

cyclemat, 2021-04-19:
> I used a two phase / three position switch and cutted the wire of the filament sensor.
> So i can choose if it's used by octoprint or by the printer.
> (You can also just add another sensor)
> 
> But it's not a good idea to try connecting the sensor to pi and printer at the same time. The pi will turn off.
> 
> Example (3phase)
> https://www.amazon.com/Baomain-Universal-Changeover-SZW26-63-Position/dp/B01IZ5ZFYC

never connect the filament sensor to pi and Printer at same you will kill the PI  the Printer mainboard works with 5v and the PI with 3.3v my idea when you want to do this is´use an optocuppler in the printer line and connect the passiv side to the PI GPIO !

DunklerPhoenix, 2021-04-20:
They are not connected at the same time with this switch.

closed 2021-05-29 by github-actions[bot]