knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #30: Layer shift on Mega X with 1.1.7-beta.1

shaddowlink, 2020-10-14:
I recently updated to 1.1.7-beta.1, loaded FW-defaults and saved. I got horrible layer shifts in the Y-axis at unregular intervals (approx. every 3-20 layers), with random amount of shift. Also always at different positions with the same model. (X and Z axis are totally fine)

here is a pic:
[image]
another model:
[image]

In the release notes I saw that this version is supposed to fix this issue, but for me it causes it. 

A downgrade to 1.1.6, loading Fw-defaults fixed this for me and my printer is running again like it is supposed to. 
tried several print settings and models before figuring it out. 

I have TMC2209 installed.

knutwurst, 2020-10-15:
Wow! That's crazy! I haven't seen a Mega X with layer shifts yet. Funny that it occurs with the 1.1.7 for you and not with 1.1.6. There will be a Beta 2 over the weekend. Maybe you would like to try this version as well? I'd be pleased :)

Thank you very much!

shaddowlink, 2020-10-16:
I only print via SD card for now. Plan to update with Octoprint over usb at some point.

Ofc I will try the new beta then.

Is there an option to collect debugging data that would be useful for you?

Sydney1101, 2020-10-19:
I have no problems with the beta 1 version on my Mega X with TMC and Octoprint. It coud be a acceleration problem... What are the values for M201 and M204? Which slicer do you use?

knutwurst, 2020-10-24:
Beta 2 is out now. ;) 

Please open a new Issue if the problem persists and link to this one.

closed 2020-10-24 by knutwurst