knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #31: current master not compilable

nablabla, 2020-10-24:
I got this
"E0_STEP_PIN, E0_DIR_PIN, or E0_ENABLE_PIN not defined for this board."
when i download 15dcb2b and just enabled KNUTWURST_MEGA_S

knutwurst, 2020-10-24:
One minute ago it compiled just fine. See: 
[image]

When did you compile the last time? Which version worked for you?

nablabla, 2020-10-24:
hi, i am searching a version that works with anycubic i3 mega s, bltouch and does UBL correctly for serveral day now. and i am so frustrated. I just found your work. So I never built it before
I downloaded the current master branch 15dcb2b and uncommented KNUTWURST_MEGA_S and this is the precompiler error that i got. It is the only thing that I did, or do I have to do anything else?

knutwurst, 2020-10-24:
I assume that you try to compile the firmware with the Arduino-IDE, correct? That's not possible anymore, because of a huge bunch of build flags which are injected in the build process by PlatformIO.

You should take a look at the WIKI: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki

There is a page where you can see how to set up your PlatformIO Environment to compile it by yourself. The hint also is shown in the sourcecode on every single define ;)

nablabla, 2020-10-24:
oh I see!
you should mention that on the front page or link the wiki, thanks. So I will try your stuff now :)

knutwurst, 2020-10-24:
[image]
Uhm...I did that? ;) 
But I just deleted the "or Arduino-IDE" on the front page to avoid confusion. Thanks! And good luck with compiling!

closed 2020-10-24 by knutwurst

nablabla, 2020-10-24:
so if I build this with platformIO (sorry I am new to this) I select teh env:MEGA_S_TMC_BLT_10 in my case, i dont have to set flags in the configuration.h right?, so they are obsolete?

nablabla, 2020-10-24:
but all the other settings, can i change those? like defining a different thermister or enable or disable endstops?