knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #32: bed leveling not working

nablabla, 2020-10-24:
Hi again, i know i said that before. I did everything as it is described in your wiki.
But still, bed leveling does not work. The result is very obviously not leveled. My bed has hills in the corners. 
The lead screws for z axis do not move at al. I must seek help now. I have tried to get this working for serveral days. not doing anything else than trying to get it working.
In used the env:mega_s_blt_tmc_11, i just changed my bed size to 210 and changed some temp and steps-settings.
compiled and uploaded it. it beeps now differently than before. the bl-toch is doing exactly as it is supposed to do.
i followed these steps:
https://github.com/knutwurst/Marlin-2-0-x-An
[AI3M_testPattern.gcode.txt](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/5432952/AI3M_testPattern.gcode.txt)
ycubic-i3-MEGA-S/wiki/BLTouch-Installation-(deutsch)
and still when printing (with the G28 M420 S1) it is not correcting my craters in my bed.
Please halp me :(
[levelingNotWorking.txt](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/5432954/levelingNotWorking.txt)

knutwurst, 2020-11-12:
Hi @nablabla ... hmm. that's weird. Can you please please upload your Startcode you use? Also it might work if you DISABLE this in your sources:

[image]

KleinWalke, 2020-11-27:
I`m having the same Problem. My trinter uses the leveling but doesn`t use it when it starts the print. 
I`m using M29 in my start code because i move my printer from time to time:

Start code:
G28 X0 Y0
G28 Z0
M501 ; read EEPROM to get Mesh data
M29 ; Auto-leveling
M420 S1 ; enable MMBL
G91
G1 X5.0 Y5.0 F7200 ; nozzle clean position
G1 Z0.1 ; raise nozzle
G1 X5.0 F7200 ; wipe excess filament at the corner
G1 Z0.2 ; raise nozzle further
G1 Y150 E20 F600 ; prime nozzle
G90
G1 Z15 F1000 ; purge 15mm of filament
M107 ; fan off

I tried older firmwares, cleared the eeprom and tried new. All didn`t help.
My printer shows me every testet point in pronterface and everything. But my printer is still ignoring that.
I actually don`t have any ideo how i can access my marlin firmware to change things.
So some help would be really nice.
You can also answer me in german. My english isn`t the very best. :D

bauland, 2020-12-01:
Command for bed leveling is G29, not M29.

KleinWalke, 2020-12-01:
My mistake. I Have G29 in my code. ^^

closed 2020-12-13 by knutwurst