knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #34: [BUG] Mega X pause stuck waiting for commands to finish

frosty5689, 2020-10-29:
### Bug Description

Press pause when printing from SD card the printer is stuck at waiting for commands to finish

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

I just use the prebuilt firmware 1.1.6 foe the Mega X without any new drivers. 

### Steps to Reproduce

1. Start printing a model from SD card
2. Pause print after a bit

**Expected behavior:**
Print head lifts off model and homes x y and print can be resumed again

**Actual behavior:**
Print head lifts off model and homes x y, but printer is stuck at waiting for commands to finish and cant be interacted with. 

#### Additional Information

* I'm using the cura start/stop code provided in the Wiki without tweaks.

knutwurst, 2020-10-30:
Hi @frosty5689 ,
does the issue persist, when you switch to the default firmware? Did you try the 1.1.7 Beta 2?

I know that there is an issue with the Mega X TFT which causes the Touchscreen to freeze, when entering pause state. This is not related to the Marlin firmware but to the TFT firmware.

I'll try to reproduce the issue, but it might be, that you need an Update for the TFT Firmware. .... I'll come back to you when i know more.

frosty5689, 2020-10-30:
I switched back to default firmware for now. It pauses fine without any
issues. One thing to note with 1.1.6, it eventually goes from waiting to
'pause failed'. I haven't tried 1.1.7 beta 2 yet. I can try that if needed.
Thanks!

frosty5689, 2020-12-02:
So, I just had a chance to flash latest 1.1.7 stable that was just released. Still the same problem where pausing the build causes it to hang at "Command has been sent. wait for response." I cleared EEPROM and saved

mopedcrosser, 2021-01-03:
Just flashed 1.1.8 as upgrade from stock firmware on a Mega X. Can confirm, same issue here. Extruder travels to pause position, beeps, but waitig for further command. Printer gets stuck, only way is power off. Using stop works fine though. Just sad u cant pause. Can take a video if needed. Great work so far by the way, thanks a lot !

knutwurst, 2021-01-19:
I'm on it :)

SpaceProper, 2021-02-07:
> I'm on it :)

Hey, wie schauts aus? Gibt es dazu schon einen Fix? Habe das selbe Problem aber will ungern auf die Stock Firmware zurück damit ich die Pause normal wieder nutzen kann.
Danke schonmal.

knutwurst, 2021-02-08:
I was in the hospital for a while and couldn't continue working. I will now take care of the issues again. Please be patient :)

hibikine, 2021-03-15:
I'm using 1.1.9 on Mega X, still confirmed.

knutwurst, 2021-03-27:
The Issue is fixed in v1.2.0, but requires the Update of the TFT firmware. You need to extract the DWIN_SET files to a micro sd card, put it into your TFT (you need to open the printer). Then boot up your printer and wait for the process to finish.
[Mega_X_TFT_firmware.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/6215324/Mega_X_TFT_firmware.zip)

frosty5689, 2021-04-04:
That's great news! This may seem like a dumb question but how do you upgrade the TFT firmware? I'll open it up and take a look, does it need to flash via JTAG or something? Also is there a stock firmware to revert to? Curious where the TFT firmware came from. Wondering if the upgrade will impact stock firmware

EliasKotlyar, 2021-05-27:
Can also confirm that MEGA-X is getting stucked on newest 1.2 Beta 3 Firmware and upgraded TFT Firmware

frosty5689, 2021-05-30:
@knutwurst can this issue be reopened or a new one started? The new TFT firmware update doesn't fix the display getting stuck after pause/stop on the Mega X. I kinda worked around this now that I'm printing directly from OctoPrint, but it makes the firmware unusuable for people that use the SDCARD to print

knutwurst, 2021-07-30:
@frosty5689 , The Issue is still present, I know and sorry for my bot for closing it :/ 
I am still working on a fix but also added the bug to the known issues in the download section.

frosty5689, 2021-08-02:
> @frosty5689 , The Issue is still present, I know and sorry for my bot for closing it :/
> I am still working on a fix but also added the bug to the known issues in the download section.

No worries! This impacts users that print from SD card. I have been using OctoPrint recently so it isn't impacting me. Would be good to have it fixed for others.

JernejL, 2021-09-10:
THIS is stil lrelevant!