knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #38: Are ARCs enabled? | Sind die ARC Funktionen in der Firmware aktiv?

DunklerPhoenix, 2020-11-09:
Heho,
Ich versuche gerade das OctoPrint ArcWelder Plugin zum laufen zu bringen. Sind in der Firmware die ARC Funktionen aktiv?

I'm currently trying to get the OctoPrint ArcWelder plugin to work. Are the ARC functions active in the firmware?

REF: https://github.com/FormerLurker/ArcWelderPlugin/issues/64

DunklerPhoenix, 2020-11-09:
Ah nevermind... 

```
//
// G2/G3 Arc Support
//
#define ARC_SUPPORT                 // Disable this feature to save ~3226 bytes
#if ENABLED(ARC_SUPPORT)
  #define MM_PER_ARC_SEGMENT      0 // (mm) Length (or minimum length) of each arc segment (default: 1)
  //#define ARC_SEGMENTS_PER_R    1 // Max segment length, MM_PER = Min
  #define MIN_ARC_SEGMENTS        0 // Minimum number of segments in a complete circle (default: 24)
  //#define ARC_SEGMENTS_PER_SEC 50 // Use feedrate to choose segment length (with MM_PER_ARC_SEGMENT as the minimum)
  //#define N_ARC_CORRECTION       25 // Number of interpolated segments between corrections
  #define ARC_P_CIRCLES             // Enable the 'P' parameter to specify complete circles
  #define CNC_WORKSPACE_PLANES      // Allow G2/G3 to operate in XY, ZX, or YZ planes
#endif
```

FormerLurker, 2020-11-09:
Not sure what will happen with a mm per arc segment of 0..  Ich habe keine Annung :). Bitte schreib mir an, ob es lauft oder nicht.

DunklerPhoenix, 2020-11-09:
Hmm... 
@knutwurst Did you change the setting MM_PER_ARC_SEGMENT to zero?

knutwurst, 2020-11-12:
Hi :) 
Yes, ARC Support is enabled and I changed the minimum for each segment to 0 because Someone who wanted to use a Laser ran into issues when printing small circles. This way it works because everything can be set up by the software.

Did you try it?

DunklerPhoenix, 2020-11-12:
Yes I tried it with the Octoprint addon ArcWelder https://github.com/FormerLurker/ArcWelderPlugin/issues/64

but somehow it generates garbage. but i think it has nothing to do with the firmware. idk.
maybe my pi is trashy

DunklerPhoenix, 2020-11-24:
@knutwurst Are you sure with the enabled Arc Support?
I get "Unknown command: G2"
[image]

`FIRMWARE_NAME:Marlin 2.0.5.4 (knutwurst, Github) KNUTWURST_AI3M_VERSION:1.1.6 DISTRIBUTION_DATE:2020-09-11 SOURCE_CODE_URL:https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S PROTOCOL_VERSION:1.0 MACHINE_TYPE:Knutwurst's Anycubic i3 MEGA EXTRUDER_COUNT:1 `

FormerLurker, 2020-11-24:
I do recommend that 2.0.6 is merged in.  Then you can change the default arc settings to this:

```
#if ENABLED(ARC_SUPPORT)
  #define MM_PER_ARC_SEGMENT      1 // (mm) Length (or minimum length) of each arc segment (default: 1)
  //#define ARC_SEGMENTS_PER_R    1 // Max segment length, MM_PER = Min
  #define MIN_ARC_SEGMENTS        24 // Minimum number of segments in a complete circle (default: 24)
  //#define ARC_SEGMENTS_PER_SEC 50 // Use feedrate to choose segment length (with MM_PER_ARC_SEGMENT as the minimum)
  #define N_ARC_CORRECTION       25 // Number of interpolated segments between corrections
  #define ARC_P_CIRCLES             // Enable the 'P' parameter to specify complete circles
  //#define CNC_WORKSPACE_PLANES      // Allow G2/G3 to operate in XY, ZX, or YZ planes
#endif
```

That should produce really nice arcs.  CNC_WORKSPACE_PLANES  probably don't need to be enabled for this build at all.    Also, a typical printer won't use P, but a laser engraver strapped to a printer just might :) I left that enabled.

FormerLurker, 2020-11-24:
Also, @DunklerPhoenix, perhaps you downloaded and built this firmware before arcs were enabled?  Maybe they weren't always?  Also, perhaps marlin responds to a naked G2 in V<2.0.6 differently (Thinkyhead was the one who explained how to do this, but maybe he meant only in the new version).  The fact that M115 doesn't report CAP:Arcs:1 does suggest it is not enabled, but I don't know that for a fact.  The firmware checker is a brand new feature.

CTRoulston, 2020-12-01:
Appears to no longer be active. 
Was working in a previous release. G2/G3 returns disconnects Octopi due to "unknown command", M115 doesn't report it and arcs are printed as straight lines.

knutwurst, 2020-12-01:
@DunklerPhoenix , as far as I can see, you youse 1.1.6, in which ARC Support is not enabled. 

Please let me know if [1.1.7 Beta 2](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/tag/1.1.7-beta.2) or [1.1.7 release](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/tag/1.1.7) works for you.

Please also try both versions, since they handle G2/G3 differently.

Thanks in advance!

@CTRoulston which version did you use?

DunklerPhoenix, 2020-12-01:
I tried version 1.1.7R yesterday and here are the images.
https://github.com/FormerLurker/ArcWelderPlugin/issues/64#issuecomment-735998625

FormerLurker is doing a check atm if its the firmware or the gcode.

CTRoulston, 2020-12-01:
Beta 3 did not work through arc-welder. I believe 1.1.6 was working.
Sorry looking through which versions I have in my downloads folder.

CTRoulston, 2020-12-01:
Also thank you so much everyone who has worked both on Marlin and the Mega-X branch :) Amazing to have all these great features available

CTRoulston, 2020-12-01:
Just remembered possibly the biggest difference (other than versions numbers), arc welder/arcs were working with TMC branch, noticing issues on TMC + BLTouch branch.

FormerLurker, 2020-12-01:
I don't see a TMC+BLTouch branch.  Was going to look and see if arcs were disabled to save space for the extra features, but don't know where to look.

CTRoulston, 2020-12-01:
MEGA_X_TMC_BLT_10_v1.1.7.hex

CTRoulston, 2020-12-01:
Previous working version (working as in arc-reactor worked, I never checked for G2/G3 as it was working as expected) was MEGA_X_TMC_v1.1.6.hex

FormerLurker, 2020-12-02:
Well crap.  I don't feel like disassembling the bin, lol.  @DunklerPhoenix's firmware check strongly suggested arcs were enabled.  I did see a mm_per_arc_segment setting of 0 being discussed.  Can't say I can see how that would work though.

CTRoulston, 2020-12-02:
@FormerLurker All good, thanks for your help. At the moment have other issues I'm addressing (signal cable came off heated bed :/) , running without arcs, fingers crossed fixed in next revision.

DunklerPhoenix, 2020-12-05:
I'm compiling the firmware MEGA_X_BLT_11 with MM_PER_ARC_SEGMENT 1 at the moment.
Will give feedback later if it works with arcwelder.

DunklerPhoenix, 2020-12-05:
Can confirm. MM_PER_ARC_SEGMENT 0 breaks the arc function. The first layer looks much better with 1.
I'll post the images tomorrow. need to sleep since 3h xD

DunklerPhoenix, 2020-12-06:
Left -> Right
No Arcs (normal gcode), Arc 0mm, Arc 1mm

[image]
[image]

knutwurst, 2020-12-13:
Wow! Okay... I'll reenable the function in the next release. Thank you for your help! I leave the issue open until it's fixed.

knutwurst, 2020-12-13:
Fixed in this commit https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/commit/08a1ceed8abaf5f357595c5d2f34c90b74dc8897 and will be available in the next release.

closed 2020-12-13 by knutwurst