knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #39: Starke Geräusche bei aktiviertem Meshleveling auf Mega X

KriegstNKeks, 2020-11-14:
Hallo,

ich habe mir auf den Anycubic Mega X die  "MEGA_X_v1.1.7-BETA3.hex" installiert.
Habe das Meshleveling durchgeführt und in Cura im Startcode aktiviert.

Jetzt habe ich nur folgendes Problem. Wenn ich drucke habe ich an mehr oder weniger der selben stelle der Y-Achse ein lautes Rattergeräusch. Dachte zuerst ein Lager des Druckbetts wäre defekt. 
Konnte aber nichts feststellen. Also habe ich das Meshleveling im Startcode deaktiviert und nochmal gedruckt und das Rattergeräusch ist weg.

An was könnte das liegen?
Ich hab den verdacht dass die Achse ständig bremsen und auf die Z-Achse warten muss, weiterfahren , bremsen usw.
Kann das sein?
Das Bett ist schon recht stark geschüsselt.

Oder liegt es an den Standardtreibern und mit TMC2208 ist es weg?

MFG

Dogfroster, 2020-11-16:
Das ist normal, das knarzen kommt davon das der die höhenunterschiede ausgleicht wenn Mesh aktiviert ist.
Also wenn dein Bed zum Beispiel 0.05 tiefer oder höher an einer stelle ist gleich er das beim druck aus in dem er die Z achse minimal anhebt und wieder absenkt.

KriegstNKeks, 2020-11-16:
Werden diese Geräusche mit Tmc2208 geringer?
Ich habe mir direkt bei Fysetc welche bestellt, sind aber noch nicht da. 
Es stört mich irgendwie wenn der Drucker sich anhört als würde er kaputtgehen.

Sydney1101, 2020-11-22:
Grundsätzlich werden die Druckergeräusche leiser wenn Du die TMCs verwendest. Mit den Original-Treiber nervt er einfach nur. Bitte prüfe nach dem Einstellen der Spannung, dass die Motoren im Dauerbetrieb nicht überhitzen. Besonders der vom Extruder.. ;-)

KriegstNKeks, 2020-12-12:
Wollte nur mitteilen dass die Geräusche jetzt weg sind.  Es lag wohl an einer Kombination aus Fehlern.
Erstens war die Z-Achse leicht schräg, scheinbar hatte sich eine der Einstellschrauben bzw. Schalter verstellt.
Die TMC2208 von Fysetc haben natürlich ebenfalls einen riesen Beitrag zur Geräuschreduktion geleistet.
Und die Temperatur der Achsenmotoren wird nur wenige Grad über Raumtemperatur. Der Extruder wurde max. 38Grad warm.
Bei 10mm retract.       Wobei das sich auch geändert hat. Ich weiß nicht ob es an dem veränderten Motortreiber oder an dem neu verbauten Titan-Heatbreak liegt, ich brauch/darf/kann jetzt nur noch 2mm Retract machen. Wenn ich mehr mache habe ich anschließend Löcher im Bauteil.   Man sollte immer nur "eine" Änderung gleichzeitig machen. Bei mir waren es jetzt Heatbreak, Motortreiber und Firmwareupdate von 1.1.7 Beta3 auf  1.1.7.

closed 2020-12-13 by knutwurst