knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #44: [BUG] Extruder stops during long print on Mega X

k7beb, 2020-12-13:
Anycubic Mega X with FYsetc TMC 2208 V1.2 drivers installed in standalone mode running knutwurst 1.1.7. 

Since installation of knutwurst 1.1.7 the extruder will stop moving after a few hours of printing (usually 3-4 hours). Print continues apparently normally except no filament is extruded and extruder motor is cold and not attempting to turn. 

If I stop the print and use the front panel to extrude filament it moves correctly, no blockage or motor/driver overheating apparent. 

Suggest possibly related to this issue: https://github.com/MarlinFirmware/Marlin/issues/14634

Reverted to 1.1.6 and seems to be printing fine now.

knutwurst, 2020-12-13:
I'm looking into it. Thanks for the information!
Please test and compare both versions. 1.1.6 and 1.1.7 because they use different stepper timings.

If something stops working after some time, mostly I assume that the stepper drivers overheat. When using LA, the extruder is working on it's limits and it might be, that the stepper shuts off.

k7beb, 2020-12-13:
After failing three times on 1.1.7 I have successfully run the print using
1.1.6. No other changes. Only the extruder had problems so possibly turning
off LA for the extruder only would solve the problem.

For your reference the print I used for testing is
https://www.thingiverse.com/thing:1698368

Thank you for your attention,
Bruce Boyle

k7beb, 2020-12-13:
Update. Completed one full print - about 11 hours - on 1.1.6 then started another and it quit after about 2 layers. So I'm inclined to agree that it is heat-related, and it seems to be better but not perfect in 1.1.6.

Sydney1101, 2020-12-15:
I also have a Mega X, but my printer works fine with the 1.17. So, it is not a problem of ther version. Normally it should be a temperature problem. In the past, I had also a problem with the temp. sensor and the printer sometimes stops. After I filled thermal paste into the brass sleeve - the problems were gone ;-). Do a PID-calibration after that..

k7beb, 2020-12-15:
The problem isn't with the hotend, it's the extruder that just stops
turning. The motor isn't hot so the presumption is that if this is in fact
a heat problem it's with the driver on the controller board.

k7beb, 2020-12-15:
Good news for you, bad for me. I just tried a separate firmware source and had the same problem. So apparently the issue is with the printer, not the firmware. Closing this issue, sorry for the trouble.

closed 2020-12-15 by k7beb

Sydney1101, 2020-12-15:
Bruce... Your problem should be the nozzle temp. sensor. Normally, when the temp deviation is too big, the printer will do a reset. On the temp. sensor there are some PTFE tubes over the wires... Maybe sometimes the wire will be unprotected and short the circuit. You can verify it very easy, when you are printing via USB... After a reset the communication will be interrupted. I hope this helps.

k7beb, 2020-12-15:
You misunderstood me. I am not getting any temperature errors and am having
no problems with the hotend temperature. The nozzle is not clogging. The
extruder motor is simply stopping turning after a few hours. Not trying to
turn, simply not turning. I can reset the printer and immediately be able
to extrude normally. Nothing to do with hotend temperatures or the hotend
in any way.