knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #45: Filament Sensor "lack of filament or filament monitor abnormal"

localhosthacker, 2020-12-17:
Hi, I flashed the i3-Mega-S (Mid2020) with new display and the TMCs. Everything works great. But the filmament everytime I want to print this Error occurs: "lack of filament or filament monitor abnormal". But it works with the standard and the david ramiro firmware.
This time i disable the sensor temporarly.

My Board is a TriGorilla v0.0.3. I installed the Marlin-Ai3M-v1.4.6-TMC2208 hex file.

Adme84, 2020-12-26:
Same issues with MEGA_S_DGUS_v1.1.7.hex. I've also disabled my filament sensor to proceed...

heizie, 2021-01-05:
same, i think this is a typical eletrical signal error.
my suggestion is, that we can set a time delay (20 ms), in oder to double check the sensor is closed or not.

heizie, 2021-01-20:
===============update================
it is actually the fault of the filament sensor. (pls read a other closed issue)
either you could use a optical sensor instead of this mechanical switch inside,
or modify the firmware and the sonsor structure, in oder to detect a "close", only when the filament is runout.
that means, sensor closed, and filament runout. no more sensor opened, filament runout

knutwurst, 2021-02-04:
Since this is not a bug in the firmware, but a hardware bug, I close the issue. But in the future I will change the firmware so that the sensor no longer reacts so sensitively.

closed 2021-02-04 by knutwurst