knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #48: [BUG] (Under extrusion with mega-s)

bauland, 2021-01-03:
### Bug Description

If I print a model with 1.1.7, just a few pla is extruded along printing. Object is very thin and can't "glu" to bed.

### My Configurations

Mega-S with BLTouch and TMC2209 (X, Y, Z and E).

### Steps to Reproduce

1. [Use 1.1.7 release firmware]
2. [Print a model]

**Expected behavior:** [my object with correct extrusion value]

**Actual behavior:** [Very very thin object]

#### Additional Information

It's ok with 1.1.6 version

jeanfbrito, 2021-01-06:
Do you calibrated the e-steps? https://teachingtechyt.github.io/calibration.html#esteps

bauland, 2021-01-06:
I have done calibration as for 1.1.6 my estep is 404.

knutwurst, 2021-01-08:
@bauland , can you tell my which firmware version you use? The filename would be really helpful. I will come back to you later :)

LFSPille, 2021-01-08:
I have the same problem on my non-upgraded mega-s.
I´ve installed this version:MEGA_1G_v1.1.8.hex

knutwurst, 2021-01-08:
@LFSPille does the problem persist with Firmware 1.1.6?

LFSPille, 2021-01-08:
I don´t know. Only installed the 1.1.8. Should i test the 1.1.6?

knutwurst, 2021-01-08:
Yes?! ;)

LFSPille, 2021-01-08:
:-)okidoki. Dann flashe ich mal neu und berichte dann.
I´ll give it a try:-)

bauland, 2021-01-08:
> 
> 
> @bauland , can you tell my which firmware version you use? The filename would be really helpful. I will come back to you later :)
> 
> EDIT: Did you load the defaults after installing the new firmware? It's essential to do that when switching to >1.1.7.

Firmware is 1.1.8. File name is: MEGA_S_TMC_BLT_10_v1.1.8.hex (I've TMC 2209 and BLTOUCH).
I did load defaults after new firmware.

LFSPille, 2021-01-08:
1.1.6 is working fine after setting the e-steps again

sw4280, 2021-01-11:
Hello, 
i don´t know if it will help, but i have the same issue with my Mega-X with TMC2208 with FW MEGA_X_TMC_v1.1.8.hex.
MEGA_X_TMC_v1.1.6.hex is also working fine for me.

knutwurst, 2021-01-12:
@LFSPille you used the wrong Firmware Version. 1G is ONLY for the first i3 Mega with only one endstop. If you own a Mega S, you need to download the MEGA_S Version. You also don't need to set the Extruder-Steps if you use the correct Firmware. Please see the Readme here: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S

Anyways, I just made some changes regarding the pulse width in the stepper driver management. Give it a try ;)

@LFSPille -> [MEGA_S_v1.1.8.1.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/5800417/MEGA_S_v1.1.8.1.hex.zip)

@bauland -> [MEGA_S_BLT_10_v1.1.8.1.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/5800416/MEGA_S_BLT_10_v1.1.8.1.hex.zip)

@sw4280 -> [MEGA_X_TMC_v1.1.8.1.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/5800418/MEGA_X_TMC_v1.1.8.1.hex.zip)

LFSPille, 2021-01-12:
Thx for the help, man. I´ll try it.:-)

bauland, 2021-01-12:
> @LFSPille you used the wrong Firmware Version. 1G is ONLY for the first i3 Mega with only one endstop. If you own a Mega S, you need to download the MEGA_S Version. You also don't need to set the [...]

Thanks, I will try as soon as possible. Is file MEGA_S_BLT_10_v1.1.8.1.hex.zip for TMC too ? (I use TMC2209).

knutwurst, 2021-01-12:
Oh.. damn! My fault... 
here you go:  [MEGA_S_TMC_BLT_10_v1.1.8.1.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/5801847/MEGA_S_TMC_BLT_10_v1.1.8.1.hex.zip)

bauland, 2021-01-12:
> 
> 
> Oh.. damn! My fault...
> here you go: [MEGA_S_TMC_BLT_10_v1.1.8.1.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/5801847/MEGA_S_TMC_BLT_10_v1.1.8.1.hex.zip)

Don't worry ...

And good news: it is working !!! 👍

LFSPille, 2021-01-12:
My fault too:-)
Now it´s working fine

Philoneous, 2021-01-12:
So this bug was not really existent then? I just flashed my Mega S with TMC2208 drivers to this Firmware, should I wait for an update or is it fine?

bauland, 2021-01-13:
This bug exists but @knutwurst has corrected it (for Mega-S TMC BLT for sure) with version **1.1.8.1** (not yet published).

knutwurst, 2021-01-13:
@Philoneous it seems that the bug can not be reproduced on all printers. If 1.1.7 and 1.1.8 work for you, it's fine. The new version will only be more relaxed in terms of signal timings. So yes. It's a bug (kind of) and it will surely added in the new release.

In the meantime, I can upload specific bugfix versions (.1 suffix), or make an "in between" release.

quattroerik, 2021-01-13:
I am running MEGA_S_TMC_BLT_11_v1.1.8.hex with Trianglelabs Titan stepper and FYTSEC TMC2209 (V3). So far could net see any under-extrusion. (smallest object printed directly on bed was 0.8mm with 0.4mm nozzle)

Philoneous, 2021-01-13:
> @Philoneous it seems that the bug can not be reproduced on all printers. If 1.1.7 and 1.1.8 work for you, it's fine. The new version will only be more relaxed in terms of signal timings. So yes. [...]

Alright, I have printed a couple of things and Have not noticed any major discrepancies do far. Thank you for the quick reply!

knutwurst, 2021-01-13:
Fixed in [Version 1.1.9](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/tag/1.1.9)9

closed 2021-01-13 by knutwurst