knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #49: [FR] laser control for the Mega Pro

hp-pepster, 2021-01-07:
Hi,

are there any plans to work on the laser control of the new Mega Pro? If so, how is the progress? I might be of help as I digged a little bit through the source code from Anycubic for the Mega Pro.

knutwurst, 2021-01-08:
Hi,
I don't see any mega pro sourcecode on github fron anycubic. Do you have a link? If so, I'll do my best to implement the laser functionality asap <3

hp-pepster, 2021-01-08:
It's not on github (yet), but a user posted it on thingiverse. As it's marlin-based, which is GPL3, anycubic has to give out the source. Link to the source: https://www.thingiverse.com/groups/anycubic-i3-mega-pro

The laser-part is very "interesting" implemented. As far as I've seen the relevant parts are in `Marlin_main.cpp` functions:
`void Read_BMP(unsigned char* gray, unsigned int y, unsigned int x)`
`void Laset_print_picture(void)`
`void prepare_laser_print()`

Maybe something in `temperature.cpp` is relevant as well.

I tried to describe the functionality as far as I understood it here: https://drucktipps3d.de/forum/topic/steuerung-laser-anycubic-mega-pro/?part=4#postid-131852 . But honestly, the code gave me an headache. Have a look at the Backup-Folder in the source for example :-) , so I might be completely mistaken.

There are some other code parts regarding the laser. One is for a basic gcode-control `inline void gcode_G6()` and the rest is for the display I think.

As other users mentioned before one has to fake the temp sensor on the nozzle with a resistor, otherwise the printer complains about a strange temperature of 0 °C (see https://drucktipps3d.de/forum/topic/steuerung-laser-anycubic-mega-pro/?part=2#postid-126858 )

Let me know if I can support you. Unfortunately I don't have the Mega Pro yet but I hope to receive it soon.

knutwurst, 2021-01-08:
Wow! Thank you very much! That's probably the dirtiest and hackiest code I've ever seen but it looks very promising!

I'll dig into it and let you know what I found out so far and if I could implement the laser functionality. Sadly I don't have Mega P to test it :/

hp-pepster, 2021-01-08:
I have to thank you :-)

As soon as I got mine I can test your code.

Wirelessless, 2021-01-25:
Hi! 
Thank you for your hard work!
I just got the Mega Pro with the LCD displaying V1.2.9 firmware and I can confirm that using the G6 command with S1-S100 turns the laser on full power and ignores any power setting (various forums suggest otherwise). I used a GCODE sequence that draws 6 different boxes at different power levels (S1-S10-S20-S30-S40-S50) and all cut the paper in the same way,
I am a little hesitant to keep using it like this as it might affect the lifespan of the laser.
I hope to see laser power support for the Mega Pro in a future release of your firmware.
Please let me know if I there is a release I can test!

sims-60, 2021-02-05:
Hi 
@knutwurst  are there any updates about the Laser?

knutwurst, 2021-02-18:
Nope. Sorry but I'll update the thread and the readme when it's done.

0o0will0o0, 2021-03-21:
I believe this issue is still relevant and warrants attention due to this firmware being the best alternative I have found in comparison to anycubic's effort for the Ai3M Pro. I believe if piezoelectric leveling and/or laser control functionality is added, it would add immense value to this printer and I'd happily donate more to the project if an update came for these features

sims-60, 2021-03-27:
hi @knutwurst  are there any updates about the laser?

sims-60, 2021-03-27:
and I have a mega pro i can test the firmware for you when you want

sims-60, 2021-03-27:
and in the firmware i have seen interesting functions there are all in the Marlin_main.cpp:
inline void gcode_G6() over the g6 command you can control the laser if its on or off

void Read_BMP(unsigned  char *gray,unsigned int y,unsigned int x) for reading the images
void Laset_print_picture(void)
void prepare_laser_print()
void laser_indicate()
void laser_init()
and in the void loop is something with the laser
i hope i can help you
@knutwurst

sims-60, 2021-03-27:
search with strg+f laser and you find also some variables

knutwurst, 2021-03-27:
HI @sims-60 , thank you very much, but I already have the original MEGA Pro firmware laying around ;) 
Take a look at my current sourcecode. The Laser is already prepared ;)

sims-60, 2021-03-27:
ok

sims-60, 2021-03-27:
when you have a beta you can compile for me the bl touch tmc _10 version and then i can test it for you @knutwurst

knutwurst, 2021-03-27:
Thanks a lot. I'll come back to you when I have any updates :)

0o0will0o0, 2021-03-27:
Is there any way to get the piezo leveling to function like the original firmware?

sims-60, 2021-03-28:
I have never used the piezo leveling before i have used the knitwurst firmware

hp-pepster, 2021-04-22:
Hi, my printer also arrived (I guess). As soon as I'm home again I could start testing. @knutwurst Are you planing to work on the laser any time soon? If you are interested I could have a look at you commits regarding the laser and help with some coding.

sims-60, 2021-05-24:
are there any updates @knutwurst ?

sims-60, 2021-05-28:
?

sims-60, 2021-05-28:
I want that the issue is open and not closed soon xD

knutwurst, 2021-06-29:
The Chiron Support is finally done and now I'll take a look at the Mega Pro Laser ;)

InoxDragoon, 2021-07-11:
FYI, if you have a Mega Pro S, the P is the same, just comes with the Laser. you can get the laser from Anycubic's website to "upgrade an S to a P.

I'm also interested in getting the laser to work on Marlin 2.x

I'm planning on upgrading to your firmware soon, so please keep up the good work!

sims-60, 2021-07-11:
ok coll with the laser i've written with the anycubic support about the laser in their shop xD

sims-60, 2021-07-11:
its possible thats its a other laser thats not so strong

sims-60, 2021-07-11:
because my anycubic laser is class v and the in the anycubic shop ist class IIIb

knutwurst, 2021-08-16:
Reopened, because it's not done yet ;)

aman-tiwari, 2021-08-17:
> Hi!
> Thank you for your hard work!
> I just got the Mega Pro with the LCD displaying V1.2.9 firmware and I can confirm that using the G6 command with S1-S100 turns the laser on full power and ignores [...]

It could be that one of the fan PWM commands is used to control the laser: https://www.youtube.com/watch?v=1cyZCpfglAg, https://marlinfw.org/docs/configuration/laser_spindle.html

mlewis-everley, 2021-08-18:
It's probably not of any help, but I noticed that some commands in marlin support a laser as well (eg: https://marlinfw.org/docs/gcode/M003.html)

I also noticed some other standard commands seem to also support a laser as well.

Just wondering if this is how laser support is intended to be added, or are we still going to have to do the weird BMP thing?

knutwurst, 2021-09-17:
Unfortunately, since I don't have a laser, I can't test the Marlin features.

Actually, I wanted to recreate exactly the anycubic feature so that the original display would continue to work.

That's why I recreated the auto leveling for the Chiron instead of just using Marlin features.

stklcode, 2021-10-02:
I do own a Mega Pro and tried a first step towards the laser.

The wiring is (mis)using the hotend heater pins (A1, A2) for the laser power.

Setting a target temperature results in _Error:MINTEMP triggered, system stopped! Heater_ID: 0_

Simulating an arbitrary temperature with a resistor between A6/7 (see linked post above) enables laser control with `M104 S0` (target temperature 0 = off) and a value greater current temperature enabled the PWM output.
In my case, the simulated temperature is 37.24°C, so with ` M104 S38` it starts with ~4Hz and ramps up with higher values. `S37` or below makes the laser stop.

So with real ugly G-code sequences I was able to do some laser engraving in stock KW 1.3.1 firmware.

That's exactly what's represented in the disabled code fragments, e.g. `WRITE(HEATER_0_PIN, 0)`.

----

There is obviously some work to do for the code fragments. Reordering and some declarations eventually makes it compiling, but not working.

```cpp
#if ENABLED(KNUTWURST_MEGA_P)
  #define LASER_FEATURE
  #define SPINDLE_LASER_PWM_PIN HEATER_0_PIN
#endif
```
results in failing sanity check
> Marlin/src/HAL/shared/../../inc/SanityCheck.h:2768:8: error: #error "SPINDLE_LASER_PWM_PIN conflicts with E0_AUTO_FAN_PIN."

Seems obvious, as the pin _is_ used for both functions...

Are there any (unpublished) updates on the laser code fragments to build on or do I have to start off from the current _master_ branch to - maybe - get it working?
(doen't seem to be too complex, but I've never worked with Marlin code til an hour ago)

knutwurst, 2021-11-02:
Rudimentary Laser Support added in [Release 1.4.1 Beta 1](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases)

Thank you so much @stklcode !

stklcode, 2021-11-02:
Is "rudimentary" really the correct term here?

Fair to say, the Anycubic features are not yet supported. This requires some work in the TFT routine, so you can't "print" bitmaps from SD.

But unlike the (misused) G6 command of the stock FW the laser is dimmable with M3. (at least on my printer 1-255 were all the same intensity, so basically it's on/off). Good enough for use with lightburn or similar software in the current state.

So, it's a different implementation.

knutwurst, 2021-11-02:
> Is "rudimentary" really the correct term here?

You're right. Your work makes it possible to use the laser in the first place. Rudimentary was more related to the entire anycubic-related implementation. In the case of the Chiron, I called "can print and level, but the display doesn't work yet" also "rudimentary". 

That should in no way reduce the value of your work and I would like to buy you a coffee ;)

stklcode, 2021-11-02:
I am not offended in any way, should have places a 😉 maybe.

Let’s see if we can get the not so straight-forward part for the built-in feature. I’ve seen the code snippets, but focused on the Marlin part to get any light out of it first place.

We should drink the coffee together. Likely never have started playing with the code without your prior work.

knutwurst, 2021-11-02:
The marlin part is a pretty good start and when everything else works, connecting the display shouldn't be a problem. Maybe wen can reuse the anycubic bitmap parsing code to feed the marlin implementation.

stklcode, 2021-11-03:
Started working on the integration based on the 1.2.9 sources that have been partially ported already: [stklcode@`feature/anycubic-laser`](https://github.com/stklcode/Marlin-2-0-x-Anycubic-i3-MEGA-S/tree/feature/anycubic-laser)

It compiles, but it completely untested so far. But no need to start from scratch, if anyone wants to join development.

closed 2021-12-09 by github-actions[bot]