knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #50: question for trigorilla 8bit v0-0-2 board / not seeing any

TKaluza, 2021-01-08:
Bought Anycubig Mega S from "official" store in Germany. Updated drivers to TMC 2208.
Board is 8bit version v0-0-02. I've picked release "MEGA_S_TMC_v1.1.8.hex" and flashed with Cura.

Sound on start up is now completly different. UI stays the same, but not seeing any "hidden" menu, with or without a SD Card. 
Can connect to Board with G-Code Software over UART. 
New files are not found on SD Card within the menu from the printer.  

Could you perhaps give me some tip/idea what to do?
Thank you very much!

TKaluza, 2021-01-09:
If this questions is inapropiate here 😭 , please tell me or close and I would try to register at facebook and ask you there 😄

knutwurst, 2021-01-09:
Hi there,
if you do not plan to install a BLTouch auto leveling sensor or if you have not installed one, it makes no difference which mainboard you have, as long as it is the Trigorilla 8 bit board. The only difference between version 1.0 and 1.1 is the pin assignment of the servo port, which is relevant for the sensor. If you look in the wiki, you will also see the appropriate instructions and also the FAQ, which answers your question;)

If you don't see a special menu, I'm assuming that you haven't read the readme: P No offense, but that just happens the most and I think I'll write new installation instructions for newbies.

If your display works with blue icons on a white background, then the version is correct, but I suspect your display has yellow text on a blue background, right? Then you have a DGUS Clone Display, for which you need the _DGUS version.

best regards
Olli

TKaluza, 2021-01-09:
@knutwurst 
You are totaly right. I'm sorry for not beeing able to rtfm 
I read it, but didn't understand on the spot. 
Would it be appreciated, if I try to structure the wiki a bit? 

All the best and thank you very much!

May I ask a another question or should I open onther report?
I ordered a BLTouch, but on my board theres 0.0.02 printed on, please see attached picture. Do you know which version I the appropiate is, or how I may determine the correct version? 

Thank you very much!

k7beb, 2021-01-13:
Same here. I have a Mega X which uses the 0.0.2 version board. I'd like to install a BL Touch but can't tell what connections to use.

Blitz88, 2021-01-16:
I have the same question, also a new Mega X...

v-a-d-e-r, 2021-01-19:
I'm also thinking about a BL Touch. But what is Trigorilla_14 v1.0/1.1? I have v0.0.2 too. What firmware to choose now?

knutwurst, 2021-01-19:
0.0.2 has nothing to do with the internal wiring. There are three different Trigorilla Board out there.

- 1.0 which is sometimes labeled 0.0.1 or only Trigorilla
- 1.0 which has new board connectors and has 0.0.2 printed on
- 1.1 which has new board connectors and also 0.0.2 printed on
- 1.2 which is used in the new Mega S and Chiron ... It does not have the two ext ports but dedicated Mega and Chiron ports and 0.0.3 is printed on the PCB. This one is currently not supported, but I do my best to get the new pin layout and update the firmware.

Ohne only difference between 1.0 and 1.1 are the servo pins.
Afaik if the pre-soldered Servo-Pins are on "S0" you got an 1.0 Board. If the Pins are on "S1", you got an 1.1 Board.

Blitz88, 2021-01-19:
+Thanks for the clarification

closed 2021-01-21 by knutwurst