knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #52: [BUG] Mega S prints off of bed

Philoneous, 2021-01-13:
### Bug Description

I tried to print a bed leveling test with a total size of 210x210mm, I have already changed the bed size in Cura according to the Wiki. There appears to be an offset as the print was ~10mm away from the front of the printbed, resulting in the back of the print not being on the printbed. I am using the Starting G-Code provided in the Wiki

### My Configurations

Your Precompiled MEGA_S_TMC_v1.1.8.hex

### Steps to Reproduce

1. Print 210x210mm object
2. Wait
3. Observe that Print is not on Buildplate :(

**Expected behavior:** Print fully on Buildplate

**Actual behavior:** Print seems to be offset in the Y direction

#### Additional Information

[image]
[image]

Philoneous, 2021-01-13:
Here is a picture of the hotend at z0:
[image]

frobert, 2021-01-15:
Same here, but by my mistake, loaded wrong firmware :/
Maybe you did the same :)

knutwurst, 2021-01-15:
A Y-offset is theoretically not possible because the zero position is specified by the limit switch. In some cases, however, the limit switch triggers too early or too late. Then the Y-axis shifts and you cannot print on the whole bed.

Anycubic built a tolerance in that the size was severely limited. I only adopted the maximum possible mechanical values in the firmware. However, this assumes that the printer is set correctly.

Check the Y limit switch. Maybe it's not sitting correctly.

quattroerik, 2021-01-15:
I have the same problem, but additionally with a self designed print head.
Using Home Offset solved the problem for me.
https://marlinfw.org/docs/gcode/M206.html
Don't forget to add G0 X0 Y0 after homing.

closed 2021-02-21 by github-actions[bot]