knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #54: Mesh Bed Leveling not activating correctly?

cylophis, 2021-01-16:
Using your firmware for a while and after 2 new AI3MS arrived with an uneven buildplate i gave MBL a shot.

After leveling and calibrating the mesh, with a EEPROM Save and restart i am lost.

I use your Start and End Code, but every print starts way over the build plate.

I put in 
M501
M420 S1
but no change

You are using a line "G0 Z0.15                                   ; lift nozzle a bit" and i thought i found the problem, but even after removing it the prints are horrible.

i read every piece of information i could find about MBL, but i am not able to archive an kind of comparable print to normal leveling.

If i just use 4 points leveling everything is fine except the difference of almost 0,14mm from the Middle to the side of the buildplate.

I am sorry in advance if this is not a bug and i am just not smart enough to do it

mrskizzex, 2021-01-31:
I'm having similar issue to this. After having done the mesh leveling and saving to eeprom and when using the provided startup GCODE, on command G0 Z0.15 ; lift nozzle a bit it doesn't lift the nozzle but actually crushes the bed so hard that the motor skips steps because it bottoms out the springs. I run the precode manually and experimented with different G0 Z values and it seems that my nozzle is just slightly dragging on the paper on Z2.35. But even after changing that and finishing the nozzle wipe, on starting actual print the nozzle just crushes the bed again and drags over it.

I don't know what the problem is as when i was using mesh leveling on https://github.com/davidramiro/Marlin-Ai3M firmware it was working without issues as this. Any help appreciated.

gloomytree, 2021-02-03:
@cylophis What method do you use to generate you mesh? Automatic (with a sensor, for instance BLTouch) or manually? What is the output of M420 V? Do you get a matrix that seems valid (you can visualize the matrix here: http://lokspace.eu/3d-printer-auto-bed-leveling-mesh-visualizer/)? 

If you do not have a BLTouch, you can generate the mesh manually, like described here https://marlinfw.org/docs/gcode/G029-mbl.html did you do that? 

M420 S1 will only work if there is a valid mesh matrix in the eeprom. 

@mrskizzex Can you provide your full starting G-Code and the output of M420 V?

Selfmade-RuLeZ, 2021-02-09:
I've the same problem. The mesh grid is not loading right. The bend bed came from a collision of the nozzle with the bed itself, have to level it again. But it doesn't metter, the mesh grid is still not activated. 
Installed BLTouch, it's working as expected. Just the firmware is not using the mesh grid. Tried with the start-gcode in this repo and trying manual with octoprint. Trying older releases for now.  

```
Send: M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +1.092 +0.940 +0.777 +0.632 +0.490
Recv:  1 +1.195 +1.002 +0.843 +0.690 +0.530
Recv:  2 +1.245 +1.042 +0.863 +0.697 +0.520
Recv:  3 +1.310 +1.070 +0.863 +0.670 +0.470
Recv:  4 +1.322 +1.055 +0.827 +0.607 +0.342
Recv: 
```

gloomytree, 2021-02-14:
Well, if M420 V does output the mesh, it IS in the firmware. What happens if you save to eprom (M500), turn off the printer and do the M420 V again? Are you aware that the firmware intentionally turns off the mesh after homing (G28)? You need to put in G29 after all G28 occurences in the starting gcode...

Selfmade-RuLeZ, 2021-02-14:
You dont need G29, just the M420 S1 after G28 to load it from the firmware as mentioned in the wiki. 
Mesh was available after restart with M420 S1 in the Terminal, it‘s just not used while printing. 
Anyways, I switched to Klipper. 
Could be also a Marlin issue or a user issue at all.

closed 2021-03-22 by github-actions[bot]