knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #55: [BUG] Printers stops extruding but still "prints"

SHAXperiment, 2021-01-17:
### Bug Description

The printer prints but suddenly the extruder stops to extrude. The printer still prints and when i stop the printer and restart the printer prints normally. This happens from time to time, not always. I checked everything and can't find a problem. The prints are perfect when the error doesn't show up.

### My Configurations

I'm using the TMC version of the last release. No custom configuration, only Extruder steps changed because of under extrusion

### Steps to Reproduce

I put the gcode on the sd card and print. I also tried other sd cards.

#### Additional Information

Here's a video that shows what happens:
http://www.shax.org/drucker/mega-s/bug.mp4

linuxq, 2021-01-17:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;}                                                                                                                                               Gesendet über BlackBerry Hub+ Posteingang für Android                                                                                                                                                                           Von: notifications@github.comGesendet: 17. Januar 2021 16:06An: Marlin-2-0-x-Anycubic-i3-MEGA-S@noreply.github.comAntworten: reply@reply.github.comCc: subscribed@noreply.github.comBetreff: [knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S] [BUG] Printers stops extruding but still "prints" (#55)  
Bug Description
The printer prints but suddenly the extruder stops to extrude. The printer still prints and when i stop the printer and restart the printer prints normally. This happens from time to time, not always. I checked everything and can't find a problem. The prints are perfect when the error doesn't show up.
My Configurations
I'm using the TMC version of the last release. No custom configuration, only Extruder steps changed because of under extrusion
Steps to Reproduce
I put the gcode on the sd card and print. I also tried other sd cards.
Additional Information
Here's a video that shows what happens:
http://www.shax.org/drucker/mega-s/bug.mp4

SHAXperiment, 2021-01-18:
Upgraded to E3D Hemera Extruder. Problem solved.

closed 2021-01-18 by SHAXperiment