knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #57: [BUG] Printing from Octoprint causes stutter

3dRikal, 2021-01-18:
### Bug Description

While printing via USB from Octoprint, especially on small circles, the printer will stutter and blob. If print is sent to SD card and processed via card, very little stuttering occurs.

### My Configurations
I3 Mega S with DBUS. On Marlin 2.0. Configs for standard stepper drivers from this site.

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
I only have the HEX file.

### Steps to Reproduce

1. Slice the retraction tower STL
2. Print from SD CARD, No issues
3. send same file to Octoprint
4. Print from Octoprint

**Expected behavior:** [What you expect to happen]
Print should be as smooth
**Actual behavior:** [What actually happens]
Stutter and difficulty to render small movements

#### Additional Information
A quick search on the internet points to a low setting on the BLOCK_BUFFER_SIZE in the Configuration_adv.h file. 
http://lokspace.eu/bad-print-quality-with-usb-or-octoprint-the-solution-is-here/

Seems that the SD Card is set properly but for some reason, the USB IO buffer is too low causing the printer to starve information.

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Nico205, 2021-01-19:
Hey,

it might not be the answer to this question, however the SD card of the Pi might be broken / too slow.
Could you try to use another SD card inside it?

Friendly Regards
Nico

3dRikal, 2021-01-19:
Got a class 3 32gb card in the Pi so, it's difficult to find a faster card. Pi is smooth running Octopi. Also, if i print mostly square parts (like psu shroud for printer) there is no problem processing the print. Only occurs when parts require a lot of movement processing which octoprint performs ok (and even arc welder plugin reprocess in littlentime) but when pushing to printer, the jerkiness starts when the small circles are pushed to printer.

Also, using 250000 baud, i'm on the fastest rate available. Really smells like buffer underflow.

Nico205, 2021-01-19:
Okay, fair enough.
Does this problem also occur if you print the file directly using USB from your PC? (if this is possible for you)

knutwurst, 2021-01-19:
You surely mean Class U3? 

Also the usb buffer connection is already highly optimized since 1.1.7. Just increasing the buffer does not work, when the data which should fill the buffer, is transmitted too slow. Also there is no room for a larger buffer, since everything else regarding the tft takes a lot of ram.

3dRikal, 2021-01-19:
Yes. I tried from PC (Ryzen 3 3700x, 32GB 3600, SSD 960 EVO) to push to printer and it also stutters. If I push the print to SD card (from PC or from Octoprint) and print via printer menu from SD, no issues (aside from some CURA flow settings I need to take care of) but the stutter is not present

3dRikal, 2021-01-19:
> You surely mean Class U3?
>
> Also the usb buffer connection is already highly optimized since 1.1.7. Just increasing the buffer does not work, when the data which should fill the buffer, is [...]

Yes, U3. Sorry i was replying from my phone :).

knutwurst, 2021-01-19:
Which Pi Version do you use? The oder ones are known to be slow a.f. with the current Octoprint version...

3dRikal, 2021-01-19:
Pi 3b+ here's a couple of shots from the same part. Pic 1, octoprint. Pic 2 SD card

[image]

[image]

quattroerik, 2021-01-19:
Did you try a different USB cable? The stock one from my Mega S broke after a few weeks...

Nico205, 2021-01-19:
These are my prints using the firmware + OctoPrint on an 3B+.
However, I've got an Anycubic i3 Mega (without S / X).

[image]
[image]

3dRikal, 2021-01-19:
printing with a new cable at the moment (finding one of these was not easy in my cable pile) but I am having the exact same behavior. Jittery, jerky printer head motion. Not thinking it's the cable. At this point, I will also open a bug at octopi... this is really bizarre.

hawkeyexp, 2021-01-23:
I'm using also a 3B+ and it works like a charm while printing.
The essental point is the sd card in most of all cases - since long times i only use transcend cards with high performance: Type **TS32GUSD300S** and it is a big difference to normal cards

I also tuned a bit the settings for 3B+ in config.txt which schould also work perfect for every 3B+ (i use this settings with multiple boards)

```
#RPi3b+
arm_freq=1400
gpu_freq=400
core_freq=460
sdram_freq=560
over_voltage=2
over_voltage_sdram=3
sdram_schmoo=0x02000020
force_turbo=1
boot_delay=1

current_limit_override=1
max_usb_current=1
avoid_warnings=2
avoid_pwm_pll=1
start_x=1
gpu_mem=128

```

3dRikal, 2021-01-23:
I've been using a Sandisk Extreme 16GB [https://www.amazon.ca/SanDisk-Extreme-microSDHC-UHS-I-SDSQXAF-032G-GN6MA/dp/B06XWMQ81P/ref=sr_1_9?dchild=1&keywords=16gb+microsd+sandisk&qid=1611417750&refinements=p_89%3ASanDisk&rnid=7590290011&s=electronics&sr=1-9](url) and no go. This is the fastest card I have although I do have a 64GB of the same type .. I wonder if I should try the larger card.

hawkeyexp, 2021-01-24:
it has nothing to do with the size - it is only a sample of my used cards which are running without any delays / errors :-)

3dRikal, 2021-01-24:
I understand that the size has no bearing. However some cards perform better than others in the same brand/spec space from my experience. The 64gb might have an edge as far as speed as it is a newer card but only a sd card speed test will tell me. As I have some prints that must be done, i reverted to sd card printing for now but i will investigate this week

3dRikal, 2021-02-04:
Closing issue since I cannot reproduce it after changing stepper drivers

closed 2021-02-04 by 3dRikal