knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #58: [BUG] (short description)

DunklerPhoenix, 2021-01-18:
### Bug Description

Printer firmware stuck if specific functions on display are used.

### Steps to Reproduce

1. Set bed temperature to 65 via display (Setup -> Temperature)
2. While heating up go to filament functions
3. Use one of the both filament functions (Feed/Remove) [message: prevent cold extrusion]
4. Printer stops heating (led of bed turn off), temperature in display gets stucked.
5. Hardreset via power switch (softreset wont work)

**Expected behavior:** printer should heat bed and nozzle at the same time

**Actual behavior:** printer doesn't react on any command

wes1993, 2021-01-30:
Hello,
have you found a solution for this problem, and also you have the same problem when start the print???

DunklerPhoenix, 2021-01-30:
nope no solution at the moment, but for me it's not critical maybe a bit annoying.

I don't have this while starting a print via octoprint. I don't use sdcard printing

wes1993, 2021-01-31:
Thanks for your reply @DunklerPhoenix, I will try to start a print from Octoprint and check!! :)

Oilpig, 2021-02-08:
I’m have same issue , I had a nozzle
Clog and after changing it out and using the on screen controls to reload filament I’ve noticed, several times that after it feeds filament in and I push the stop button couple times to start cooling process it seems to just cut out and keep nozzle hot. Hard reset seems the only way to fix it , also causes octoprint to drop the connection,
This has also happened when trying to perform same steps thru octoprint ,

wes1993, 2021-02-09:
Hello!! :)
Finally I have solved the problem!!
See here:

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/71

Best Regards
Stefano

closed 2021-03-17 by github-actions[bot]