knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #59: T

sims-60, 2021-01-20:
### Description

### Feature Workflow

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

maikkoe, 2021-01-20:
Moin,

dort sollte der Anschluss sein:
[image]

knutwurst, 2021-01-20:
@sims-60 schau mal in's Wiki. Dort findest du auch eine Anleitung mit Bildern :) 

Lediglich das v1.2 (0.0.3) Board wird noch nicht unterstützt. Ich arbeite aber dran.

sims-60, 2021-01-20:
also kann ich es auf den selben stecker stecken und muss nur die firmware mit _11 draufladen?

maikkoe, 2021-01-20:
Wichtig ist, dass du darauf achtest die Pins richtig zu belegen. Ich hab das 0.0.2 und habe die _10 installiert. Hat alles wunderbar geklappt wie in der Anleitung beschrieben, habe den Sensor selbst erst jetzt am Wochenende nachgerüstet.

sims-60, 2021-01-20:
bei mit ist die steckerleiste auf s1 festgelötet soll ich dann _10 oder _11 nehmen

maikkoe, 2021-01-20:
Wie du auf meinem Bild siehst, ist sie bei mir ebenfalls dort verlötet. Die _10 sollte da die richtige sein, ansonsten wenn das nicht geht, kannst du immer noch die 11er flashen

sims-60, 2021-01-20:
und soll ich dann die pins wie in der anleitung vertauschen

sims-60, 2021-01-20:
dachte nur das wenn ich die falsche firmware drauf flashe oder so das ich mit dann den bl touch schrotten kann

maikkoe, 2021-01-20:
Das kann ich dir so leider aus dem Kopf nicht sagen, ich meine ich musste es bei mir machen. Auf dem Board ist aber Links eine Beschriftung was welcher Pin ist. Bei mir, von oben nach unten S/-/+

sims-60, 2021-01-20:
danke bei mir ist es auch s - +

maikkoe, 2021-01-20:
Kontrolliere es bitte dennoch anhand deiner Verkabelung/Sensor.

sims-60, 2021-01-20:
iich mache erst jetzt die verkabelung xD

sims-60, 2021-01-20:
danke funkttioniert jetzt alles habe auch die _10 firmware

closed 2021-01-21 by knutwurst

kallejoken, 2021-02-18:
Ich möchte auch gerne den bltouch nachrüsten, nun ist auf dem Mainboard v1.3 aufgedruckt, kann ich diesen nun nicht verbauen?
[image]
[image]

sims-60, 2021-02-19:
hi

sollte denke ich gehen aber su sicht links nebem der steckerleiste +-s und am sensor ist 
weiß / orange = Signal  
schwarz / braun = Masse/Ground
rot = +5V 
muss schauen dass es extat so dann auf dem board ist