knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #62: [BUG] (possible) BLTouch behavior after system start or after first leveling

hawkeyexp, 2021-01-22:
### Bug Description

Sometimes during startup the BLTouch doesn't test it's pin (in/out).
If this has happened and you start autoleveling it will result in pin doesn't coming out and BLTouch starts to blink resulting in an error.
Restart didn't solve the problem.
But: If system is in this behavior and pin is soft manually pushed out and followed by a restart all works fine again!
System does startup BLTouch does test it's pin and autoleveling works without any problems.

### My Configurations

Precompiled firmware MEGA_S_DGUS_TMC_BLT_11_v1.1.9.hex Board is v0.0.3 / S1 is used.

Is this behavior known in any way or possible a known solution available?
If not i will add a video.

Thx

this-snow, 2021-01-22:
Is your BLTouch wiring okay? Is it an genuine product? I could not reproduce this error. The pin self-tests on every start.

hawkeyexp, 2021-01-22:
Hi, yes it's a original one. And as usual i tried to catch the behavior on a video and it isn't shown :-( I try to catch it this evening again (und ich werd ihn schon noch erwischen :-))

https://photos.app.goo.gl/bAnKPX2WF31nQij87

About wiring: should fin fine cause first attempts shows a fine working process...

https://photos.app.goo.gl/A5jFzr5XskybAZG39

PS: wiring is hard soldered excluding the connection on the board - this is a usual pin header connector.

hawkeyexp, 2021-01-22:
Here we go i catched it :-):

https://photos.app.goo.gl/HYv1i4B6Tk73DvA77

I have reproduced all inital steps and started from plain with releveling the sensor like in your gide.
The power on is after the sensor calibration as succesfull done.
I looks to me like after the inital process the state of sensor is wrong compared to requested state inside the firmware and results in this error.

hawkeyexp, 2021-01-25:
I have checked some pages etc. in the www and compared the knutwurst configuration.h to the template from manufacturer - i think it can be possible a problem with my edition of bltouch (v3.1) ? To be sure sensor is ok i ordered a new one today.

hawkeyexp, 2021-01-27:
Problem is solved - i replaced the bltouch with new one (both original) and now it works like a charm!

Seems it was doa :-(

closed 2021-01-27 by hawkeyexp