knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #65: [FR] Support for 4Max Pro 2.0 (and original 4MP)

alfrankgit, 2021-01-24:
Hi! I really appreciate the effort you put into the Anycubic Mega Firmwares, since Anycubic doesn't do anything.

How about supporting the 4Max Pro (2.0 and original version) also? They have the same boards and displays.

For the 4Max Pro 2.0 I already patched your V1.1.9 in the Mega P TMC sections, see the attached Configuration.h and Configuration_adv.h. It has a BMG clone extruder, TMCs for X and Y, other thermistors.

But I am no programmer so I can't add a new environment to your project. But that would be nice to have, wouldn't it?

I also have three of the original 4Max Pro but none still in the original configuration, so I'm not sure what that was. IIRC it doesn't have TMCs and it definitely has the original DWIN display. But it has the Titan clone extruder. And hotend thermistor type 11. But I can test a configuration for that one too, if you can create an env for it.

PS: My native language is German.

[4MP_2.0_KW1.1.9.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/5861850/4MP_2.0_KW1.1.9.zip)

knutwurst, 2021-02-06:
Hello,
so since the hardware seems to be the same, that wouldn't be a problem and I think it's great that you have already done the preparatory work :)

But I'm currently working on support for the Chiron - and by that I also mean support for all display functions, including leveling. There is also very strong demand for support for the Mega Pro Laser.

I'll see how quickly it goes with the 4MAX Pro, but I can't promise that I can take care of it right away. But I'll write it on my agenda :)

alfrankgit, 2021-02-08:
Hi,
and thanks a lot for your reply and infos.

As a moderator on drucktipps3d.de I'd gladly welcome you to our forum where I already recommended your excellent firmware quite a few times.

knutwurst, 2021-10-30:
I've included the 4MAX pro 2.0 build environment for stock and TMC. I actually don't know if all motor directions are correct, so please give me some feedback, what's not working correctly
[KW-131-4MAXPRO2.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/7446174/KW-131-4MAXPRO2.zip)
.

alfrankgit, 2021-10-30:
Thanks for including the 4Max Pro 2.0 now, but I can't test it myself anymore because I don't own the 4Max Pro 2.0 anymore. But there are users on drucktipps3d.de who do and are aware of your firmware and my "hacks".
https://drucktipps3d.de/forum/topic/anycubic-4max-pro-2-0/?part=23#postid-180447

knutwurst, 2021-10-30:
@alfrankgit Maybe you can shed light on the darkness. I saw in your config that you are using the Build Environment of the MEGA_P and activate both MEGA_P, TMC and DGUS2 for it.

Are these settings for the stick 4MAX or for one in which all drivers have been substituted for TMC2208?

If you built MEGA_P_TMC for the 4MAX stock, I had a mistake in thinking and maybe the directions of the two versions (TMC and non-TMC) are now reversed  :D

I'm looking forward for some "testers" ;)

alfrankgit, 2021-10-30:
I didn't change any physical stepper drivers on the 4Max Pro 2.0.
All my changes in the configuration files are marked with //AFR, either as a new comment to disable a line or as a comment with the original content of the line followed by a line tight the changes I made.

knutwurst, 2021-10-30:
...then... the 4MAX_TMC version should work on stock. Let's wait for feedback. 

And thank you again for your provided config!

JonnyHaystack, 2021-10-30:
I've been using alfrank's config with MEGA_P_TMC environment. I can test with the new environment at some point :smile:

knutwurst, 2021-10-30:
After getting some feedback I added configurations for both, Stock and full TMC Versions. Now also for the original 4MAX Metal. 
The case light (controllable via touchscreen) should also work. Please let me know if there are any other issues. If not, I'll release the version in the upcoming days.

[KW_4MAX_4MAXPRO2_v1.3.1.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/7447746/KW_4MAX_4MAXPRO2_v1.3.1.zip)

q4wey, 2021-10-31:
> After getting some feedback I added configurations for both, Stock and full TMC Versions. Now also for the original 4MAX Metal. The case light (controllable via touchscreen) should also work. Please [...]

I'm new to this thread and have limited experience with marlin but I tested your hex files currently I have found the file titled (4MAXP2_v1.3.1-4MP2) to work but with 2 glitches. I don't know if these glitches are unique to my machine 4maxv2
or in the firmware itself. The display will run two intro cycles on startup and the stop print button will flicker until the tool head is parked. 

I hope this info is of anyhelp.

MasterYod4, 2021-10-31:
Hello first of all, I would like to thank you very much for your work!
I tested version 4MAXP2_v1.3.1-4MP2 on my 4Max Pro 2.0. What I noticed is that the light button now works but the normal 4 point leveling in Anycubic Menu doesn't work.
As described by q4wey, the intro runs twice but I couldn't detect the stop button flickering.

CruisenRainerWahnsinn, 2021-10-31:
> Hello first of all, I would like to thank you very much for your work! I tested version 4MAXP2_v1.3.1-4MP2 on my 4Max Pro 2.0. What I noticed is that the light button now works but the normal 4 [...]

Yes, same for me (tested 4MP2 stock version). The original "Leveling"-Button isn't working (like in the self compiled versions before).

@MasterYod4 
You can go to "print" -> "Special Menu" to get 4 point or even Mesh bed leveling.

MasterYod4, 2021-10-31:
@CruisenRainerWahnsinn Thanks for the hint.

knutwurst, 2021-10-31:
> The original "Leveling"-Button isn't working

I know... that's because it's not implemented at all. I just included the build paths and ne configurations and "hacked" the light control in. This isn't final at all. It's just a milestone. I'll take a look at anycubic's implementation of the leveling menu. If it's straight forward, i'll implement it also.

In the meantime, you can use the specialmenu like @CruisenRainerWahnsinn mentioned :)

JonnyHaystack, 2021-10-31:
Is the filament runout sensor working? I think some other people may have got it working in forks that I've seen

knutwurst, 2021-10-31:
@JonnyHaystack it's currently disabled until I figured out the correct pin.

hanebuechenes, 2021-11-01:
Is there also a version for the 4max pro 1? I have one and i can test it!

hanebuechenes, 2021-11-01:
Tested the version KW_4MAX_4MAXPRO2_v1.3.1.zip on stock 4max V1. seems to work nice, only the Y axis is wrong direction. Can i change this somewhere, or have it to be compiled again?

CruisenRainerWahnsinn, 2021-11-01:
> 
> 
> Tested the version KW_4MAX_4MAXPRO2_v1.3.1.zip on stock 4max V1. seems to work nice, only the Y axis is wrong direction. Can i change this somewhere, or have it to be compiled again?

Which hex file did you chose of the zip? There are 4 files.

I'm not sure if it is true for the 4max (1), but if you use TMC Version of 4MP2 the direction is switched. So the TMC version of it might fix this issue.

MasterYod4, 2021-11-02:
I played around a bit today and noticed the following:

Set flow rate - works
Preheat Ultrabase - works
4 Point Leveling - only drives up to point A, all other points are not approached
Mesh leverling - works
PID Tune Hotend - does not work
PID Tune Ultrabase - works

Yury-MonZon, 2021-11-03:
I second what MasterYod4 said. 

Also, I've tried both TMC and non-TMC version on 4Max Pro 2.0(which has TMC installed by default). TMC version has Z axis reversed.

On both versions I can read/send commands via USB, I've tried 57.6k, 115.2k and 250k baudrate - got corrupted data. The cable and drivers are fine - I can successfully flash the board. 

Stopping the printing process from the touchscreen shows me like a thousand of popup screens that stop is done. It doesn't happen if I pause the print and then stop it.
I'm able to test multiple times, if it could be useful to developer - let me know.

Daventrik, 2021-11-03:
i would be glad to help with testing but i didn't flashed my printer with these versions yet, still on stock but i have the "round corners" on a square and i hope that the new firmware can solve that issue since the profile tuning is not solving this. How can i help?

knutwurst, 2021-11-03:
1.4.0 Beta 2 is out now.

CruisenRainerWahnsinn, 2021-11-03:
> 
> 
> i would be glad to help with testing but i didn't flashed my printer with these versions yet, still on stock but i have the "round corners" on a square and i hope that the new firmware can solve that issue since the profile tuning is not solving this. How can i help?

Try setting K factor to 0.2

Send "M900 K0.20" and safe with M500
Try 190°C (if PLA)
Get a e3d V6 nozzle (fits on stock hotend)
=> big improvement

q4wey, 2021-11-04:
> 1.4.0 Beta 2 is out now.

Tested 1.3.0 beta 2 no changed just complied and uploaded. No menu pops up in the print menu for mesh leveling etc.

Daventrik, 2021-11-04:
> Try setting K factor to 0.2
>
> Send "M900 K0.20" and safe with M500 Try 190°C (if PLA) Get a e3d V6 nozzle (fits on stock hotend) => big improvement

I will buy a e3d v6 nozzle but what would be recommended? Brass? copper plated, stainless steel? Thank you!

Yury-MonZon, 2021-11-04:
Version 1.4.0.-b2 doesn't have linear advance support
`Unknown command: M900`

knutwurst, 2021-11-05:
Fixed in dd169b7 and will be included in the next release.

Yury-MonZon, 2021-11-05:
You are the man! That was quick!  Thanks a lot!
Are you planning to release 1.4.0-beta.3 with those fixes any time soon?

q4wey, 2021-11-06:
@knutwurst 

Tested the new release 1.40.0.beta 2 today. When using the built in remove filament tool while a print is paused the filament will begin to remove but then the printer reboots so you cant resume the print.

Your doing great work on this mate, many thanks.

knutwurst, 2021-11-06:
@Yury-MonZon dunno... I have to fix some more things as you can see ;) In the upcoming days I think. But you can check out the master branch and compile it by yourself :) 

@q4wey Does this only apply to the 4MAX Pro v2?

CruisenRainerWahnsinn, 2021-11-06:
Didn't test the latest alphas yet. Still very happy about 1.3.1 :) already

**Concerning ENV and 4max types:**
There are several 4max out. Just wondering if env "4max" is meant to be "4max pro (1)". There should be 4max, 4max pro, and 4 max pro 2 (afaik)

**Concerning acceleation/jerk parameters that are working very nice for me:**
```
Recv: echo:; Steps per unit:
Recv: echo: M92 X100.00 Y80.00 Z800.00 E423.50
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X150.00 Y150.00 Z25.00 E30.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X2000.00 Y2000.00 Z60.00 E10000.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P1500.00 R1500.00 T1500.00
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.02
```

[image]
:

knutwurst, 2021-11-06:
@CruisenRainerWahnsinn take a look at the Beta (not alpha). ;) All 4MAX variants should be covered. Beta 3 is on the way with more improvements.

CruisenRainerWahnsinn, 2021-11-06:
> 
> 
> @CruisenRainerWahnsinn take a look at the Beta (not alpha). ;) All 4MAX variants should be covered. Beta 3 is on the way with more improvements.

I'll check them out 100%. Thank's a lot for your work!

Yury-MonZon, 2021-11-06:
@CruisenRainerWahnsinn Print acceleration 1500? Looks a bit high to me. What are your print speeds with these settings? Also, how did you manage to install e3d v6 nozzle? It is longer, isn't it?

CruisenRainerWahnsinn, 2021-11-06:
> 
> 
> @CruisenRainerWahnsinn Print acceleration 1500? Looks a bit high to me, how did you get rid of the ringing? What are your print speeds with these settings? Also, how did you manage to install e3d v6 nozzle? It is longer, isn't it?

It's working great for me like this. 

This is the latest Cura Profile I'm using (from the print you see):
https://www.dropbox.com/s/qxn14x4ppobsnq3/4MaxPro2.0_PLA_0.2mm_192C.curaprofile?dl=0

The only thing you need to dadpt to the e3d v6 nozzle, is the height of the fan duct (needs little sanding or reconstruction). The e3d v6 nozzle fits the stock hotend perfectly.

q4wey, 2021-11-06:
> @Yury-MonZon dunno... I have to fix some more things as you can see ;) In the upcoming days I think. But you can check out the master branch and compile it by yourself :)
> 
> @q4wey Does this only apply to the 4MAX Pro v2?

yes my apologies this was tested on the 4max pro 2.0 i should have specified

q4wey, 2021-11-14:
Tested 4MAXP2_DGUS_v1.4.0-b3.hex on the 4max pro 2.0. When printing and the pause button it pressed a message of SD pullout will pop on the screen until the printer head it parked. after the head is parked 3 beeps are made but when you try to resume the print it just wont resume.

Yury-MonZon, 2021-11-14:
> it just wont resume
I also have this sometimes. Wondering why is this not a constant thing?

knutwurst, 2021-11-14:
@q4wey and @Yury-MonZon are you sure that you have the DGUS variant with Anycubic 0.0.2 Display? There is a Pause/Resume bug in the original DGUS II.

Yury-MonZon, 2021-11-14:
I'm sorry, didn't know that they are different. How do I get the model of the screen?

CruisenRainerWahnsinn, 2021-11-15:
> 
> 
> I'm sorry, didn't know that they are different. How do I get the model of the screen?

There are some sources here:

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(English)#2-identification-of-the-display

&

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Types-of-Anycubic-Touchscreens

Yury-MonZon, 2021-11-15:
@knutwurst Yeah, I have Anycubic 0.0.2 display (DGUS clone) - the text 'Ready' is sharp and small.

luknz, 2021-11-16:
Hi, first of all thank you very much for your effort! Knutwurst has been running flawlessly on my i3 Mega for months. Today I also flashed my Anycubic 4Max Pro 2.0 with V 1.4.0-beta.3. Since then he has been making strange whistling noises when printing, like an old Internet modem. Does anyone know what could be causing this?

Truggy, 2021-11-20:
Not sure what I did wrong. I have v002 board 4mp v2.

when I stop a print the system crashes and is ina continuous reboot until hard switch is switched off.

I have the clone screen as well - same behavior with stock steppers or all tmc 2208. Did reset and load defaults in eprom.

thanks,
John

Yury-MonZon, 2021-11-28:
> continuous reboot until hard switch is switched off
When I do this I have multiple messages on the screen. So I use pause and then stop. And after that printer doesn't print when you start a new print.

Truggy, 2021-11-28:
That the same as I have. Ok. Wasnt sure if something was up with the mainboard.

dspitzer, 2021-11-30:
After a lot of fiddling I managed to install a BLTouch in the 4Max Pro 2 and with the latest master the auto levelling works perfectly - thanks a lot @knutwurst ! 

I'm using the 4MAXP2_DGUS_BLT_10 and the only parameter that I changed was the NOZZLE_TO_PROBE_OFFSET.

A few quirks that I noticed so far: After booting, the display immediately reboots (it basically shows the startup animation twice). Also, stopping a print results in a boot loop - I have to turn the printer off every time I stop a print.

Truggy, 2021-12-01:
I get the same but I used the non blt firmware.

SchimmelSepp, 2021-12-01:
Hey good morning,
Is it just recommended to use CURA or do i must use CURA with the Firmware from Knutwurst. Because i think Prusa works better for details & corners.

CruisenRainerWahnsinn, 2021-12-04:
@SchimmelSepp 
Most users I crossed, were using cura, some prusa. I saw good results on both.

Concerning Display and crashes:
I recently found out that my octoprint server is solving the bugs you're describing (but the double starting animation)
My machine don't crash and i can navigate through the whole menu (on the display) the whole time - while printing and always can access the special menu, which isn't the case for non-print-server-users.

SchimmelSepp, 2021-12-04:
> @SchimmelSepp Most users I crossed, were using cura, some prusa. I saw good results on both.
>
> Concerning display and crashes: I recently found out that my octoprint server is solving the bugs you're [...]

Yeah, i use prusa and the result with KW-Firmware is amazing instead of the Standard. Me either had just the Problem when i hit Stop the Printer reboots it selfs 2-6 times (idk why it does sometime 2 times and sometime up to 6) but yesterday i printed for 9 Hours and it stopped and didnt finished the printing, but the printer gived me via the Display the text „Print is finished“. And the Layerd looked good and didnt hat print fails.

Truggy, 2022-01-04:
Do we think a release 4 will happen?

CruisenRainerWahnsinn, 2022-01-17:
@knutwurst 

Just compiled and installed the beta 4 from today on my totally stock 4MP2.

Needed to chose DGUS-Version as env, because the stock 4MP2-env had no "special menu".
Maybe it got mixed up. 

I wil run this vversion now and report all i find here.

CruisenRainerWahnsinn, 2022-01-20:
> @knutwurst

The DGUS-Version is working fine for the 100% stock printer!
No issues so far (with octoprint involved)

knutwurst, 2022-01-21:
There is nothing mixed up afaik. There are (older) 4MP2 printers that have an original DWIN/DGUS II display. They need the firmware without DGUS in the name. The newer printers with Anycubic 0.0.2 "DGUS Clone" need the _DGUS firmware.

That was already the case with the Mega X.

JonnyHaystack, 2022-01-21:
That does sound like a somewhat confusing naming convention :stuck_out_tongue:

CruisenRainerWahnsinn, 2022-01-21:
Haha, yes! 

Thanks for your work!

I'm 100% sure ja have the clone screen:
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(English)#23-the-anycubic-002-display-dgus-clone

Special menu looked like this since 1.3.0 (self compiled)
https://raw.githubusercontent.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/master/images/manual_dgus_specialmenu.jpg

Since 1.4.0b4  i have to chose DGUS to get the special menu back and it is still looking like before (not like original DWIN)
(I have a stock printer - no changes made on drivers or display)

CruisenRainerWahnsinn, 2022-01-22:
Otoprint error message which cause "promptly stop" on 4MP2-DGUS, when using mesh bed leveling with M420 S1 in the start gcode. without the command it has no error message and prints.

```
Fehler vom Drucker gemeldet
Die Firmware Deines Druckers hat einen Fehler gemeldet. Darum hat OctoPrint die Verbindung geschlossen. Gemeldeter Fehler: Failed to enable Bed Leveling
```

CruisenRainerWahnsinn, 2022-01-22:
#https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/234

[17 more replies]

closed 2022-04-29 by github-actions[bot]