knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #68: Junction Deviation controll

dima4kas, 2021-01-29:
Hi all,

I really appriciate that given configuration for Anycubics printer.
I'm useing the i3 Mega S with latest Knutwurst version.
running well.

I tried to figure out the Junction Deviation given by Marlin. I would like to use this functionallity instead of Classik Jerk controll.

Regarding Marlin documentation the CLASSIC JERK overide the JD controll.
In line 1074 of you configuration "CLASSIC JERK" is enabeled.
#if ENABLED(CLASSIC_JERK)

means I have no chance to change this option with M205 (line 1123)
#if DISABLED(CLASSIC_JERK)

I do not know what will happen if M205 J0.013 (only as example) is in the G-Code, but with M503 I can see that M205 is allways with JERK even I changed it with M500.

Maybe I'm wrong, then let me know what I have to do on the Printer to change this controll to JD.

THANK You.

knutwurst, 2021-01-29:
Hi @dima4kas,
after a long test period with A-B tests with regard to the 16 Mhz 8 bit CPUs, no junction deviation has been used since version 1.1.7.
If you use a Cartesian printer, JD has no advantage anyway, since exactly one axis is controlled for each direction of movement. That is why JD only has the effect on Cartesian printers that the value is converted internally into normal jerk values. Since this takes place at runtime, these calculations consume too much computing power, which is then reflected in poorer print results.

When it comes to firmware, I attach great importance to the highest possible print quality and most people want the best out of the box result. That's why JD is disabled.

But you can reactivate it manually! I have saved all values ​​in the firmware so that you can switch JD on again at any time: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/a5446988883a62d899077c8e9883b02e9512587a/Marlin/Configuration.h#L1074

Of course you have to compile the firmware yourself with PlatformIO. But you can also install 1.1.6 to test it. Everything here still works with JD. 

Best,
Olli

dima4kas, 2021-01-29:
Hi Olli,

thank for that fast response. 
What I understood is that JD is changing the path, whereas jerk is only controlling the acceleration of the axis. 
I do not know who to see what the CPU is telling to the motors, but how do you know that in the case of cartesian machine CPU is changing JD into normal jerk. 
Al normal CNC machines are controlled by exactly one motor. That is no reason to don't use JD. 
A please maybe one of you has deep knowledge, so let me know how do you tested it and where I can see what CPU is telling to motors. 

Thank you

Br
Dimi

knutwurst, 2021-02-04:
It's not "changing JD into normal jerk" but the behaviour is the same. JD is actually there to calculate the cornering speed based on the angle and the common speeds of the motors involved. All axes are always included. Here's a good explanation of how it works: https://reprap.org/forum/read.php?1,739819

Unfortunately, I cannot give you any information about how much computing time is required and how which motors are controlled. I have only tested it with several printers (and Atmega2560 hardware) and found that with many small steps in Gcode, the computing power was insufficient to calculate the juction deviation for each segment in the necessary time. The result was blobs and stuttering.

You are welcome to do your own test series. Of course you can also print with JD and I would be happy if you share your experiences.

However, at the moment I believe that JD is not suitable for this type of printer. At least not with the hardware.

closed 2021-03-12 by github-actions[bot]