knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #69: Head Crash with 5V BLTouch 3.1

agentchimendez, 2021-01-30:
### Bug Description

Installed BLTouch 3.1(genuine) and now the head crashes during the first steps of BLTouch calibration. Ignores input from both z-stop switches on Mega S and does not recognize input from BLTouch pin. 

### My Configurations

Issue persists with both MEGA_S_DGUS_BLT_10_v1.1.9 and MEGA_S_DGUS_BLT_11_v1.1.9.

Compiled new build with 5V mode options as seen in [Issue:25](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/25#issue-709729122). The FORCE_SW_MODE does seem to have taken effect as crashing with this compilation is much slower than with precompiled .hex. But otherwise the issue persists when beginning the BLTouch calibration from the menu.

I have not worked with Marlin modification before and suspect I may have not adjusted other definitions properly. My mainboard fans now whine. I have attached the Configuration.h and Configuration_adv.h files.

I followed the alternate wiring guide found in the [wiki](https://raw.githubusercontent.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/master/images/bltouch_wiring_alt.jpg) A multimeter probe indicates a stable connection to black and white wires.

### Troubleshooting Information

M280 P0 S10/60/90 all work as expected. M119 only ever shows open but that appears to be expected behaviour.

When pin is manually pushed with a finger one can feel the Hall Effect vibration. A multimeter shows approximately 5v current when pin is triggered manually.

QR code on BLTouch leads to this [datasheet](https://5020dafe-17d8-4c4c-bf3b-914a8fdd5140.filesusr.com/ugd/f5a1c8_d40d077cf5c24918bd25b6524f649f11.pdf) which indicates a 4.8-5.1v operating range.

### Steps to Reproduce

1. Install BLTouch using [Knutwurst's Guide](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/BLTouch-Installation-(english))
2. Follow configuration to test 2. Pin moves as expected but If allowed to continue levelling from menu head will crash.
3. Follow configuration steps to 20.
4. Head will crash on first attempt to trigger BLTouch.
5. Printer ignores both Z endstops and will continue moving downwards until switched off.
6. Repeatable behaviour on all builds. 

**Expected behavior:**  BLTouch Z- triggers and indicates mesh point.
**Actual behavior:** Z- is not triggered in Marlin and head crashes.

#### Additional Information

I have not yet tried compiling in debug mode. From research it appears that this may need to be my next step in order for the Terminal to show a pin trigger event. Any advice on this point would be appreciated. 

[config5Vflags.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/5897148/config5Vflags.zip)

agentchimendez, 2021-02-05:
Compiled debug mode firmware. Used M43 to attempt pin debugging.

M43 shows "Recv: PIN:   2   Port: E4        Z_MIN_PROBE_PIN" but I am unclear is this is indicating BLTouch or not.

M43 S runs BLTouch debug and fails on detecting trigger after 30 seconds. Same failure with M43 S P0 or P2.

Running M280 P0 S10 and then M43 P0 or P2 W does not detect manual trigger.

M43 E detects endstops as expected but not BLTouch trigger.

M43 W causes printer to time out and eventually disconnect Octoprint. Copied output below. PIN: 11 message shows every time regardless of input. Unclear if PIN: 15 message is from triggering BLTouch as timeout and disconnect occurs after these two lines. No change to output if Z-stops are triggered rather than BLTouch.

Send: M43 W
Recv: Watching pins
Recv: PIN:  11   Port: B5        SERVO0_PIN
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
[...]
Recv: PIN:  15   Port: J0        <unused/unknown>
Recv: PIN:  15   Port: J0        <unused/unknown>

Not sure where to go next. Any suggestions?

agentchimendez, 2021-02-07:
Turns out I'm just terrible at both soldering and multimetering.

Got pin detection with M43 S in debug mode after repairing my awful soldering. Reassembling and reloading the base firmware to confirm with final tests. 

Ticket should close later today or tomorrow.

For other people who get as frustrated as I was this month; always remember stubborness will eventually make up for lack of skill.

agentchimendez, 2021-02-08:
Probe is working. Closing ticket.

Next problem is Z-offset. No matter what value I set with M851 it only marginally changes the gap between nozzle and bed. M503 returns the last set Z-offset however. 

Thanks again for the great firmware and great wiki resources. Sorry for jamming up the forum but hopefully it will help others in the future.

closed 2021-02-08 by agentchimendez

knutwurst, 2021-02-08:
Hi @agentchimendez !
I'm a little embarrassed that I didn't answer, but I was in the hospital for a while. But now I feel better. I have to admit that I am thrilled with how thoroughly you described the problem and how well you investigated (and fixed!) the problem. I am pleased that everything is now working and I thank you very much for the helpful information. There are certainly many others out there who have a similar problem and then don't know where to look. Thank you and have fun with your printer!