knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #71: Not a Bug just an HELP

wes1993, 2021-01-30:
Hello to everyone!!! 
I have bought from amazon the Original BLtouch V3.1:
 https://www.amazon.it/Antclabs-Bltouch-livellamento-sensore-stampante/dp/B01FFV2TOS/ref=sr_1_5?__mk_it_IT=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=bl+touch&qid=1612027158&quartzVehicle=2590-535&replacementKeywords=touch&sr=8-5

I don'w know why but seems that won't work well, i have some question for check if work well:
1. it's normal that the red light flash also when the printer it's in standby?
2. if the printer start the printing process the red and lblue light stay on
2. it's normal that whe i do the calibration the bed is very distant from the hot end?
3. it's normal that if i lauch the bed calibration without pronterface start but after go to home go to the first part when it's supposed to start the first calibration and hang here with red light flashing?

I hope someone can help me
Best Regards
Stefano

wes1993, 2021-01-30:
I have attached a photo of when the sensor find the final point:

[image]

wes1993, 2021-02-01:
UPDATE!!!
I have found that with your firmware 1.1.1 with BLTouch support downloaded from here (https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/tag/1.1.1) i don't have any problem!!!
So what is the difference related to the HotEnd temperature etc from 1.1.1 and1.1.9?

I really hope someone could help me!!
@knutwurst Again thanks alot for your beautiful work!!!!

wes1993, 2021-02-01:
I think i have finally solved the problem.
I will do more test in the next days but the solution is:
Flash Firmware
Power Off Printer
Turn ON
Pronterface
M502 (Load setting from firmware)
M500 (save settings in EEPROM)
Power Off printer
Turn ON
Print

wes1993, 2021-02-01:
I have claim victory too soon...
Same problem... the print start and after some seconds the hotend temperature go down.. from 245 to 226 degree in some seconds...

wes1993, 2021-02-01:
NEW UPDATE:
Disabling the fan of the HotEnd seems that could print, so there is a way to make the hotend controller more faster when the temperature down?

knutwurst, 2021-02-02:
Hi @wes1993 ,
Unfortunately, I am currently in the hospital, which is why it is difficult to respond to bug reports promptly. As far as I can see, this is not a firmware bug. Your problem is already described in the wiki or FAQ: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/FAQ-(english)

Your fan may run at a maximum of 70% if it is the original one. With the firmware he gets more power than with the original. This is also in the FAQ. In addition, if the temperature cannot be maintained, you should either pull a silicone sock over the nozzle or do the PID tuning for your preferred temperature.

wes1993, 2021-02-02:
Hello @knutwurst 
I am very sorry that you are in the hospital, I hope is not at all serious 🤞
Thanks again for your work and your help!! You firmware is wonderful!! :)
I have seen that in cura if I change the fan speed settings nothing change... so now I know why if I put 70%, 50% etc nothing change.
Thanks for you help!! :)
I wish you a quick recovery and sorry for my bad English... :( I’m Italian :(

wes1993, 2021-02-02:
P.S. I have seen now the wiki... sorry for my stupid oversight

closed 2021-02-02 by wes1993