knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #72: [BUG] Mesh bed leveling: not saving to EPPROM

OkiGraphics, 2021-01-31:
### Bug Description

I had to re-level the bed before large print. I re-leveled trough a special menu directly via display. After going trough all the points, I hit save to EEPROM and then end mesh bed leveling. I have restarted the printer. When I go to mesh bed leveling again, it starts again on old Z values. I tested it several times, I even made the first point way too off on purpose (so I feel no resistance), but If i start to re-level again, I feel strong resistance (as before re-leveling).

### My Configurations

"MEGA_S_DGUS_v1.1.9.hex"

### Steps to Reproduce

1. Run trough all mesh bed leveling points
2. Save to EEPROM.
3. Start mesh bed leveling again.

**Expected behavior:** [What you expect to happen]
Starting the next mesh leveling process on to positions saved to EEPROM last time.
**Actual behavior:** [What actually happens]
Mesh bed leveling process starts on old (incorrect) positions.

#### Additional Information
What might by important, last time I (sucessfully) did mesh bed leveling via Pronterface.

CruisenRainerWahnsinn, 2021-04-08:
Are there any news/solutions available to use mesh bed leveling reliably? It worked with ponterface but with octoprint it seems like mesh data got lost.

knutwurst, 2021-04-09:
Hi @CruisenRainerWahnsinn and @OkiGraphics .
Can you please test the new 1.2.0-alpha.1 version if the issue still persists? I've fixed a mesh loading bug, but it might be another issue.

knutwurst, 2021-04-10:
1.2.0-alpha.2 is released. Please test if the problem still persists.

CruisenRainerWahnsinn, 2021-04-11:
> 
> 
> 1.2.0-alpha.2 is released. Please test if the problem still persists.

Tested. Thank you!

MEGA_P_TMC + config files from here: 
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/65
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/5875815/4MP_2.0_KW1.1.9.zip

Still got the error when trying to print via octoprint:

[image]

`Recv:  T:67.84 /205.00 B:60.12 /60.00 @:127 B@:51 W:?
Recv:  T:68.45 /205.00 B:60.14 /60.00 @:127 B@:48
Recv:  T:69.68 /205.00 B:60.16 /60.00 @:127 B@:44 W:?
Recv: echo:busy: processing
Recv:  T:71.47 /205.00 B:60.17 /60.00 @:127 B@:40 W:?
Recv:  T:72.06 /205.00 B:60.23 /60.00 @:127 B@:29
Recv:  T:73.31 /205.00 B:60.24 /60.00 @:127 B@:25 W:?
Recv: echo:busy: processing
Recv:  T:75.14 /205.00 B:60.32 /60.00 @:127 B@:7 W:?
Recv:  T:76.02 /205.00 B:60.32 /60.00 @:127 B@:4
Recv:  T:77.23 /205.00 B:60.32 /60.00 @:127 B@:0 W:?
Recv: echo:busy: processing
Recv:  T:79.02 /205.00 B:60.32 /60.00 @:127 B@:0 W:?
Recv:  T:79.67 /205.00 B:60.32 /60.00 @:127 B@:0
Recv:  T:80.83 /205.00 B:60.32 /60.00 @:127 B@:0 W:?
Recv: echo:busy: processing
Recv:  T:82.62 /205.00 B:60.27 /60.00 @:127 B@:0 W:?
Recv:  T:83.18 /205.00 B:60.31 /60.00 @:127 B@:0
Recv:  T:84.37 /205.00 B:60.26 /60.00 @:127 B@:0 W:?
Recv: echo:busy: processing
Recv:  T:86.11 /205.00 B:60.18 /60.00 @:127 B@:0 W:?
Recv:  T:86.69 /205.00 B:60.16 /60.00 @:127 B@:0
Recv:  T:87.85 /205.00 B:60.16 /60.00 @:127 B@:0 W:?
Recv: echo:busy: processing
Recv:  T:89.54 /205.00 B:60.14 /60.00 @:127 B@:0 W:?
Recv:  T:90.09 /205.00 B:60.11 /60.00 @:127 B@:0
Recv:  T:91.26 /205.00 B:60.03 /60.00 @:127 B@:8 W:?
Recv: echo:busy: processing
Recv:  T:92.96 /205.00 B:60.00 /60.00 @:127 B@:11 W:?
Recv:  T:93.78 /205.00 B:59.96 /60.00 @:127 B@:17
Recv:  T:94.93 /205.00 B:59.95 /60.00 @:127 B@:18 W:?
Recv: echo:busy: processing
Recv:  T:96.60 /205.00 B:59.82 /60.00 @:127 B@:43 W:?
Recv:  T:97.18 /205.00 B:59.80 /60.00 @:127 B@:45
Recv:  T:98.29 /205.00 B:59.80 /60.00 @:127 B@:47 W:?
Recv: echo:busy: processing
Recv:  T:99.96 /205.00 B:59.78 /60.00 @:127 B@:54 W:?
Recv:  T:100.50 /205.00 B:59.77 /60.00 @:127 B@:57
Recv:  T:101.60 /205.00 B:59.72 /60.00 @:127 B@:69 W:?
Recv: echo:busy: processing
Recv:  T:103.20 /205.00 B:59.74 /60.00 @:127 B@:69 W:?
Recv:  T:103.75 /205.00 B:59.67 /60.00 @:127 B@:84
Recv:  T:104.79 /205.00 B:59.63 /60.00 @:127 B@:95 W:?
Recv: echo:busy: processing
Recv:  T:106.42 /205.00 B:59.68 /60.00 @:127 B@:92 W:?
Recv:  T:106.94 /205.00 B:59.67 /60.00 @:127 B@:97
Recv:  T:108.02 /205.00 B:59.72 /60.00 @:127 B@:92 W:?
Recv: echo:busy: processing
Recv:  T:109.85 /205.00 B:59.80 /60.00 @:127 B@:84 W:?
Recv:  T:110.38 /205.00 B:59.80 /60.00 @:127 B@:86
Recv:  T:111.45 /205.00 B:59.80 /60.00 @:127 B@:89 W:?
Recv: echo:busy: processing
Recv:  T:113.06 /205.00 B:59.85 /60.00 @:127 B@:86 W:?
Recv:  T:113.58 /205.00 B:59.87 /60.00 @:127 B@:85
Recv:  T:114.63 /205.00 B:59.96 /60.00 @:127 B@:70 W:?
Recv: echo:busy: processing
[... 221 lines cut ...]
Recv: ok
Send: N20 M82*43
Recv: ok
Send: N21 G28*32
Recv:  T:205.63 /205.00 B:60.00 /60.00 @:65 B@:48
Recv: echo:busy: processing
Recv:  T:206.05 /205.00 B:60.00 /60.00 @:58 B@:51
Recv: echo:busy: processing
Recv:  T:207.19 /205.00 B:60.14 /60.00 @:39 B@:26
Recv: echo:busy: processing
Recv:  T:208.32 /205.00 B:60.16 /60.00 @:24 B@:20
Recv: X:-6.00 Y:0.00 Z:0.00 E:4.00 Count X:-600 Y:0 Z:0
Recv: ok
Send: N22 M420 S1*87
Recv: echo:Invalid mesh.
Recv: Error:Failed to enable Bed Leveling
Changing monitoring state from "Printing" to "Error: Failed to enable Bed Leveling"
Send: M112
Send: N23 M112*16
Send: N24 M104 T0 S0*23
Send: N25 M140 S0*82
Changing monitoring state from "Error: Failed to enable Bed Leveling" to "Offline (Error: Failed to enable Bed Leveling)"
Connection closed, closing down monitor`

Any suggestions welcome!

knutwurst, 2021-04-23:
I don't know what's the difference to the 4MAX pro, but the 1.2.0 seems to work on all supported printers. :/

closed 2021-05-29 by github-actions[bot]