knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #73: [BUG] Too awesome

gloomytree, 2021-02-03:
### Bug Description

This repo and fw is too awesome. It has a documentation, continuous development, structure, activity... 

**Expected behavior:** Stale repo, flooding issues, no reaction from dev

**Actual behavior:** The oposite. 

#### Additional Information

I know opening an issue to say thank you is not the proper way, but here I am, rebel me. Ty for the hard work and sharing. 

Also important: 

We're no strangers to love
You know the rules and so do I
A full commitment's what I'm thinking of
You wouldn't get this from any other guy
I just wanna tell you how I'm feeling
Gotta make you understand

Never gonna give you up
Never gonna let you down
Never gonna run around and desert you
Never gonna make you cry
Never gonna say goodbye
Never gonna tell a lie and hurt you

knutwurst, 2021-02-04:
I think that's the nicest thing I've read during this time (and during the pandemic). Plus, I just laughed out loud when I read it. Thank you! I am really happy that my free time work is valued <3

3dRikal, 2021-02-04:
I totally agree... This FW has been my initiation to Anycubic mods and control. After the learning curve (Marlin, 3d Printing, Tuning, etc...) I come back to this page to see new bugs and see if I can reproduce or at least try and find a workaround or a fix. I will give back to this community that has so graciously given to me.

boredomwontgetus, 2021-02-15:
I have not tried it yet. Will give it a shot soon, tho. 

But just by the looks of the repo i already wanted to agree to this Issue!
The amount of work put into here looks incredible. All my respect to the dev/maintainer! 

Keep up the good work, mate. We appreciate it a lot.

closed 2021-03-23 by github-actions[bot]

sims-60, 2021-03-30:
I want thst it is opened and not closed