knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #74: [BUG] after filament out detection hotend cools down and can't reheat

brunoosti, 2021-02-04:
Not sure if it's a bug or thermal protection. Have experienced it throughout versions.
If I do not attend the printer for some time after it detects run-out filament, it cools off the hotend (maintaining the bed temp) not being able to heat it again. If I try to reheat over the printer menu I can hear the tic tic tic sound meaning it wasn't able to  reach the desired temp. Bug? Feature? Should I always be around when the printer detects filament run-out?  My filament sensor is giving some false detections, though...

pwtdo, 2021-03-01:
Just experienced the exact same problem this morning. Woke up to a filament out error (filament was fine), and after pressing OK the printer was not able to re-heat the hotend. Bed temperature was maintained throughout.

3dRikal, 2021-03-04:
I also have noticed this but it is not systematic. The only way to gain back control is to reset the printer.

I'm using the mega s, dgus, tmc version of the firmware (1.1.9)

brunoosti, 2021-04-04:
Yesterday had it again. Not sure if its caused by user error or the printer that won't reheat. Testing, I removed the filament to see if  printer would not heat again, but after decreasing the temp and pressing continue, the printer started heating the hotend to the desired temp, and printing after that.

3dRikal, 2021-04-10:
I think this is related to the bug with the filament sensor already opened and knutwurst is aware. Seems to be related to Marlin 2 and the way it is handling the sensor and is a more complex bug than anticipated. We can probably link these bugs and close this one..

brunoosti, 2021-04-10:
Sure. Didn't know those where related. 
Indeed when I have a long print to do and I know filament won't run out, I tend to disable the sensor.

closed 2021-04-10 by brunoosti